espressif/arduino-esp32 #5239: Olimex ESP32-POE Ethernet PHY and arduino-esp32 2.0.0/IDF 4.4

Simonefardella, 2021-06-03:
Hi guys,

In these days i'm trying to use the arduino-esp32 2.0.0 beta version on my test lab boards ([OLIMEX ESP32-POE](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware)). 

Most of the functionalities are okay, but when i try to use the onboard ETH lan chip (LAN8720) nothing works.

I know that "nothing works" is not correct in the IT/Electronics field, but i'll try to explain more in depth in the following paragraphs.

Preliminary informations: this board uses the GPIO12 as ETH_PHY_POWER and the GPIO17 as ETH_CLK_MODE.

I'm using the standard sketch that works for other versions of arduino-esp32 in order to power up the ETH, acquire an IP from DHCP server and do a sample request to google: [here it's the sketch](https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_Arduino/ESP32_PoE_Ethernet_Arduino.ino)

but when i open the arduino console i see those messages:

[image]

At this point, it seems that the ETH phy chip isn't initialized correctly, so i've picked up my oscilloscope and this is what i've seen:

Older esp32-arduino (1.0.4 and 1.0.6), probe on GPIO17, as you can see, there is a clean and nice looking 50MHZ signal coming out from GPIO17, as expected:

[image]

When i use the esp32-arduino 2.0.0, probing GPIO17 there isn't any signal coming out from that pin, as you can see:

[image]

...so, i've asked to gitter, and @lbernstone clearly says that starting from esp-arduino 2.0.0 the eth clock [gpio output is hardcoded in sdkconfig](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig#L616) and if i want to modify it i need to compile my own libraries. 

Anyhow...i've repeated the same probing test on GPIO0 (...thinking that the phy clock out comes out from the "default" GPIO0 pin...), but also in this case, nothing comes out from that:

[image]

This seems odd...i mean, in the past using a define (`#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT`) everything was okay and no custom libraries were needed...how it's possible that now anyone who is not using the same pinout definition for ETH phy chips of the ESP32-Ethernet-Kit needs a custom compiled libraries?

Please clarify those aspects and help us to make the phy chips work again, because i think that i've misinterpreted those changes and maybe is more simple than reported!

friedrichengel, 2021-06-03:
same problem here. Are using also the clock on an other pin.  So would be great to have an soultion that is not building your own Librarie.

rene-dev, 2021-06-03:
you are not passing a clock mode argument to eth.begin in your sketch.
https://github.com/espressif/arduino-esp32/blob/e6ba8c7ac9c55fda98253c9aa4fdc1aba9caa44e/libraries/WiFi/src/ETH.cpp#L114
that should still work.

rene-dev, 2021-06-03:
oh, it looks like the argument was removed. That is not good, as I also require this.

Simonefardella, 2021-06-04:
Exactly, this parameter was removed...but i think that is not the only problem (i've tested hardcoding this parameter in `ETH.cpp` but the problem is the same...) 
This is absolutely not a good choice...think like Olimex or other companies that uses ETH PHY chips...it's totally out of mind that every company has to build and mantain their own eth library!

friedrichengel, 2021-06-04:
i have hope that they will change something like that in the final release.We are using the same pinout like olimex uses.

Simonefardella, 2021-06-04:
Absolutely an issue, a clarification is needed! @me-no-dev...maybe you're the right guy! 😀

sauttefk, 2021-06-04:
Hmm... I'm also missing this. I've added this feature 4 years ago and now it has vanished... :-(
https://github.com/espressif/arduino-esp32/pull/916

Simonefardella, 2021-06-04:
@sauttefk, initially i was suspecting that there was only this problem (missing `eth_clock_mode_t clock_mode` parameter), so i've manually patched the `ETH.cpp` setting manually  `eth_config.clock_mode = 3` (RMII clock output from GPIO17).

Nothing changes, same result.

Simonefardella, 2021-06-04:
I can drop a minor update about that, i've tried to compile a "plain" esp-idf project (not esp32-arduino), basically i've used the iperf example, configured GPIO17 as ETH_CLOCK_GPIO17_OUT, flashed using idf.py and it works.

I've tested it using IDF 4.4, here is my sdkconfig: <details><summary>Click to expand</summary>
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
# end of Bootloader config
[... 1258 lines cut ...]
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```
</details>

So, i think that something on esp32-arduino side is not aligned and and needs to be fixed.

me-no-dev, 2021-06-10:
Hi All. ESP-IDF API has changed much since v3.3. RMII Clock out is no longer controlled from the app, so some things ended up misconfigured :) Please try using the github version and `ethernet` branch to try out ethernet with clock on GPIO17. If someone has a board with another clock pin, please let me know so we can try a fix for that too :)

Simonefardella, 2021-06-10:
I'll try it right now, thanks @me-no-dev: i have a bunch of boards on my hand i can test it and let you know

PR #5274 (2021-06-16): Enable clock output for Ethernet RMII on ESP32
fixes #5239
commit 1f4f2b6e (2021-06-16): Enable clock output for Ethernet RMII on ESP32 (#5274)

fixes #5239

* Add back clock mode argument
closed 2021-06-16 by me-no-dev