espressif/arduino-esp32 #5242: use xTaskCreatePinnedToCore in WIFI_MODE_APSTA

Rezaasghariir, 2021-06-03:
----------------------------- Remove above -----------------------------

### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module
|<B>IDE name</B>|Arduino in VSCode|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|Mac OSX|Ubuntu|

### Description:
hello 
when i use xTaskCreatePinnedToCore  in WIFI_MODE_APSTA wifi mode my accesspoint active but not connnect to local wifi.
when remove xTaskCreatePinnedToCore  work true.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
    xTaskCreatePinnedToCore(serialCheckTask,   // task
                            "serialCheckTask", // name for task
                            8192,              // size of task stack
                            NULL,              // parameters
                            1,                 // priority
                            &serialCheckTaskHandle,
                            1 // core
    );

lbernstone, 2021-06-03:
This is a real time OS.  You must put a delay/yield in your task if you want other code to run. https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/FreeRTOS/FreeRTOS.ino

Rezaasghariir, 2021-06-03:
this is my task 
//
void serialCheckTask(void *pvParameters)
{
    uint8_t data = 0;
    uint8_t client = CLIENT_ALL; // who sent the data
    while (true)
    { // run continuously
        while (any_client_has_data())
        {
            if (Serial.available())
            {
                client = CLIENT_SERIAL;
                data = Serial.read();
            }
            else if (inputBuffer.available())
            {
                client = CLIENT_INPUT;
                data = inputBuffer.read();
            }
            else
            {
                //currently is wifi or BT but better to prepare both can be live
#ifdef ENABLE_BLUETOOTH
                if (SerialBT.hasClient() && SerialBT.available())
                {
                    client = CLIENT_BT;
                    data = SerialBT.read();
                    //Serial.write(data);  // echo all data to serial
                }
                else
                {
#endif
#if defined(ENABLE_WIFI) && defined(ENABLE_HTTP) && defined(ENABLE_SERIAL2SOCKET_IN)
                    if (Serial2Socket.available())
                    {
                        client = CLIENT_WEBUI;
                        data = Serial2Socket.read();
                    }
                    else
                    {
#endif
#if defined(ENABLE_WIFI) && defined(ENABLE_TELNET)
                        if (telnet_server.available())
                        {
                            client = CLIENT_TELNET;
                            data = telnet_server.read();
                        }
#endif
#if defined(ENABLE_WIFI) && defined(ENABLE_HTTP) && defined(ENABLE_SERIAL2SOCKET_IN)
                    }
#endif
#ifdef ENABLE_BLUETOOTH
                }
#endif
            }
            // Pick off realtime command characters directly from the serial stream. These characters are
            // not passed into the main buffer, but these set system state flag bits for realtime execution.
            if (is_realtime_command(data))
                execute_realtime_command(data, client);
            else
            {
                vTaskEnterCritical(&myMutex);
                client_buffer[client].write(data);
                vTaskExitCritical(&myMutex);
            }
        } // if something available
        COMMANDS::handle();
#ifdef ENABLE_WIFI
        wifi_config.handle();
#endif
#ifdef ENABLE_BLUETOOTH
        bt_config.handle();
#endif
#if defined(ENABLE_WIFI) && defined(ENABLE_HTTP) && defined(ENABLE_SERIAL2SOCKET_IN)
        Serial2Socket.handle_flush();
#endif
        vTaskDelay(1 / portTICK_RATE_MS); // Yield to other tasks
    }                                     // while(true)
}

//
i don't have problem  in only STA mode but in APSTA mode wifi dont conect to station

TheRealDJ, 2021-06-13:
Does the ESP32 crash?

Rezaasghariir, 2021-06-13:
No don't crashed. If fine my mistake.

closed 2021-06-13 by Rezaasghariir