espressif/arduino-esp32 #5246: Can't set AP hostname on ESP32

ToMe25, 2021-06-04:
### Hardware:
Board:							ESP32 DevKitC-VE-ND
Core Installation version:			3.2.1*
IDE name:						Platform.io, Eclipse 2021-03 (4.19.0)
Flash Frequency:					40Mhz
PSRAM enabled:                                    no, but installed on the board
Upload Speed:					921600
Computer OS:                                        Ubuntu 20.04

\* I couldn't figure out what version exactly was meant, this is what platformio shows as the platform version.

### Description:
I tried to set the hostname of an ESP32 in AP mode, so I can connect with it using the hostname from another device that joins this WiFi network.
For this I first set up all the basic WiFi stuff, and made a simple WebServer to test this with a browser.
The system I am coding this on is an Ubuntu 20.04 desktop, however I tried connecting to the server from various systems.
I unfortunately don't have an error message, since no method call fails.
My problem is that while the WebServer responds to opening http://192.168.4.1/, but not http://esp-ap/.
`esp-ap` also can't be pinged, while `192.168.4.1` can.
Is there something I can do to make this work?

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <WebServer.h>

const char *AP_SSID = "esp-ap";

IPAddress apIp(192, 168, 4, 1);
WebServer server;

void onAPStart(WiFiEvent_t event) {
	if (WiFi.softAPsetHostname(AP_SSID)) {
		Serial.println("Setting hostname");
	}
}

void setup() {
	Serial.begin(115200);
	delay(3000);
	WiFi.mode(WIFI_AP);
	WiFi.onEvent(onAPStart, SYSTEM_EVENT_AP_START);

	Serial.println("Opening AP...");
	WiFi.softAPConfig(apIp, apIp, IPAddress(255, 255, 255, 0));
	WiFi.softAP(AP_SSID);

	server.on("/", HTTP_GET, []() {
		server.send(200, "text/plain", "Some Response");
	});

	server.begin();

	Serial.println("Server ready.");
}

void loop() {
	server.handleClient();
	delay(20);
}

```

### Debug Messages:
```
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
Opening AP...
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
Setting hostname
Server ready.
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 16 - AP_STACONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED
[V][WebServer.cpp:292] handleClient(): New client
[V][Parsing.cpp:121] _parseRequest(): method: GET url: / search: 
[V][Parsing.cpp:227] _parseRequest(): headerName: Host
[V][Parsing.cpp:228] _parseRequest(): headerValue: 192.168.4.1
[V][Parsing.cpp:227] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:228] _parseRequest(): headerValue: Mozilla/5.0 (X11; Linux aarch64; rv:78.0) Gecko/20100101 Firefox/78.0
[V][Parsing.cpp:227] _parseRequest(): headerName: Accept
[V][Parsing.cpp:228] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
[V][Parsing.cpp:227] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:228] _parseRequest(): headerValue: de,en-US;q=0.7,en;q=0.3
[V][Parsing.cpp:227] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:228] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:227] _parseRequest(): headerName: DNT
[V][Parsing.cpp:228] _parseRequest(): headerValue: 1
[V][Parsing.cpp:227] _parseRequest(): headerName: Connection
[V][Parsing.cpp:228] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:227] _parseRequest(): headerName: Upgrade-Insecure-Requests
[V][Parsing.cpp:228] _parseRequest(): headerValue: 1
[V][Parsing.cpp:255] _parseArguments(): args: 
[V][Parsing.cpp:238] _parseRequest(): Request: /
[V][Parsing.cpp:239] _parseRequest():  Arguments:
```
During the run that produced this log I first tried to connect to http://192.168.4.1/, and then to http://esp-ap/ from a Raspberry Pi.
The first caused these log entries and worked, while the second seems to have caused no logging and didn't work.

sansillusion, 2021-06-04:
This is not how webservers/DNS works. An ssid is a wifi ap name not a
hostname. You may want to look in mDNS to set a name like esp.local if your
computer/rpi supports mDNS/Bonjour.

ToMe25, 2021-06-04:
Does this mean `softAPsetHostname` sets the SSID, not the device hostname?

I am aware of mDNS and am using it in my actual application, however I would prefer to use a hostname as that is more widely supported.

I assumed in AP mode the ESP32 would create some kind of minimal DNS DHCP combo to allow connected devices to reach each other using their hostnames, and `softAPsetHostname` would set its own host name on this server.
Is that wrong?

sansillusion, 2021-06-04:
DNS is meant for internet not local networks. This is why mDNS was invented.

ToMe25, 2021-06-04:
Does the ESP32 AP support connected devices addressing each other by their hostname without mDNS?

lbernstone, 2021-06-04:
No.  It is not a dns server, it is a wifi AP.  There is a DNSServer library, but it is not automatically populated by the internal dhcp server (you could do this in a callback if you really wanted the pain). As sansillusion says, this is exactly what mDNS is for.

ToMe25, 2021-06-04:
Thanks for the answer, seems I was wrong then.
Sorry, seems like I didn't fully understand what you meant @sansillusion.
I understood that as "This is a better way to do that" instead of answering whether what I wanted to do is supported.

Unfortunately the default DNSServer library seems to only support one record, which just doesn't really help.
I have not looked for very long, but I couldn't find a better one yet, and I don't think its worth the effort to implement my own.

So what does softAPsetHostname do then?

ToMe25, 2021-06-04:
> you could do this in a callback if you really wanted the pain

Assuming I did want to try this, which I am not sure about yet, where could I get the requested hostname from the dhcp server without changing dhcp internals?

If I end up doing this, I would make it a separate¹ open source library, and try to make the DNS server able to have manually added records, automatically add dhcp hostname requests, and optionally² fall back to an external DNS server if it has an internet connection and none of the local records match.

¹ separate as in not part of what I'm working on atm, not as in I am against having it a standard library.
² both as in I might not implement that, and as in I would make this something that can be enabled or disabled if I add it.

Is there some event/callback for getting the requested hostname that I haven't found yet?

lbernstone, 2021-06-04:
I think this would take significant changes to [dhcpserver](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c).  You would need to build and manage a list (or read from file), then when the SYSTEM_EVENT_AP_STACONNECTED event happens you ensure that the right ip address is given out, along with the hostname.  Giving out the dns info would then just be a lookup in the table.

ToMe25, 2021-06-04:
I have spent some more time looking into this, and unfortunately I don't think I will be able to implement this, as it seems to require changes to the dhcp server.
I thought I read somewhere that stuff was event based, so I hoped I could use those events, but it doesn't look like it.

Thanks for helping me :)

However I am still confused as to what `softAPsetHostname` is supposed to do then.

sansillusion, 2021-06-05:
I believe it's to set the DHCP client id. Like in my router it will display
the name with mac address instead of just the mac address.

ToMe25, 2021-06-05:
Thats what `setHostname` does, but why would I want that in AP mode when the ESP is likely the dhcp server?
Also looking at the source, `softAPsetHostname` specifically sets the hostname for the AP network interface, which will in most cases probably not be connected to an external dhcp server.
Setting the hostname in AP mode has to be meant to do something, right?
Or is that just for internal use?

TheRealDJ, 2021-06-13:
```softAPsetHostname()``` calls ```esp_netif_set_hostname()```. EspressIF's documentation for this function ([https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html) says

**The configured hostname overrides the default configuration value CONFIG_LWIP_LOCAL_HOSTNAME. Please note that when the hostname is altered after interface started/connected the changes would only be reflected once the interface restarts/reconnects**

Try calling ```WiFi.softAPsetHostname()``` before ```WiFi.mode(WIFI_AP)```?

ToMe25, 2021-06-13:
Thanks for the hint, however unfortunately it didn't seem to change anything.
I tried changing the test file to this, with the same result:
```cpp
#include <WebServer.h>

const char *AP_SSID = "esp-ap";

IPAddress apIp(192, 168, 4, 1);
WebServer server;

void setup() {
	Serial.begin(115200);
	delay(3000);
	WiFi.softAPsetHostname(AP_SSID);
	WiFi.mode(WIFI_AP);

	Serial.println("Opening AP...");
	WiFi.softAPConfig(apIp, apIp, IPAddress(255, 255, 255, 0));
	WiFi.softAP(AP_SSID);

	server.on("/", HTTP_GET, []() {
		server.send(200, "text/plain", "Some Response");
	});

	server.begin();

	Serial.println("Server ready.");
}

void loop() {
	server.handleClient();
	delay(20);
}
```
I also tried adding `WiFi.disconnect(1, 1);` directly before and/or after `WiFi.mode(WIFI_AP);` with this changed version, and I still can't ping `esp-ap` or reach http://esp-ap/ in a web browser.

ToMe25, 2021-09-02:
Would it make sense to open a feature request to add the mac address and the requested host name to [esp_netif_lwip.c#esp_netif_dhcps_cb](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/lwip/esp_netif_lwip.c#L670) and [esp_netif_types.h#ip_event_ap_staipassigned_t](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/include/esp_netif_types.h#L139)?
The callback is called [here](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c#L693).
At this position the mac address should be available as part of `m->chaddr`.
And the requested host name should be available somewhere in `m->options`.

Adding these two things should make it possible to work around this issue, and especially the mac address should be useful in many situations.
This would be a relatively small change, however I'm not sure if this would make sense to request since it would be an api break.

ToMe25, 2021-09-09:
I opened a feature request for that change.
It can be found here: espressif/esp-idf#7541

VojtechBartoska, 2022-03-30:
Hello @ToMe25 , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

ToMe25, 2022-03-31:
Using version `2.0.3-RC1` with platformio and this lightly updated sketch [esp-ap](http://esp-ap/) still wasn't reachable.

Here the updated sketch:
```cpp
#include <WebServer.h>

const char *AP_SSID = "esp-ap";

IPAddress apIp(192, 168, 4, 1);
WebServer server;

void onAPStart(arduino_event_id_t event) {
	if (WiFi.softAPsetHostname(AP_SSID)) {
		Serial.println("Setting hostname");
	}
}

void setup() {
	Serial.begin(115200);
	delay(3000);
	WiFi.disconnect(1, 1);
	WiFi.softAPsetHostname(AP_SSID);
	WiFi.mode(WIFI_AP);
	WiFi.onEvent(onAPStart, ARDUINO_EVENT_WIFI_AP_START);

	Serial.println("Opening AP...");
	WiFi.softAPConfig(apIp, apIp, IPAddress(255, 255, 255, 0));
	WiFi.softAP(AP_SSID);

	server.on("/", HTTP_GET, []() {
		server.send(200, "text/plain", "Some Response");
	});

	server.begin();

	Serial.println("Server ready.");
}

void loop() {
	server.handleClient();
	delay(20);
}
```
Here the log created by this updated sketch:
```
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[  3196][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
Opening AP...
[  3288][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
[  3288][V][WiFiGeneric.cpp:325] _arduino_event_cb(): AP Started
[  3296][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 10 - AP_START
Setting hostname
[  3305][V][WiFiGeneric.cpp:154] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
[  3317][V][WiFiGeneric.cpp:328] _arduino_event_cb(): AP Stopped
[  3326][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 11 - AP_STOP
[  3327][V][WiFiGeneric.cpp:325] _arduino_event_cb(): AP Started
[  3332][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 10 - AP_START
Setting hostname
Server ready.
[ 65592][V][WiFiGeneric.cpp:337] _arduino_event_cb(): AP Station Connected: MAC: 48:51:b7:77:72:b6, AID: 1
[ 65593][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[ 66295][V][WiFiGeneric.cpp:347] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
[ 66295][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[ 90758][V][WebServer.cpp:294] handleClient(): New client: client.localIP()=192.168.4.1
[ 90823][V][Parsing.cpp:121] _parseRequest(): method: GET url: / search: 
[ 90824][V][Parsing.cpp:227] _parseRequest(): headerName: Host
[ 90825][V][Parsing.cpp:228] _parseRequest(): headerValue: 192.168.4.1
[... 18 lines cut ...]
[ 91310][V][Parsing.cpp:121] _parseRequest(): method: GET url: /favicon.ico search: 
[ 91311][V][Parsing.cpp:227] _parseRequest(): headerName: Host
[ 91312][V][Parsing.cpp:228] _parseRequest(): headerValue: 192.168.4.1
[ 91319][V][Parsing.cpp:227] _parseRequest(): headerName: Connection
[ 91325][V][Parsing.cpp:228] _parseRequest(): headerValue: keep-alive
[ 91331][V][Parsing.cpp:227] _parseRequest(): headerName: User-Agent
[ 91337][V][Parsing.cpp:228] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/99.0.4844.84 Safari/537.36
[ 91352][V][Parsing.cpp:227] _parseRequest(): headerName: DNT
[ 91358][V][Parsing.cpp:228] _parseRequest(): headerValue: 1
[ 91363][V][Parsing.cpp:227] _parseRequest(): headerName: Accept
[ 91369][V][Parsing.cpp:228] _parseRequest(): headerValue: image/avif,image/webp,image/apng,image/svg+xml,image/*,*/*;q=0.8
[ 91380][V][Parsing.cpp:227] _parseRequest(): headerName: Referer
[ 91385][V][Parsing.cpp:228] _parseRequest(): headerValue: http://192.168.4.1/
[ 91392][V][Parsing.cpp:227] _parseRequest(): headerName: Accept-Encoding
[ 91399][V][Parsing.cpp:228] _parseRequest(): headerValue: gzip, deflate
[ 91406][V][Parsing.cpp:227] _parseRequest(): headerName: Accept-Language
[ 91412][V][Parsing.cpp:228] _parseRequest(): headerValue: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
[ 91420][V][Parsing.cpp:255] _parseArguments(): args: 
[ 91425][V][Parsing.cpp:238] _parseRequest(): Request: /favicon.ico
[ 91431][V][Parsing.cpp:239] _parseRequest():  Arguments: 
[ 91443][E][WebServer.cpp:635] _handleRequest(): request handler not found
```

Again a request to [192.168.4.1](http://192.168.4.1/) works, while [esp-ap](http://esp-ap/) doesn't.
This time the test was done using Google Chrome on a Windows 10 notebook, as well as firefox on an Android phone, tho that one was before the last restart so its not in this log.

me-no-dev, 2022-03-31:
I think you are mistaking DHCP hostname with DNS/mDNS/SSDP resolution. Check those libraries in order to resolve the IP by calling `http://esp-ap` on windows, or `http://esp-ap.local` on Linux and Mac

ToMe25, 2022-03-31:
Uh, what does the AP hostname do then?
Does it just set an internal variable used for nothing then?

TheRealDJ, 2022-03-31:
```
// global
#include <DNSServer.h>
DNSServer dnsServer;

// setup after AP start (so IP exists)
dnsServer.start(53, "esp-ap", WiFi.softAPIP());

// you can also put a * to resolve ALL DNS to this IP
dnsServer.start(53, "*", WiFi.softAPIP());
```

TheRealDJ, 2022-03-31:
So ... 
```
#include <WebServer.h>
#include <DNSServer.h>

const char *AP_SSID = "esp-ap";

IPAddress apIp(192, 168, 4, 1);
WebServer server;
DNSServer dnsServer;

void onAPStart(arduino_event_id_t event) {
	if (WiFi.softAPsetHostname(AP_SSID)) {
		Serial.println("Setting hostname");
	}
}

void setup() {
	Serial.begin(115200);
	delay(3000);
	WiFi.disconnect(1, 1);
	WiFi.softAPsetHostname(AP_SSID);
	WiFi.mode(WIFI_AP);
	WiFi.onEvent(onAPStart, ARDUINO_EVENT_WIFI_AP_START);

	Serial.println("Opening AP...");
	WiFi.softAPConfig(apIp, apIp, IPAddress(255, 255, 255, 0));
	WiFi.softAP(AP_SSID);

	server.on("/", HTTP_GET, []() {
		server.send(200, "text/plain", "Some Response");
	});

	server.begin();
	dnsServer.start(53, "esp-ap", WiFi.softAPIP());

	Serial.println("Server ready.");
}

void loop() {
	server.handleClient();
	delay(20);
}
```

me-no-dev, 2022-03-31:
> Uh, what does the AP hostname do then?
> Does it just set an internal variable used for nothing then?

it sets the interface host name. Name resolution is a whole another thing in networking.

ToMe25, 2022-03-31:
> it sets the interface host name. Name resolution is a whole another thing in networking.

And what is that interface hostname used for on the AP interface?
I understand now that apparently it isn't supposed to do any DNS resolution with that.
And thats fine, but I would like to understand what it does with the AP hostname instead before closing this.

ToMe25, 2022-03-31:
I just realized I really shouldn't keep this open since the issue was just me misunderstanding what exactly it was meant to do.
I will close this issue in a day, regardless of whether I figure out what it does with that hostname.

Also I apologize for my last comment sounding more aggressive then it was meant to :)

ToMe25, 2022-04-02:
Closing this since it seems like it was never supposed to do what I thought it should.
As I should have done months ago.

closed 2022-04-02 by ToMe25