espressif/arduino-esp32 #5247: esp32 c3 的hardwareSerial库不能用，串口不能收发

BG2CRW, 2021-06-05:
esp32 c3 的hardwareSerial库不能用，串口不能收发

BG2CRW, 2021-06-05:
代码如下：

HardwareSerial mySerial1(1);
void setup() {
  mySerial1.begin(115200, SERIAL_8N1, 5, 18); //rx:5,tx:18
}

void loop() {
	mySerial1.println("hello world");
	delay(1000);

}

lbernstone, 2021-06-05:
Yes.  HardwareSerial needs to be rewritten to use the new driver/hal.

VojtechBartoska, 2021-06-07:
@BG2CRW Please keep your issue in English here.

lengbuyu, 2021-06-09:
@VojtechBartoska 
He's mean is 
ESP32 C3 can't use HardwareSerial Library, the serial port cannot receive and send.

Code：
HardwareSerial mySerial1(1);
void setup() {
mySerial1.begin(115200, SERIAL_8N1, 5, 18); //rx:5,tx:18
}

void loop() {
mySerial1.println("hello world");
delay(1000);
}

Just help translate.

lybgo, 2021-08-03:
When can we solve this problem?
I have products that also need to use Serial1 communication

PR #5549 (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF
## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.
commit b1d072df (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF (#5549)

## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.
closed 2021-08-23 by me-no-dev