espressif/arduino-esp32 #5249: WebServer webpages will not run java script in browser

roblatour, 2021-06-06:
using
#include <WebServer.h>

and the following code;

`  String Template = "<html><head><title>test</title></head><body>Hello and goodbye<script type='text/javascript'>setTimeout(function ( ){self.close();}, 5000 );</script></body></html>";

  Template.replace("$", Text);

  server.send(200, "text/html", Template);`

the webpage is sent and opened in the browser on my w10 machine as expected, but the java/script does not get run.

The same code works fine with ESPAsyncWebServer - that is to say the webpage is sent and opened in my w10 browser + the script is run.  However, I can not use ESPAsyncWebServer for other reasons.

Hoping this can be fixed in WebServer.h

chegewara, 2021-06-06:
1) on esp32 side all works fine
2) scripts are running in browser, so even if there is problem it is in browser or script, esp32 just delivers scripts, not running it

roblatour, 2021-06-06:
Yes, that's what I would have thought too.  

However, when I  create a .html file and drag it to the browser it works.  Also, when I send the same HTML via ESPAsyncWebServer's send command, it works.  But when I send it via the WebServer's send command it does not work. 

If needed I can create a sketch that shows this.   

Also, when I send it via WebServer and open the HTML in the browser to examine it, the HTML all looks fine.

Very strange.

chegewara, 2021-06-06:
I would suggest to add some `console.log()` to js script and see in browser console.

lbernstone, 2021-06-07:
I didn't look closely at your code to make sure your template is working, but I know that WebServer can deliver javascript just fine.  I have an example at https://github.com/lbernstone/rrdtool_ESP32/blob/master/examples/basicDB/basicDB.ino

roblatour, 2021-06-07:
Thanks - just saw your post now, I will look at your example. 

However, I have been working on an example of where it is not working for me ... (below) ... 

Perhaps I am doing something wrong?

`#include <Arduino.h>
#include <WiFi.h>
#include <WebServer.h>
WebServer server(80);

const char * ssid = "****";
const char * password =  "****";

const String index_html = "<!DOCTYPE html><head><title>test</title></head><body><h1>test</h1><form name='DataForm' action='/get'><input name='StayOpen' type='submit' value='StayOpen'>&nbsp;&nbsp;<input name='SelfClose' type='submit' value='SelfClose'></form></body></html>";

void HandleNotFound() {
  server.send(404, "text/plain", "not found");
}

void handleDataIn() {

  String ResponsePage;
  bool StayOpenWasClicked = false;

  for (uint8_t i = 0; i < server.args(); i++) {
    if (server.argName(i) == "StayOpen") {
      StayOpenWasClicked = true;
      break;
    }
  }

  if (StayOpenWasClicked) 
     ResponsePage = "<html><head><title>test</title></head><body>StayOpen clicked, this page should stay open and will</body></html>";
   else 
     ResponsePage = "<html><head><title>test</title></head><body>SelfClose clicked, this page should self-close but will not<script type='text/javascript'>setTimeout(function ( ){self.close();}, 5000 );</script></body></html>";
 
  server.send(200, "text/html", ResponsePage.c_str());

}

void setup() {

  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("Waiting to connect");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  server.on("/", HTTP_GET, [index_html]() {
    server.sendHeader("Connection", "close");
    server.send(200, "text/html", index_html.c_str());
  });

  server.on("/get", handleDataIn);
  server.onNotFound(HandleNotFound);
  server.begin();

  Serial.println("");
  Serial.println("Enter the following in your browser:");
  Serial.println(WiFi.localIP());

}

void loop() {
  server.handleClient();
  delay(2);
}`

lbernstone, 2021-06-07:
index_html is a global variable.  You don't need to capture it in your lambda for "/", and doing so may very well end up with an empty string.  See my example.

roblatour, 2021-06-07:
Thank you.  

However, there is no problem with index_html - it presents just fine.

The problem is tied more closely to the handleDataIn() routine, where the response page gets sent to the browser in its entirety but that page's script does not get executed in the browser.

Also, just to confirm the script works fine.  If you put the contents of the response page in a separate file with a '.html' extension and drag it into your browser it works as expected - auto closing after five seconds  (see attached file example2.txt that would need to be renamed example2.html for testing).  This only to confirm it is not a problem with the script itself.

[example2.txt](https://github.com/espressif/arduino-esp32/files/6608661/example2.txt)

Running the example from my earlier post may provide a better appreciation of the issue.

Again, with thanks

roblatour, 2021-06-07:
I think this may be the problem:

https://stackoverflow.com/questions/25937212/window-close-doesnt-work-scripts-may-close-only-the-windows-that-were-opene

closed 2021-06-07 by roblatour