espressif/arduino-esp32 #5250: VFSFileImpl(): fopen failed (IDFGH-5357)

pnadar-mv, 2021-06-07:
- Development Kit:      [ESP32-DevKitC]
- Module or chip used:  [ESP32-WROOM-32D]
- Operating System:     [Windows]
- Using an IDE?: [Yes.Visual Studio Code]
- Power Supply:         [external 5V]

Esp32 gives error as [E][vfs_api.cpp:243] VFSFileImpl(): fopen(/spiffs/ConnectionInfo.inf) failed.
The code contains the data folder containing  empty file as ConnectionInfo.inf and is uploaded using Visual Studio Code Platformio.
But once the data is written to the file ConnectionInfo.inf, on reopening the file for the second time or so, I get the error as fopen(/spiffs/ConnectionInfo.inf) failed.
What could be the issue?

## Debug Logs

```
Writing Key/Value Information to file: /ConnectionInfo.inf
[E][vfs_api.cpp:243] VFSFileImpl(): fopen(/spiffs/ConnectionInfo.inf) failed
Failed to open SPIFFS File: /ConnectionInfo.inf
Connection Info Ret Val:
CANNOT_OPEN_FILE_IN_WRITE_MODE
Writing Key/Value Information to file: /DeviceInfo.inf
[E][vfs_api.cpp:243] VFSFileImpl(): fopen(/spiffs/DeviceInfo.inf) failed
Failed to open SPIFFS File: /DeviceInfo.inf
DeviceInfo Info Ret Val:
CANNOT_OPEN_FILE_IN_WRITE_MODE
```

igrr, 2021-06-07:
I have transferred this issue from esp-idf as the vfs_api.cpp mentioned here seems to be part of arduino-esp32. However there doesn't seem to be enough information for this issue to be actionable. @pnadar-mv I recommend you to create a small sketch which demonstrates this issue, and attach it or link it from the issue description.

pnadar-mv, 2021-06-07:
Thanks @igrr for acknowledging the issue here. I would love to elaborate the issue a bit. I have attached a small sketch below which  shows how the data is written into the file and retrieved from it. Also I have attached the output which shows fopen failed. This might not be the case every time, in fact the code works successfully in some of the Esp32 devices.

Some additional Info:
Using an IDE: Visual Studio Code
Extensions used in Visual Studio Code: Platformio IDE V1.10.0, C/C++ V1.4.0

The Firmware contains the data folder containing empty files as ConnectionInfo.inf, DeviceInfo.inf, AlertSettings.inf, DeviceModelSensorInfo.inf, CalibrationInfo.inf, CalibValuesInfo.inf and is uploaded using Visual Studio Code Platformio.
But once the data is written to the file ConnectionInfo.inf or in DeviceInfo.inf and even in the remaining .inf files, 
on rewriting the files for the second time or so, the Esp32 throws an error as fopen(/spiffs/ConnectionInfo.inf) failed.

Additional Firmware Logic:
Also the Esp32 has some files saved spontaneously in SPIFFS when the internet goes down and the files gets automatically deleted when the internet is available again.

The Partition used in ESP32-WROOM-32D is as attached below for your reference.
https://drive.google.com/file/d/1fvK7-xWMPKhxi8abKYsvl26Mveavukjl/view?usp=sharing
Also a code snippet is attached where data is written to a file and and retrieved from the file.
https://drive.google.com/file/d/1n7r_81J2WfHqahiW0Wnc6R1uV_OWe7fB/view?usp=sharing
The output too is attached below.
[Output.txt](https://github.com/espressif/arduino-esp32/files/6608234/Output.txt)

lbernstone, 2021-06-08:
I don't have an actual answer for you, but can tell you that spiffs has distinct limits.  I believe it only allows 256 files total.  There is no such thing as a directory in spiffs, just file names with "/" in them, and some code to simulate hierarchy based on those slashes.  The smallest file possible is a full sector, so even though you have only written 100 bytes, that file will take 4096 bytes on the flash.  LittleFS is a drop-in replacement for SPIFFS that should get around some of those limitations, and provide equivalent or better file integrity.

igrr, 2021-06-08:
> spiffs has distinct limits. I believe it only allows 256 files total.

@lbernstone I'm curious, do you have any reference to this?

lbernstone, 2021-06-08:
I don't.  I'll retract it since you know it to be otherwise :smile:

pnadar-mv, 2021-06-08:
Hi @lbernstone and @igrr.
Thanks for answering my query.
**Just curious, do we really have to upload 'empty files' in the data folder within the firmware?**
I did this because of the suggestion provided in the link below. Please refer the 5th point.
https://fasani.de/2018/11/24/platformio-an-alternative-to-arduino-ide-and-a-complete-ecosystem-for-iot/
It says that :-
5. FS Data. If you are using SPIFFS that in Arduino is the /data folder inside your sketch you will notice that in PlatformIO this folder needs to be at the root level. So you can simply move it one directory above. To upload SPIFFS data use the following command in the terminal:
pio run --target uploadfs
**IMPORTANT: As difference with Arduino if you try to save a file in /1.JPG here you will get it saved on:
/spiffs/1.JPG so create this folder on /data or you will get; VFSFileImpl(): fopen(/spiffs/1.jpg) failed**

pnadar-mv, 2021-06-10:
Hi @lbernstone @igrr .
Just referred a similar issue on the following link.
https://github.com/espressif/arduino-esp32/issues/1763

**Is the flash storage longevity about 10000 cycles for SPIFFS?** Also here it says that the 'file' needs one page for a header, and filesize/(pagesize - metadata) pages for data. MetaData is overhead that SPIFFs uses to organize the data, 12 to 32 bytes.

I actually used the function totalBytes() and it showed 896321 bytes for the above Partition Scheme used in my case and for **77** files each around 100 bytes stored in SPIFFS, the usedBytes() returned 67268 bytes.

closed 2021-09-06 by stale[bot]

AllanOricil, 2024-03-11:
Im facing a similar issue. Can someone help me. There is nothing on the internet about this error in that line

[image]

````c
void write_secret(byte* secret, unsigned int length){
  init_sd_card_reader();
  File file = SD.open("/test.txt", FILE_WRITE);
  if(!file){
    Serial.println("file not open");
  }else{
    for (int i = 0; i < length; i++) {
      file.print((char) secret[i]);
    }
    file.flush();
    file.close();
    Serial.println("New secret added");
  }
}
````

WallNutss, 2025-03-13:
Hey, if anyone still need this, I'll leave my answer right here. I also found this issue when tried to write some empty file to SD Card. Funny thing is, when I tried flash, the code using ArduinoIDE, everything works fine. But when I tried to flash the code using the PlatformIO development framework. This is my code

`
void SDCardModule::createEmptyJsonFileIfNotExists(const char* filePath) {
    ESP_LOGI(LOG_TAG, "Start creating Empty JSON File");

    // Check if the file exists, if not, create it
    if (!SD.exists(filePath)) {
      ESP_LOGI(LOG_TAG, "File does not exist, creating a new one.");

      File file = SD.open(filePath, FILE_WRITE);    
      if (file) {
        JsonDocument doc;
  
        // Serialize the empty JSON object to the file
        if (serializeJson(doc, file) == 0) {
          ESP_LOGI(LOG_TAG, "Failed to write empty JSON object to the file");
        } else {
          ESP_LOGI(LOG_TAG, "Empty JSON object written to file successfully");
        }

        file.close();
      } else {
        ESP_LOGI(LOG_TAG, "Failed to open file for writing");
      }
    }else{
        ESP_LOGI(LOG_TAG, "There's already JSON file of %s available!", filePath);
    }
}
`

Where as you can see in this log, it failed to write them
`
I (2516) SD_CARD: Start creating Empty JSON File
I (2522) SD_CARD: File does not exist, creating a new one.
[  2215][E][vfs_api.cpp:332] VFSFileImpl(): fopen(/sd/rfids.json) failed
I (2544) SD_CARD: Failed to open file for writing
`

 I thought there was some difference in the source code used by ArduinoIDE and PlatformIO, and I see nothing fishy in those two codes. I found that turn's out the support for FAT32 long filename system by default is turned off, and all I need to do is to enable this to true. This [blog](https://pcbartists.com/firmware/esp32-firmware/esp32-cannot-save-json-file-to-sd-card/) fix my issue right away. Hope this help someone