espressif/arduino-esp32 #5252: BluetoothSerial.end() kernel panic

olicooper, 2021-06-07:
### Hardware:
Board:							Lolin32 Lite
Core Installation version:			1.0.6
IDE name:						Platform.io
Flash Frequency:					default
PSRAM enabled:                                    NA
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:

I am getting a kernel panic when using `BluetoothSerial.end()` while a client is connected..

1. Start BT `SerialBT.begin("deviceName")` (`enableSerialBtMode()`)
2. Connect an Android 10 device using Android app `Serial Bluetooth Terminal`. The device is already paired so connection starts immediately.
3. Stop (disable) BT while android device is still connected `SerialBT.end()` (`disableSerialBtMode()`)
4. kernel panic!

### Sketch
My sketch is spread across many files with many classes so it is too complex to place here. Snippets have been extracted below..

- On device setup:
```cpp
void serialBtCb(esp_spp_cb_event_t event, esp_spp_cb_param_t *param) {
    if (event == ESP_SPP_SRV_OPEN_EVT) {        
        // log "Client connected"
    } else if (event == ESP_SPP_CLOSE_EVT) {
        // log "Client disconnected"
    }
}
void setup() {
    SerialBT.enableSSP();
    SerialBT.onConfirmRequest(confirmRequestCallback);
    SerialBT.onAuthComplete(authCompleteCallback);
    SerialBT.register_callback(&serialBtCb);
}
```

- Enabling serial BT mode:
```cpp
void enableSerialBtMode() {
    SerialBT.begin(BT_DEVICE_NAME);
}
```

- Disable serial BT mode:
```cpp

void disableSerialBtMode() {
    // If the user initiated a pairing request, reject the request
    if (confirmRequestPending) {
        confirmRequestPending = false;
        if (btStarted()) SerialBT.confirmReply(false);
    }
    SerialBT.end(); // KERNEL PANIC HERE
}
```

### Exception output:

The kernel seems to load a variable from an illegal address `EXCVADDR: 0x00000028` when calling the method `BluetoothSerial.end()`. It seems that it is trying to close the SSP connection but failing at `btc_spp_write`.

```
2021-06-07T17:08:29 [I][bt]: Switch BT mode: 2 -> 0
2021-06-07T17:08:29 [D][bt]: Client disconnected
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401022b5  PS      : 0x00060330  A0      : 0x80102a12  A1      : 0x3ffcf620  
A2      : 0x00000000  A3      : 0x3ffc1620  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060520  A7      : 0x00000000  A8      : 0x00000005  A9      : 0x3ffcf620  
A10     : 0x00000000  A11     : 0x00060523  A12     : 0x00060520  A13     : 0x3ffcf6a0
A14     : 0x070f0000  A15     : 0x3ffe5024  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000028  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x401022b5:0x3ffcf620 0x40102a0f:0x3ffcf670 0x400f79c5:0x3ffcf690 0x4009081a:0x3ffcf6c0
  #0  0x401022b5:0x3ffcf620 in btc_spp_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/spp/btc_spp.c:1305
  #1  0x40102a0f:0x3ffcf670 in btc_spp_call_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/spp/btc_spp.c:851
  #2  0x400f79c5:0x3ffcf690 in btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/btc/core/btc_task.c:160
  #3  0x4009081a:0x3ffcf6c0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

olicooper, 2021-06-08:
Manually disconnecting the client, then adding a delay before calling `end()` has allowed me to avoid the kernel panic (see below), but I don't think this is a very good fix because the `end()` code calls disconnect anyway so it should work without it.

```cpp
SerialBT.disconnect();
delay(200);
SerialBT.end();
```

I have noticed the same is happening for BLE too (kernel panic when a client is connected and `BLEDevice::deinit(false)` is called), but again disconnecting the client and then waiting with a delay fixes the issue.

Any suggestions? Is this a bug?

### Memory Usage
- I thought I'd also share the memory usage if it helps.. Calling `esp_get_free_heap_size`, `esp_get_free_internal_heap_size`, `esp_get_minimum_free_heap_size` just before disabling and after disabling BLE and SerialBluetooth.

| BT Mode | Before | After |
| --- | --- | --- |
| BLE | free=126212 free_internal=126212 minfree=125396 | free=220292 free_internal=220292 minfree=125396 |
| SerialBluetooth | free=115516 free_internal=115516 minfree=110756 | free=220236 free_internal=220236 minfree=110608 |

Cycling between enabled/disabled SerialBluetooth seems to have 56bytes of memory leak. The BLE library doesn't manage memory at all (I have to manually store pointers to all descriptors/characteristics/services etc. that I create so I can free them when I disable BLE - otherwise I get 5000 bytes of memory leak).

P-R-O-C-H-Y, 2021-11-11:
Hi @olicooper,
I have made some testing and you should upgrade the arduino-esp32 to latest supported version `2.0.1 (2.0.0)`.
There is no kernel panic happening when calling SerialBT.end() when connected devices were not disconnected before.

About the memory leaks I made some tests too.
```
void loop() {
  SerialBT.begin("ESP32test");
  reportMemoryUse("after begin begin"); //print memory usage
  SerialBT.end();
  delay(5000);
}
```

These are the outputs:
`version 1.0.6`
```
MEMORY USE: 283572 total, 206756 free, 113792 max.alloc - max.alloc internal 113792 - heap internal 4324084
MEMORY USE: 283556 total, 206700 free, 113792 max.alloc - max.alloc internal 113792 - heap internal 4324524
MEMORY USE: 283520 total, 206500 free, 113792 max.alloc - max.alloc internal 113792 - heap internal 4324104
MEMORY USE: 283504 total, 206448 free, 113792 max.alloc - max.alloc internal 113792 - heap internal 4324328
MEMORY USE: 283488 total, 206396 free, 113792 max.alloc - max.alloc internal 113792 - heap internal 4324000
MEMORY USE: 283468 total, 206340 free, 113792 max.alloc - max.alloc internal 113792 - heap internal 4324220
MEMORY USE: 283452 total, 206288 free, 113792 max.alloc - max.alloc internal 113792 - heap internal 4324168
MEMORY USE: 283436 total, 206236 free, 113792 max.alloc - max.alloc internal 113792 - heap internal 4323840
MEMORY USE: 283408 total, 206176 free, 113792 max.alloc - max.alloc internal 113792 - heap internal 4323780
MEMORY USE: 283392 total, 206124 free, 113792 max.alloc - max.alloc internal 113792 - heap internal 4323728
```
`version 2.0.1`
```
MEMORY USE: 272936 total, 195872 free, 65524 max.alloc - max.alloc internal 65524 - heap internal 4332311
MEMORY USE: 272936 total, 195872 free, 65524 max.alloc - max.alloc internal 65524 - heap internal 4332291
MEMORY USE: 272936 total, 195872 free, 65524 max.alloc - max.alloc internal 65524 - heap internal 4332311
MEMORY USE: 272936 total, 195872 free, 65524 max.alloc - max.alloc internal 65524 - heap internal 4332311
MEMORY USE: 272936 total, 195872 free, 65524 max.alloc - max.alloc internal 65524 - heap internal 4332311
MEMORY USE: 272936 total, 195872 free, 65524 max.alloc - max.alloc internal 65524 - heap internal 4332311
MEMORY USE: 272936 total, 195872 free, 65524 max.alloc - max.alloc internal 65524 - heap internal 4332119
MEMORY USE: 272936 total, 195872 free, 65524 max.alloc - max.alloc internal 65524 - heap internal 4332311
MEMORY USE: 272936 total, 195872 free, 65524 max.alloc - max.alloc internal 65524 - heap internal 4332311
MEMORY USE: 272936 total, 195872 free, 65524 max.alloc - max.alloc internal 65524 - heap internal 4332311
```
I can confirm that in version `1.0.6` there is a memory leak every time SerialBT.begin() and SerialBT.end() is called.
This issue is fixed in the version `2.0.1`.

closed 2021-11-11 by P-R-O-C-H-Y