espressif/arduino-esp32 #5255: platformio-build-esp32*.py scripts do not respect upload.extra_flags

maxgerhardt, 2021-06-08:
### Hardware:
Board:							Feather ESP32-S2 and related
Core Installation version:			        current git version, a618fc1361b08aa968407dd81b7b065cca207f46
IDE name:						PlatformIO
Flash Frequency:					80Mhz
PSRAM enabled:                                       no
Upload Speed:					       115200
Computer OS:                                         Windows 10

### Description:

Looking at the Arduino IDE description of the board

https://github.com/espressif/arduino-esp32/blob/a618fc1361b08aa968407dd81b7b065cca207f46/boards.txt#L3772-L3774

and the [PlatformIO board definition](https://github.com/platformio/platform-espressif32/blob/develop/boards/featheresp32-s2.json), I do not see anything in the [build script](https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build-esp32s2.py) that would implement the `extra_flags` with the TinyUF2 bootloader in PlatformIO.

I would have expected that the information of this "extra flash image" is encoded in e.g. the board JSON definition and in the builder script, which can then use `FLASH_EXTRA_IMAGES` to trigger the uploading of that binary file.

https://github.com/platformio/platform-espressif32/blob/4cb06337d678312dbcb293f508d7de7d7124ee68/builder/main.py#L318-L319

so e.g. in or after 

https://github.com/espressif/arduino-esp32/blob/a618fc1361b08aa968407dd81b7b065cca207f46/tools/platformio-build-esp32s2.py#L289-L293

I would have expected a check if the board needs extra flash images (currently in the `boards.txt`, 4 boards use extra flags), and then add these to the array, as e.g. with
```python
    ("0x2d0000", join(FRAMEWORK_DIR, "variants", env.BoardConfig().get("build.variant"), "tinyuf2.bin")
```
or somewhere between the lines. With that missing, these 4 boards are probably only working within the Arduino IDE right now. 

This would also resolve [this](https://community.platformio.org/t/board-definition-for-adafruit-magtag/20747) where people are asking how to use new board definitions with PlatformIO.

brentru, 2021-08-10:
Bumping this issue - having the same issue as @maxgerhardt

me-no-dev, 2021-08-11:
this issue is particular to PlatformIO. Please take the problem there :) This should be done in the PIO board config (or maybe added to the project)

cc: @ivankravets @valeros

maxgerhardt, 2021-08-11:
The current builder scripts already declare the extra images to flash, and the TinyUF2 bootloader is specific to Arduino-ESP32. In my view this *must* by handled in the `platformio-build-*.py`, not in platform-espressif32. The platform scripts directly call into the framework's script, expecting them to do the heay platform-specific lifting ([source](https://github.com/platformio/platform-espressif32/blob/develop/builder/frameworks/arduino.py))

me-no-dev, 2021-08-11:
the TinyUF2 bootloader is specific to particular boards. It is not used in general to be in `platformio-build-*.py`

VojtechBartoska, 2022-03-30:
Hello @maxgerhardt , I'm closing this issue as there is nothing to do on our side. Thanks for your contribution.

closed 2022-03-30 by VojtechBartoska

maxgerhardt, 2022-03-30:
Doing it in *only* platform-espressif32 scripts is way more complicated. The board definitions for the boards using the TinyUF2 bootloader should be expanded by an attribute (say, board.build.arduino.tinyuf2_bootloader = true) which can then be reacted to in the builder scripts **in this respository**. Otherwise code for some bootloaders is in the builder script here and some others in some weird post-fixup-things-not-done-in-arduino-esp32-script. Please rethink this design decision.

VojtechBartoska, 2022-03-30:
Thanks for fast reply, @me-no-dev can you please take a look in meantime on last comment and if it's applicable? Thanks!

VojtechBartoska, 2022-04-06:
Hello, at this moment we are not able to consider this, marking us won't fix for now.