espressif/arduino-esp32 #5258: ARDUINO_SERIAL_PORT=1 needed for (some) boards in v2.0.0

denschub, 2021-06-08:
### Hardware:
Board: Adafruit MagTag ESP32-S2, and probably a lot more
Core Installation version: 5.1.1
IDE name: Platform.io
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: macOS 11.4

### Description:

In https://github.com/espressif/arduino-esp32/commit/5502879a, a new flag `ARDUINO_SERIAL_PORT` was added. Looking at the following code:

denschub, 2021-08-21:
\*nudge\*

VojtechBartoska, 2022-03-30:
Hello @denschub , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? Serial was refactored lately. You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

denschub, 2022-04-08:
Hi @VojtechBartoska, please excuse me, I currently do not have hardware available that would allow me to test this. And since https://github.com/espressif/arduino-esp32/issues/5255#issuecomment-1090175300 was closed as wontfix, I probably won't have the chance to test that in an upcoming project anyway.

I think it's fair to close this for now, and if someone feels strongly, they will file a new bug or make noise here. :)