espressif/arduino-esp32 #5260: Can't compile latest master with platformio

xtrinch, 2021-06-08:
Hello,

trying to compile latest master with platformio.

```
[env:esp32s2doit-devkit-v1]
platform = espressif32
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
framework = arduino
```

No problems using the alpha release, the latest master however gives me this compilation error:

```
Indexing .pio/build/esp32s2doit-devkit-v1/libFrameworkArduino.a
Linking .pio/build/esp32s2doit-devkit-v1/firmware.elf
Retrieving maximum program size .pio/build/esp32s2doit-devkit-v1/firmware.elf
Checking size .pio/build/esp32s2doit-devkit-v1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.5% (used 37808 bytes from 327680 bytes)
Flash: [=======   ]  74.4% (used 975330 bytes from 1310720 bytes)
Building .pio/build/esp32s2doit-devkit-v1/firmware.bin
esptool.py v3.0

A fatal error occurred: Segment loaded at 0x3f000120 lands in same 64KB flash mapping as segment loaded at 0x3f000020. Can't generate binary. Suggest changing linker script or ELF to merge sections.
*** [.pio/build/esp32s2doit-devkit-v1/firmware.bin] Error 2
```

Found some other people on the internet having the same problem: https://community.platformio.org/t/esp32-arduino-platformio-support-for-2-0-0-version/21745/2

Is there anything I can update or configure to make this work?

Jason2866, 2021-06-12:
You have to update esptool.py to version 3.1

xtrinch, 2021-06-12:
Nice, that does it. Thanks :)

VojtechBartoska, 2021-06-17:
@xtrinch Can I consider this issue closed?

xtrinch, 2021-06-17:
Yes certainly, closing