espressif/arduino-esp32 #5262: WiFi.begin() stops trying to connect

am-canes, 2021-06-09:
Board: ESP32 Dev Module
Core Installation version: 1.0.5
IDE name: Platform.io
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Windows 10

Using WiFi.begin() will esp32 try to connect indefinitely?
Is there any reason the code could stop trying to connect to wifi? Maybe a timeout?

Sometimes it seems that my code stop trying to connect and I don't know the reason. 
I need a way to detect this case. Is it possible?

chansheunglong, 2021-06-10:
Adding `WiFi.reconnect()` in your code when you are looping and waiting for a connection to be established. Not sure if there is any timeout but I have been using this line to ensure my device is trying to connect to the network.

jellewie, 2021-06-11:
Just look for the definition, and (try) to figure it out yourself?
WiFi:begin() isn't declared in the ESP32 stuff, so it must already be declared before by Arduino, so their we find
[arduino-libraries/WiFi/blob/master/src/WiFi.cpp#L59](https://github.com/arduino-libraries/WiFi/blob/master/src/WiFi.cpp#L59)    [#L79](https://github.com/arduino-libraries/WiFi/blob/master/src/WiFi.cpp#L79)   [#L98](https://github.com/arduino-libraries/WiFi/blob/master/src/WiFi.cpp#L98)

Which includes a 'WL_MAX_ATTEMPT_CONNECTION' what would return 'WL_CONNECT_FAILED'

So if you would have simply did
`Serial.println(WiFi.begin())`
you would also have spotted this

And as always, Please mark this bug report closed if al your questions have been answered.

lbernstone, 2021-06-11:
> WiFi:begin() isn't declared in the ESP32 stuff, so it must already be declared before by Arduino, so their we find
??
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.h#L42

jellewie, 2021-06-11:
> > WiFi:begin() isn't declared in the ESP32 stuff, so it must already be declared before by Arduino, so their we find
> > ??
> > https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.h#L42

It is weird that my text-editor search tool didn't catch [that](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.cpp#L148). Thanks for the heads up!

but `Serial.println(WiFi.begin())` could still have been used to print the 'status()'

am-canes, 2021-06-11:
Using WiFiClientEvents I found that the problem occurs when wifi is disconnected and tries to connect again and it is disconnected with reason AUTH_LEAVE.
In this case wifi stops trying to connect. 
Now I am doing WiFi.reconnect() when this condition is detected.

Thanks.

closed 2021-06-11 by am-canes