espressif/arduino-esp32 #5270: ESP-NOW - facing problems to exchange data

mggil7, 2021-06-09:
Hello, I am facing problems to use espnow and Exchange data between two ESP8266. 
Already tried with two ESP32 and had the same problem, which is I am able to send data ( a text message) but it fails to be delivered. 
Probably it is a silly detail but I am stuck and do not know what to do next. Find below master and slave codes. 
Any help or advice would be greatly appreciated.
MASTER:
#include <ESP8266WiFi.h>
extern "C" {
#include <espnow.h>
}
// This is the slave MAC Address which receives the data
// uint8_t mac[] = {0x48, 0x3F, 0xDA, 0x7E, 0x30, 0xF8}; //AP MAC MASTER'S ADDRESS
uint8_t mac[] = {0x48, 0x3F, 0xDA, 0x7E, 0x5A, 0xCA}; //AP MAC SLAVE'S ADDRESS

  #define WIFI_CHANNEL 4
  int prevstate_1 = LOW;
  int prevstate_2 = LOW;

// Data structure, must be the same for the slave

struct __attribute__((packed))DataStruct {
  char text[32];
};
    DataStruct button_1;
    DataStruct button_2;
    
// Callback when data is sent
void OnDataSent(uint8_t *mac, uint8_t sendStatus) {
  Serial.print("Last Packet Send Status: ");
  if (sendStatus == 0){
    Serial.println("Delivery success");
  }
  else{
    Serial.println("Delivery fail");
  }
}

//=====================================================================================================
void setup() {
  pinMode(D1, INPUT_PULLUP);
  pinMode(D4, INPUT_PULLUP);
  pinMode(D2, OUTPUT);
  pinMode(D3, OUTPUT);

  Serial.begin(115200); Serial.println();
  Serial.println("Starting EspnowController.ino");
  WiFi.mode(WIFI_STA); // Station mode for esp-now controller
  WiFi.disconnect();
  Serial.printf("This mac: %s, ", WiFi.macAddress().c_str());
  Serial.printf("slave mac: %02X%02X%02X%02X%02X%02X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
  Serial.printf(", channel: %i\n", WIFI_CHANNEL);
  if (esp_now_init() != 0)
  {
    Serial.println("*** ESP_Now initialization failed");
  }
  esp_now_set_self_role(ESP_NOW_ROLE_CONTROLLER);
  esp_now_add_peer(mac, ESP_NOW_ROLE_SLAVE, WIFI_CHANNEL, NULL, 0);
  strcpy(button_1.text, "Button 01 pressed");
  strcpy(button_2.text, "Button 02 pressed");
  Serial.println("Setup finished");
    // Once ESPNow is successfully Init, we will register for Send CB to
  // get the status of Trasnmitted packet
  esp_now_register_send_cb(OnDataSent);
}

//======================================================================================================

void loop() {
  sendData();
}

//======================================================================================================
void sendData() {

  int currentstate_1 = digitalRead(D4);
  if (prevstate_1 != currentstate_1) {
    if (currentstate_1 == LOW) {
      uint8_t bs[sizeof(button_1)];
      memcpy(bs, &button_1, sizeof(button_1));
      esp_now_send(mac, bs, sizeof(button_1));
      Serial.println(button_1.text);
      digitalWrite(D2, !digitalRead(D2));
    }
  } prevstate_1 = currentstate_1;

  int currentstate_2 = digitalRead(D1);
  if (prevstate_2 != currentstate_2) {
    if (currentstate_2 == LOW) {
      uint8_t bs[sizeof(button_2)];
      memcpy(bs, &button_2, sizeof(button_2));
      esp_now_send(mac, bs, sizeof(button_2));
      Serial.println(button_2.text);
      digitalWrite(D3, !digitalRead(D3));
    }
  } prevstate_2 = currentstate_2;
}
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
SLAVE

#include <ESP8266WiFi.h>
extern "C" {
#include <espnow.h>
#include <user_interface.h>
}
 uint8_t mac[] = {0x48, 0x3F, 0xDA, 0x7E, 0x30, 0xF8}; //AP MAC SLAVE'S ADDRESS

//============================================================================================

int Led1 = D0;
int Led2 = D1;
  
void initVariant() {
  WiFi.mode(WIFI_AP);
  wifi_set_macaddr(SOFTAP_IF, &mac[0]);
}

//=============================================================================================
#define WIFI_CHANNEL 4

// must match the controller struct
struct __attribute__((packed))DataStruct { 
  char text[32];
  unsigned int time;};

DataStruct receivedData;

//=============================================================================================
void setup() {
  Serial.begin(115200); Serial.println();
  Serial.println("Starting EspnowSlave.ino");
  
  Serial.print("This node AP mac: "); Serial.println(WiFi.softAPmacAddress());
  Serial.print("This node STA mac: "); Serial.println(WiFi.macAddress());
  Serial.printf(", channel: %i\n", WIFI_CHANNEL);
  pinMode(Led1,OUTPUT);
  pinMode(Led2,OUTPUT);

  if (esp_now_init() != 0) {
    Serial.println("*** ESP_Now init failed");
    while (true) {};
  }

  esp_now_set_self_role(ESP_NOW_ROLE_SLAVE);
  esp_now_register_recv_cb(receiveCallBackFunction);
  Serial.println("End of setup - waiting for messages");
}

//==============================================================================================

void loop() {
}

//==============================================================================================
//void receiveCallBackFunction(uint8_t *senderMac, uint8_t *incomingData, uint8_t len) {
  void receiveCallBackFunction(uint8_t *mac, uint8_t *incomingData, uint8_t len) {

    Serial.print("INCOMING=");
      Serial.println(*incomingData);

       Serial.print("len=");
      Serial.println(len);
  memcpy(&receivedData, incomingData, sizeof(receivedData));

    //  Serial.print("INCOMING_AFTER=");
      //Serial.println(incomingData);

       Serial.print("len_AFTER=");
      Serial.println(receivedData.text);

  String DataCompare = String(receivedData.text);

  if(DataCompare == "Button 01 pressed"){
  digitalWrite(Led1, !digitalRead(Led1));
  Serial.println(" Message = " + DataCompare);
 }
  if(DataCompare == "Button 02 pressed"){
  digitalWrite(Led2, !digitalRead(Led2));
  Serial.println(" Message = " + DataCompare);
   
  }

VojtechBartoska, 2021-06-15:
Thank you for reporting this issue @mggil7.

Did you try this example?
https://github.com/espressif/esp-idf/tree/master/examples/wifi/espnow

We are working on this issue internally.

VojtechBartoska, 2021-06-17:
https://jira.espressif.com:8443/browse/WIFI-3841

TD-er, 2021-08-21:
Just to remove the stale label, as I'm also curious about what the fix will be.

I only can send data via ESP-now between ESP32 and ESP8266 when using (PlatformIO version) espressif32@2.1.0
Later versions do not allow to send data.

VojtechBartoska, 2021-12-14:
@TD-er and @mggil7 Do you have any updates up to the latest comment?

closed 2021-12-14 by VojtechBartoska

aidarishkulov, 2022-02-22:
How to measure WiFi signal strength ESP_NOW?

zhanzhaocheng, 2022-02-24:
ESP-NOW does not provide an interface to obtain RSSI directly, but you can refer to [this method](https://github.com/espressif/esp-now/blob/master/components/espnow/src/espnow.c#L132-L140) to obtain RSSI