espressif/arduino-esp32 #5271: Migrate to ArduinoCore-API

dok-net, 2021-06-10:
Migrating / porting Arduino Core for ESP8266 to use https://github.com/arduino/ArduinoCore-API should bring closure to debates about Arduino compatibility. 

According to the README there, at the time of entering this issue, in excerpts:
```
All Arduino official cores are being ported to the new structure so they take advantage of this single repo.

[...]
Some cores have been ported to the new structure, for example:

megaAVR (https://github.com/arduino/ArduinoCore-megaAVR)
nRF52-mbedos (https://github.com/arduino/ArduinoCore-nRF528x-mbedos)
classic AVR (https://github.com/arduino/ArduinoCore-avr/tree/api)
SAMD (https://github.com/arduino/ArduinoCore-samd/tree/api)
```

me-no-dev, 2021-06-10:
this is planned

dok-net, 2021-06-10:
I expected no less, just felt adding adding an issue to keep track is a good idea :-) Thanks!!!

igrr, 2021-06-15:
Linking the discussion on the same topic in esp8266/Arduino: https://github.com/esp8266/Arduino/issues/8112.
and quoting dok-net:

> Makes it look more and more like it's best to let matters settle before adopting that.

igrr, 2022-04-05:
Seems like the conclusion for now is that migrating to arduino-core API is not currently doable. We'll close this issue to keep the backlog manageable.

closed 2022-04-05 by igrr