espressif/arduino-esp32 #5273: ESP323-C3 UART communication

arijitdas123student, 2021-06-10:
### Hardware:
Board:							ESP32C3 Dev Module
Core Installation version:			2.0.0-alpha1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    No
Upload Speed:				       115200
Computer OS:                                        Ubuntu 20.04

### Description:
Hi there, 
I'm new to UART and would like to ask for some help from receiving accelerometer data from an Arduino Nano 33 BLE Sense to an ESP32-C3 DevKitM-1  via UART.
Can somebody please help me out ? 

Thanks

lbernstone, 2021-06-10:
There is currently an issue with uart1 on the C3. #5247

arijitdas123student, 2021-06-10:
Thanks for sharing this @lbernstone!

chansheunglong, 2021-06-10:
Does that mean currently with Version 2.0.0-alpha1 Uart1 is not usable? Until the HardwareSerial library was updated

lbernstone, 2021-06-10:
Yes, experimental hardware support is experimental :smile:

arijitdas123student, 2021-06-11:
@lbernstone By any means have you guys tested the ADXL accelerometer sensors with the ESP32C3 DevKitM1 ?

arijitdas123student, 2021-06-11:
What I can see is that the default ADXL example script does compile for ESP32C3 from Arduino IDE....but since I presently don't have the sensor with me I wanted to check if anyone from you might have got to communicate with the ADXL sensors for ESP32C3 ?

Jason2866, 2021-06-17:
See #5287

lybgo, 2021-08-03:
@lbernstone 
When can we solve this problem?
I have products that also need to use Serial1 communication

chansheunglong, 2021-08-06:
> @lbernstone
> When can we solve this problem?
> I have products that also need to use Serial1 communication

Just wait
If is product go with ESP8266/ESP32
C3 is too new to be in a product

SuGlider, 2021-08-17:
I have created a PR to solve this issues, as well as other Arduino UART issues.
I confirm that the PR solves C3 Serial1 issue.

If you want to help testing it and then send commentaries, please use
https://github.com/SuGlider/arduino-esp32/tree/uart_refactoring
PR #5549

closed 2021-08-23 by me-no-dev

QGB, 2022-05-16:
```

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403814ee
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4ab8657f6f5aa80b81c7dd649db3bd64a98999ed7c65acc5a2ab77ce80caf525
Expected: 6babbc7b9ea63e04e0fc89ec224f6268aa3f84392018f73ab33721f768ce2eb6
Attempting to boot anyway...
entry 0x403ce000
Hello world!
E (160) uart: uart_set_pin(673): tx_io_num error
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40384a83
file: "C:\Users\qgb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32\esp32-hal-uart.c" line 168
func: uartBegin
expression: uart_set_pin(uart_nr, txPin, rxPin, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE)

abort() was called at PC 0x40384a87 on core 0
Core  0 register dump:
MEPC    : 0x403818d0  RA      : 0x40384a92  SP      : 0x3fc93260  GP      : 0x3fc8b400
TP      : 0x3fc89ad8  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x3fc9328c  S1      : 0x3fc9328c  A0      : 0x3fc93298  A1      : 0x3fc9327a
A2      : 0x00000000  A3      : 0x3fc932c5  A4      : 0x00000001  A5      : 0x3fc8d000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000001  S3      : 0x3fc8adb0
S4      : 0x0000000c  S5      : 0x00000000  S6      : 0x0000001c  S7      : 0x0000001b
S8      : 0x00000100  S9      : 0x00000000  S10     : 0x3fc8adb0  S11     : 0x0001c200
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc93260: 0x3fc8adb0 0x00000000 0x3fc93278 0x40389656 0x0000001c 0x00000000 0x00000030 0x3fc8bb14
3fc93280: 0x3fc9328c 0x3fc8bb30 0x3fc93278 0x38333034 0x37386134 0x40384a00 0x726f6261 0x20292874
3fc932a0: 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x34383330 0x20373861 0x63206e6f
3fc932c0: 0x2065726f 0x00000030 0x00000000 0x07f0d48f 0x3c030a58 0x3fc8ada4 0x00000001 0x40384a8a
3fc932e0: 0x00000001 0x3fc8ada4 0x00000001 0x42001bb6 0x3c0305f4 0x0001c200 0x00000003 0x00000000
3fc93300: 0x00000001 0x00000000 0x3fc8e170 0x00000000 0x00000000 0xffc1ffff 0x00000000 0x00000000
3fc93320: 0x00000000 0x00000000 0x00004e20 0x00000001 0x00000070 0x00000000 0x0800001c 0x0001c200
3fc93340: 0x0000001c 0x0000001b 0x3fc8c1fc 0x42000cc6 0x00000070 0x00000000 0x00000000 0x00000000
3fc93360: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93380: 0x00000000 0x0001c000 0x3fc8c000 0x420000d0 0x00000000 0x00000000 0x00000000 0x00000000
3fc933a0: 0x00000000 0x3fc8d000 0x42001000 0x42001512 0x00000000 0x00000000 0x00000000 0x40386fe8
3fc933c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc933e0: 0x27146414 0x01010107 0x00000001 0x00000000 0x00000000 0x00000000 0xffffffff 0x7fefffff
3fc93400: 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3fe00000 0x00000000 0x3ff80000
3fc93420: 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000 0x3ff00000
3fc93440: 0x00000000 0x40240000 0x00000000 0x401c0000 0x00000000 0x40140000 0x00000000 0x43500000
3fc93460: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fc93180 0x3fc8d264 0x3fc8cbec
3fc93480: 0x3fc8cbec 0x3fc93474 0x3fc8cbe4 0x00000018 0x1b565a63 0xc25c9055 0x3fc93474 0x00000000
3fc934a0: 0x00000001 0x3fc91470 0x706f6f6c 0x6b736154 0xa3f86800 0x00e112d0 0x00000000 0x3fc93460
3fc934c0: 0x00000001 0x00000002 0x00000000 0x00000000 0x00000000 0x3fc8dbfc 0x3fc8dc64 0x3fc8dccc
3fc934e0: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x420144fa 0x00000000
3fc93500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93520: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93540: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93560: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc935a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc935c0: 0x00000000 0x3f000000 0x00000080 0x31282045 0x20293036 0x74726175 0x6175203a 0x735f7472
3fc935e0: 0x705f7465 0x36286e69 0x3a293337 0x5f787420 0x6e5f6f69 0x65206d75 0x726f7272 0x013cf90a
3fc93600: 0xee00baf5 0x34449a9a 0xb595e154 0x211089ba 0x66675bd0 0x8ae67da7 0x006fdd63 0x010cc346
[... 48 lines cut ...]
3fc93340: 0x0000001c 0x0000001b 0x3fc8c1fc 0x42000cc6 0x00000070 0x00000000 0x00000000 0x00000000
3fc93360: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93380: 0x00000000 0x0001c000 0x3fc8c000 0x420000d0 0x00000000 0x00000000 0x00000000 0x00000000
3fc933a0: 0x00000000 0x3fc8d000 0x42001000 0x42001512 0x00000000 0x00000000 0x00000000 0x40386fe8
3fc933c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc933e0: 0x27146414 0x01010107 0x00000001 0x00000000 0x00000000 0x00000000 0xffffffff 0x7fefffff
3fc93400: 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3fe00000 0x00000000 0x3ff80000
[... cut ...]
3fc93560: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc93580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc935a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc935c0: 0x00000000 0x3f000000 0x00000080 0x31282045 0x20293036 0x74726175 0x6175203a 0x735f7472
3fc935e0: 0x705f7465 0x36286e69 0x3a293337 0x5f787420 0x6e5f6f69 0x65206d75 0x726f7272 0x013cf90a
3fc93600: 0xee00baf5 0x34449a9a 0xb595e154 0x211089ba 0x66675bd0 0x8a

```

lbernstone, 2022-05-16:
Please don't hijack issues. This issue was for a previous version. Open a new issue and follow the posting instructions.

PR #5549 (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF
## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.
commit b1d072df (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF (#5549)

## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.