espressif/arduino-esp32 #5275: Building as ESP-IDF Component: fatal error: soc/soc_caps.h: No such file or directory

Rajssss, 2021-06-10:
### Hardware:
Board:							ESP32 WROVER B DevKitC
Core Installation version:			1.0.6 (master)
IDE name:						IDF component
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
Describe your problem here
Hello, I am trying to use arduino-core as IDF Component with blink example from esp-idf v3.3.5. 
I have followed this guide: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md
I have just called digitalWrite in my code.
But while building, `make all` gave the following error:

`C:/Users/Rajssss/Documents/Microcontroller/ESP/tools/msys32/home/Rajssss/esp/esp-eclipse-manual-work/esp_arduino_core_idf_test/components/arduino/cores/esp32/esp32-hal-gpio.h:28:26: fatal error: soc/soc_caps.h: No such file or directory``

This is strange but, I have checked the idf/components/soc & arduino/tools/sdk/esp32/include/soc  but its not there.
Please help me out. Thanks.

chegewara, 2021-06-10:
> Core Installation version: 1.0.6 (master)

v1.0.6 is not the same as master. Master branch is now esp-idf v4.3+.

closed 2021-06-10 by Rajssss

hobzcalvin, 2021-12-10:
I'm also having this error, trying to build an existing project with v3.3.4. Can someone explain what might be going on here? I can't figure it out from the above. Thanks!