espressif/arduino-esp32 #5276: ESP32 ESP-WHO IDF with Arduino component can't build software

RickvF, 2021-06-10:
### Hardware:
Board:						ESP-EYE DevKit for AIoT Developers
IDE name:				         ESP-IDF, ESP-WHO, Arduino Esp32 component
Upload Speed:					115200
Computer OS:                                  Windows 10 (Virtual box with Ubuntu 20.04.2.0 Desktop AMD64)

### Description:
I have an ESP-EYE DevKit for AIoT Developers and wanted to develop my own program for it based on the examples/esp32/camera/CameraWebServer example as was available in the Arduino IDE. I developed the program with the Arduino IDE. The program includes HardwareSerial in order to communicate via the DOUT and DIN pin to another ESP-32. Also, I made some extensions to the webserver to support multiple web pages. Most of the source code is below.

This all is working pretty well via the Arduino IDE on Windows 10. However, when I connect with some browsers to the webserver running on my ESP-EYE, the Camera and options bar cannot be displayed due to an HTTP header issue (`Header field are too long for the server to interpret`). This HTTP header issue can be solved by changing the header length in the sdkconfig file (`Max HTTP Request Header Length` and `Max HTTP URI Length`). When I tried to do this in my Arduino installation (on various locations over the file system within Windows 10), did not improve anything.

So I got the suggestion to try to build the environment of ESP-IDF in Ubuntu. For this, I created a virtual box with Ubuntu 20.04.2.0-desktop-amd64 version.

Next, I followed this tutorial: https://gist.github.com/abobija/2721427f18d78e439fa9875cac960ffa. With this, I was able to run the Hello World example on my ESP-EYE. 
Then I followed this tutorial to include the Arduino component: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md. I also managed to run the hello world example.

In order to run the CameraWebServer example, instead of the ESP-IDF from the tutorial, I learned I needed the ESP-WHO environment, with the camera_web_server example as base.
So followed all those steps:
```
The steps to build esp-who SDK on Ubuntu are as follows:

Step 1. Install Prerequisites ：
sudo apt-get install git wget libncurses-dev flex bison gperf python python-pip python-setuptools python-serial 
python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache libffi-dev libssl-dev

Step 2. Get ESP-WHO:
mkdir esp-who
cd ~/esp-who
git clone --recursive https://github.com/espressif/esp-who.git
cd esp-who
git submodule update -f --init --recursive  
git pull

Step 3. Set up the esp-idf environment variables under esp-who:
cd ~/esp-who/esp-idf
pwd
export IDF_PATH=~/esp-who/esp-idf

Step 4. Set up the tools 
cd ~/esp-who/esp-idf
./install.sh
. ./export.sh

Step 5. Start a Project：
cd ..
cd ~/esp-who/examples/single_chip/camera_web_server
idf.py build
idf.py flash monitor
```

Afterwards, I was able to successfully run the camera_web_server example on my ESP-EYE.

Again I followed this tutorial: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md to include the Arduino component to the camera_web_server example folder, and I modified the program files in the camera_web_server example folder to my developed program. 

Files I had in the camera_web_server/main folder after copying my own program to it:
- app_httpd.cpp
- main.cpp
- camera_index.h
- camera_pins.h
- communication.h
- pages.h
- include (this was a folder)
- www (this was a folder)
- CMakeLists.txt (with modified COMPONENT_SRCS to: COMPONENT_SRCS "main.cpp")
- component.mk
- Kconfig.projbuild

Run `make menuconfig` and modified the HTTP request and HTTP URi Header length. Set in the Arduino configuration the flash size to Hugh App.

When I then run `idf.py build` this results in the error:
[... 930 lines cut ...]
        httpd_register_uri_handler(camera_httpd, &decline_uri);
        httpd_register_uri_handler(camera_httpd, &calling_uri);
    }

    config.server_port += 1;
    config.ctrl_port += 1;
//    Serial.printf("Starting stream server on port: '%d'\n", config.server_port);
    if (httpd_start(&stream_httpd, &config) == ESP_OK) {
        httpd_register_uri_handler(stream_httpd, &stream_uri);
    }
}
```

communication.h
```
#ifndef Communication_h
#define Communication_h

#include <HardwareSerial.h>

extern String message;
extern HardwareSerial Myserial;

#endif
```

me-no-dev, 2021-06-15:
What IDF version are you using?
Do not mix `make menuconfig` with `idf.py build`! Use `idf.py menuconfig` instead.

RickvF, 2021-06-15:
@me-no-dev I just used the master branch of https://github.com/espressif/esp-idf

As shown for the examples with the esp-who in combination with the esp32-arduino component I executed both the idf.py build and menu flash monitor commands and got different results.

But are you suggesting the esp-who with esp32-arduino component should work and I should follow the idf.py menuconfig, idf.py build and idf.py flash monitor commands?

lbernstone, 2021-06-15:
Does esp-who compile on its own with esp-idf master?  These components may not compile on every version of esp-idf.  You typically need to match the version of arduino-esp32 with a specific version of idf (https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/config/sdkconfig.h#L661).  Currently, arduino-esp32 master compiles on esp-idf master.  Earlier versions of arduino are unlikely to work.  If you then want to put esp-who into the mix, you need to be sure it compiles separately on the version of esp-idf you are using.

RickvF, 2021-06-15:
> 
> 
> Does esp-who compile on its own with esp-idf master? These components may not compile on every version of esp-idf. You typically need to match the version of arduino-esp32 with a specific version of idf (https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/config/sdkconfig.h#L661). Currently, arduino-esp32 master compiles on esp-idf master. Earlier versions of arduino are unlikely to work with idf master. If you then want to put esp-who into the mix, you need to be sure it compiles separately on the version of esp-idf you are using.

For the ESP-who I used following tutorial provided by Espressif. The ESP-IDF was part of the ESP-WHO folder after cloning. I did not do anything extra to get the ESP-IDF. As seen also here the master branch is used for ESP-WHO

he steps to build esp-who SDK on Ubuntu are as follows:

Step 1. Install Prerequisites ：
sudo apt-get install git wget libncurses-dev flex bison gperf python python-pip python-setuptools python-serial 
python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache libffi-dev libssl-dev

Step 2. Get ESP-WHO:
mkdir esp-who
cd ~/esp-who
git clone --recursive https://github.com/espressif/esp-who.git
cd esp-who
git submodule update -f --init --recursive  
git pull

Step 3. Set up the esp-idf environment variables under esp-who:
cd ~/esp-who/esp-idf
pwd
export IDF_PATH=~/esp-who/esp-idf

Step 4. Set up the tools 
cd ~/esp-who/esp-idf
./install.sh
. ./export.sh

Step 5. Start a Project：
cd ..
cd ~/esp-who/examples/single_chip/camera_web_server
idf.py build
idf.py flash monitor

lbernstone, 2021-06-15:
esp-who expects idf4.0.  arduino-esp32 expects idf4.4(ish).  It will not be possible to integrate this without a significant effort on your part.  It might be possible to get this working with the 4.0 release container at https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder.

RickvF, 2021-06-15:
Thanks for the tips.
For my understanding, this will allow me to modify the HTTP headers correctly and then use the Arduino IDE to create the rest of the project, flash it and so on? So I don't need the esp-idf anymore

RickvF, 2021-06-15:
And when I want to use the ESP-IDF I have to come up with a different solution for the setup and loop function of my main file and a different solution of the HardwareSerial I am currently using?

lbernstone, 2021-06-15:
the docker container includes the esp-idf at a specific point in time (in this case idf4.0 about 8 months ago) along with the arduino that compiled then.  You can clone esp-who into the container and build your project there, or else build the arduino libraries there and copy them out into your normal environment.  I can provide no guarantees that any of this will work, since you are trying to mix and match pieces.  I don't know where your HTTP header issues are coming from. If that is your actual problem, it would probably be more direct to solve that problem, instead of going down a very long rabbit hole.

RickvF, 2021-06-15:
That is indeed the initial issue I am trying to solve, which took me already a long road down the rabbit hole. But I will try every possible solution.

me-no-dev, 2021-06-15:
generally you do not need the full esp-who. All you need is esp-face :) https://github.com/espressif/esp-dl

ESP-IDF master should compile fine with arduino master, esp32-camera master and esp-dl master.

RickvF, 2021-06-16:
That sounds as an easier solution.

So I have to install the ESP-IDF and add in the components folder of the ESP-IDF installation the following components:

- https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md
- https://github.com/espressif/esp32-camera
- https://github.com/espressif/esp-dl

Then I can take the hello world example and edit it with my sketch based on the examples/esp32/camera/CameraWebServer from the Arduino IDE?

RickvF, 2021-06-16:
Got a bit further already with that approach.

main.cpp:115: undefined reference to `startCameraServer()'

made a step and now got this reference you have to check why arduino accepts the reference but esp-idf not

RickvF, 2021-06-16:
It is still missing some files which are not part of the esp-dl

../main/app_httpd.cpp:22:10: fatal error: fb_gfx.h: No such file or directory
 #include "fb_gfx.h"
          ^~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

me-no-dev, 2021-06-17:
ahh :) fb_gfx component you need to copy from ESP-WHO. It is used to draw the frames around faces

RickvF, 2021-06-17:
Thanks for your help! That got me further. `idf.py build` went successful.

Only got a problem with `idf.py flash monitor` it is resetting constantly with:

```
Wrote 826272 bytes (515765 compressed) at 0x00010000 in 12.5 seconds (effective 528.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 105...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (105 compressed) at 0x00008000 in 0.1 seconds (effective 441.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Running idf_monitor in directory /home/rick/Desktop/hello_world
Executing "/home/rick/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/rick/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /home/rick/Desktop/hello_world/build/hello-world.elf -m '/home/rick/.espressif/python_env/idf4.4_py3.8_env/bin/python' '/home/rick/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

st:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6744
load:0x40078000,len:14236
ho 0 tail 12 room 4
load:0x40080400,len:3716
0x40080400: _init at ??:?

entry 0x40080680
I (28) boot: ESP-IDF v4.4-dev-1594-g1d7068e4b 2nd stage bootloader
I (28) boot: compile time 20:04:55
I (28) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 8MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00177000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=22478h (140408) map
I (163) esp_image: segment 1: paddr=000324a0 vaddr=3ffb0000 size=04c84h ( 19588) load
I (171) esp_image: segment 2: paddr=0003712c vaddr=40080000 size=08eech ( 36588) load
I (186) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=8c13ch (573756) map
I (394) esp_image: segment 4: paddr=000cc164 vaddr=40088eec size=0d9f8h ( 55800) load
I (417) esp_image: segment 5: paddr=000d9b64 vaddr=50000000 size=00010h (    16) load
I (429) boot: Loaded app from partition at offset 0x10000
I (429) boot: Disabling RNG early entropy source...
I (440) cpu_start: Pro cpu up.
I (441) cpu_start: Starting app cpu, entry point is 0x40081354
0x40081354: call_start_cpu1 at /home/rick/esp/esp-idf/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (457) cpu_start: Pro cpu start user code
I (457) cpu_start: cpu freq: 160000000
I (457) cpu_start: Application information:
I (461) cpu_start: Project name:     hello-world
I (467) cpu_start: App version:      1
I (471) cpu_start: Compile time:     Jun 17 2021 20:04:38
I (477) cpu_start: ELF file SHA256:  e260b84c100cbb67...
I (483) cpu_start: ESP-IDF:          v4.4-dev-1594-g1d7068e4b
I (490) heap_init: Initializing. RAM available for dynamic allocation:
I (497) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (503) heap_init: At 3FFBAB18 len 000254E8 (149 KiB): DRAM
I (509) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (515) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (522) heap_init: At 400968E4 len 0000971C (37 KiB): IRAM
I (529) spi_flash: detected chip: generic
I (533) spi_flash: flash io: dio
E (537) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.
assertion "flash_ret == ESP_OK" failed: file "IDF/components/esp_system/startup.c", line 321, function: do_core_init

abort() was called at PC 0x40149937 on core 0
0x40149937: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

Backtrace:0x400d402b:0x3ffe3b000x40089a35:0x3ffe3b20 0x4009098e:0x3ffe3b40 0x40149937:0x3ffe3bb0 0x400d373d:0x3ffe3be0 0x400d389b:0x3ffe3c00 0x40081561:0x3ffe3c40 0x400791d0:0x3ffe3c80  |<-CORRUPTED
0x400d402b: panic_abort at /home/rick/esp/esp-idf/components/esp_system/panic.c:379

0x40089a35: esp_system_abort at /home/rick/esp/esp-idf/components/esp_system/esp_system.c:126

0x4009098e: abort at /home/rick/esp/esp-idf/components/newlib/abort.c:46

0x40149937: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400d373d: do_core_init at /home/rick/esp/esp-idf/components/esp_system/startup.c:321 (discriminator 1)

0x400d389b: start_cpu0_default at /home/rick/esp/esp-idf/components/esp_system/startup.c:391 (discriminator 3)

0x40081561: call_start_cpu0 at /home/rick/esp/esp-idf/components/esp_system/port/cpu_start.c:544

ELF file SHA256: e260b84c100cbb67

Rebooting...
ets Jun  8 2016 00:22:57
```

me-no-dev, 2021-06-18:
`E (537) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.`

You have selected 8MB flash but have 4MB on the module

RickvF, 2021-06-18:
Feeling a bit stupid right now......... found it and fixed it. Camera is working.
Next step adding the mic.

Thanks for the help!

RickvF, 2021-06-21:
Got only one error left which appears after a while.

W (303806) httpd_txrx: httpd_sock_err: error in send : 11
W (303806) httpd_uri: httpd)uri: uri hadnler execution failed
W (304806) httpd_txrx: httpd_sock_err: error in send : 11
E (304806) example: httpd_resp_send_chunk(): #5 error
W (304806) httpd_uri: httpd_uri: uri handler execution failed
W (304816) httpd_uri: httpd_uri: URI '/favicon.ico'not found
W (304816) httpd_txrx: httpd_resp_send_error: 404 Not Found - This URI does not exist

But in all my webpage I even didn't specify favicon.ico. Also in the example webservers I didn't saw the use of a favicon.

lbernstone, 2021-06-21:
that request comes from the web browser.  It wants to decorate the tab.  You can provide one as a static file if you really want.

sansillusion, 2021-06-21:
You may be interested in serving the icon inline im your main html to prevent multiple requests.

Found some details here about it:
https://davidwalsh.name/blank-favicon

lbernstone, 2021-06-22:
That's an interesting idea, but if you are serving more than one page, you would need to embed it in every page (or use a template).

closed 2021-09-19 by stale[bot]