espressif/arduino-esp32 #5281: ESP fails to boot: rst:0x0

coriolan-v, 2021-06-11:
Hello,

I have an issue with my ESP32-WROOM-32UE module on my custom PCB.
About 10% of the time on a small amount of PCBs, the MCU does not boot when I connect the power.
I get the following error:

`rst:0x0 (NO_MEAN),boot:0x6 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
ets_main.c 404`

I have searched everywhere for the description of this error rst:0x0 but it seems to be the only one not described [here ](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection) or [here ](https://github.com/espressif/esp-idf/blob/release/v3.0/components/esp32/include/rom/rtc.h#L80)

For comparison, this is what I get when my MCU boots correctly the rest of the time:
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac`

For the record, this is the status of the strapped GPIOS:
EN is 3.3V
GPIO2 is pulled down trough a 5.1kOhm resistor.
GPIO12 is pulled down trough a 10kOhm resistor.
GPIO15 is left floating.
[The 3.3V efuse is burnt](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12-esp32-only).

Any ideas? Thanks!

lbernstone, 2021-06-11:
It sounds like a power timing issue.  Espressif recommends putting a 1uF capacitor on the enable pin to ensure that the rail has reached 3V3 before starting boot. From the hardware design guide:
```
2.2 Power-on Sequence and System Reset
2.2.1 Power-on Sequence
ESP32 uses a 3.3 V system power supply. The chip should be activated after the power rails have stabilized. This
is achieved by delaying the activation of CHIP_PU (Pin9) after the 3.3 V rails have been brought up. More details
can be found in Section Power Scheme in ESP32 Datasheet.
Notice:
To ensure the power supply to the ESP32 chip during power-up, it is advised to add an RC delay circuit at the
CHIP_PU pin. The recommended setting for the RC delay circuit is usually R = 10 kΩ and C = 1 μF. However,
specific parameters should be adjusted based on the power-up timing of the power supply and the power-up and reset sequence timing of the chip.
```

coriolan-v, 2021-06-11:
Hello,
Thanks for that tip, that's really interesting. On my PCB I had a 0.1uF, I tried soldering a 1uF, and a 10uF, but no luck,..
Any other ideas?

[image]

coriolan-v, 2021-06-11:
I have done more research and I think I've pinned it down to GPIO14. I have connected it to VBUS trough a voltage divider bridge. 

When the MCU boots trough VBUS, GPIO14 is HIGH, and it sometimes doesn't boot.
When the MCU boots trough the battery, GPIO14 is LOW, and the it boots every time...
Any ideas why? I can't find any data about GPIO 14.

Thanks

TheRealDJ, 2021-06-13:
I can't seem to find anything about this in the ESP32 PDFs, but [https://randomnerdtutorials.com/esp32-pinout-reference-gpios/](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/) indicates that ```GPIO14``` and ```GPIO15``` output PWM on boot. Someone have a o-scope handy and could check this? I need a o-scope!

coriolan-v, 2021-06-14:
I can check tomorrow on the oscilloscope, but regardless of this, why would this prevent boot in some cases?

coriolan-v, 2021-06-14:
I have ran more tests by erasing the flash memory and uploading a simple Hello World.
Now I get this error message when it fails to boot, which happens abuot 10% of the time:

`ets Jul 29 2019 12:21:46
rst:0x6 (SDIO_RESET),boot:0x2 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_FEO_V2))
waiting for download`

Is this something to do with the GPIO0 signal? It's pulled up to 3.3V with a 5.1k resistor.

chansheunglong, 2021-06-15:
What is your voltage at GPIO0? [#577](https://github.com/espressif/arduino-esp32/issues/577) might have your answer.

coriolan-v, 2021-06-15:
I thought it was related to GPIO0 too so I shorted the resistor between
GPIO0 and 3.3V, still no difference.

closed 2021-09-06 by stale[bot]

vishesh-varma, 2023-02-09:
Were you able to figure out the final issue? And the solution? I've recently started seeing this same issue on a few custom boards that I am making for a project, but I haven't been able to pin down the final issue.

coriolan-v, 2023-02-18:
I've sorted it, I had small 0.1uF capacitors next to the VCC/3.3V pin. I replaced them with 2 massive 33uF capacitors and it solved the problem!