espressif/arduino-esp32 #5282: Issue with millis and micros with interrupt.

bhavikbhansali, 2021-06-11:
### Hardware:
Board:							Custom WROOM32D based board (selecting ESP32 Dev Module while uploading sketch)
Core Installation version:			1..0.4
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					80MHz
PSRAM enabled:                                    Disable
Upload Speed:				       921600
Computer OS:                                        Windows 10

### Description:
I am trying to calculate the time between two interrupt using millis and/or micros. It works but during the serial print it shows values with the error or you can say wrong value. attaching the code and serial monitor result will describe the issue. 

As per the source of the interrupt, the output value is 101 in the serial monitor.

what I am trying to do is calculate the RPM of a motor using a proximity sensor.

According to me I think the reason is due to millis and/or micros used with interrupt cause the issue. (I am not sure just guessing)

### Sketch: 
```cpp

//Change the code below by your sketch
#define  pin 19

volatile unsigned long time_prv = 0;
volatile unsigned long time_cur = 0;
int time_diff = 0;
volatile int counter = 0;

void IRAM_ATTR ttt() {
  time_prv = time_cur;
  time_cur = micros();
  counter++;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  pinMode(pin, INPUT);

  Serial.println("Wait for 2 Sec Delay");
  delay(5000);
  Serial.println("Over and Start");
  counter = 0;
  attachInterrupt(digitalPinToInterrupt(pin), ttt, FALLING);
}

void loop() {
  // put your main code here, to run repeatedly:
 time_diff = time_cur - time_prv;
 if(time_diff != 0)
 {
  Serial.println((60000000 / time_diff));
 }
}
```

### Debug Messages:
```
[serial_monitor_output.txt](https://github.com/espressif/arduino-esp32/files/6640111/serial_monitor_output.txt)

```
Thanks
Bhavik

TheRealDJ, 2021-06-13:
Not sure that this will fix the issue, but why don't you define ```time_diff``` as ```volatile``` and calculate within ```ttt()```. This will reduce processing in ```loop()``` ... since you have no delay in your ```loop()``` it could make a difference.

bhavikbhansali, 2021-06-14:
@TheRealDJ Thanks for response. I tried the mentioned thing and it gives the same result.
I dont know whats going on but i think facing very weird issue. Any other suggestion.

hpwit, 2021-06-20:
Try using ESP. getCycleCount() instead of micro()
Which gives you the number of cycles. So get get back in rpm you’ll need to do 240000000/timediff

closed 2021-09-19 by stale[bot]