espressif/arduino-esp32 #5285: 2.0.0-alpha MD5Builder linkage issue

tobozo, 2021-06-13:
Trying to compile this example sketch from a Linux PC (x86_64) using Arduino 1.18.15 along with sdk 2.0.0-alpha:

[image]

I get a linkage error after this line:

```
~/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2020r3/bin/xtensa-esp32-elf-g++ -Wl,--Map=/tmp/arduino_build_349781/SD_Update.ino.map -L~/.arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/tools/sdk/esp32/lib -L~/.arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/tools/sdk/esp32/ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32_out.ld -T esp32.project.ld -T esp32.peripherals.ld -mlongcalls -Wl,--cref -Wl,--gc-sections -fno-rtti -fno-lto -Wl,--wrap=mbedtls_mpi_exp_mod -u esp_app_desc -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_include_pthread_local_storage_impl -u ld_include_panic_highint_hdl -u start_app -u start_app_other_cores -u vfs_include_syscalls_impl -u call_user_start_cpu0 -Wl,--undefined=uxTopUsedPriority -u app_main -u newlib_include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u __cxa_guard_dummy -DESP32 -DCORE_DEBUG_LEVEL=2 -DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -DARDUINO_SERIAL_PORT=0 -Wl,--start-group /tmp/arduino_build_349781/sketch/SD_Update.ino.cpp.o /tmp/arduino_build_349781/libraries/Update/HttpsOTAUpdate.cpp.o /tmp/arduino_build_349781/libraries/Update/Updater.cpp.o /tmp/arduino_build_349781/libraries/FS/FS.cpp.o /tmp/arduino_build_349781/libraries/FS/vfs_api.cpp.o /tmp/arduino_build_349781/libraries/SD/sd_diskio_crc.c.o /tmp/arduino_build_349781/libraries/SD/SD.cpp.o /tmp/arduino_build_349781/libraries/SD/sd_diskio.cpp.o /tmp/arduino_build_349781/libraries/SPI/SPI.cpp.o /tmp/arduino_cache_328182/core/core_3c169ca6a9045a9187ddb2bfeb0989ba.a -lmbedtls -lefuse -lapp_update -lbootloader_support -lesp_ipc -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -llwip -llog -lheap -lsoc -lesp_hw_support -lesp_pm -lesp_ringbuf -ldriver -lxtensa -lperfmon -lesp32 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lasio -lbt -lcbor -lunity -lcmock -lcoap -lconsole -lnghttp -lesp-tls -lesp_adc_cal -lesp_hid -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lprotobuf-c -lprotocomm -lmdns -lesp_local_ctrl -lsdmmc -lesp_serial_slave_link -lesp_websocket_client -lexpat -lwear_levelling -lfatfs -lfreemodbus -ljsmn -ljson -llibsodium -lmqtt -lopenssl -lspiffs -lulp -lwifi_provisioning -lbutton -ljson_parser -ljson_generator -lesp_schedule -lesp_rainmaker -lqrcode -lws2812_led -lesp_littlefs -lesp-dsp -lesp-face -lesp32-camera -lfb_gfx -lasio -lcbor -lcmock -lunity -lcoap -lesp_hid -lesp_local_ctrl -lesp_websocket_client -lexpat -lfreemodbus -ljsmn -llibsodium -lbutton -lesp_rainmaker -lmqtt -lwifi_provisioning -lprotocomm -lprotobuf-c -ljson -ljson_parser -ljson_generator -lesp_schedule -lqrcode -lws2812_led -lesp-dsp -lesp-face -lpe -lfd -lfr -ldetection_cat_face -ldetection -ldl -lesp32-camera -lfb_gfx -lbt -lbtdm_app -lesp_adc_cal -lmdns -lconsole -lfatfs -lwear_levelling -lopenssl -lspiffs -lesp_littlefs -lmbedtls -lefuse -lapp_update -lbootloader_support -lesp_ipc -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -llwip -llog -lheap -lsoc -lesp_hw_support -lesp_pm -lesp_ringbuf -ldriver -lxtensa -lperfmon -lesp32 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lsdmmc -lesp_serial_slave_link -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lphy -lrtc -lmbedtls -lefuse -lapp_update -lbootloader_support -lesp_ipc -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -llwip -llog -lheap -lsoc -lesp_hw_support -lesp_pm -lesp_ringbuf -ldriver -lxtensa -lperfmon -lesp32 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lsdmmc -lesp_serial_slave_link -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lphy -lrtc -lmbedtls -lefuse -lapp_update -lbootloader_support -lesp_ipc -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -llwip -llog -lheap -lsoc -lesp_hw_support -lesp_pm -lesp_ringbuf -ldriver -lxtensa -lperfmon -lesp32 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lsdmmc -lesp_serial_slave_link -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lphy -lrtc -lmbedtls -lefuse -lapp_update -lbootloader_support -lesp_ipc -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -llwip -llog -lheap -lsoc -lesp_hw_support -lesp_pm -lesp_ringbuf -ldriver -lxtensa -lperfmon -lesp32 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lsdmmc -lesp_serial_slave_link -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lphy -lrtc -lmbedtls -lefuse -lapp_update -lbootloader_support -lesp_ipc -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -llwip -llog -lheap -lsoc -lesp_hw_support -lesp_pm -lesp_ringbuf -ldriver -lxtensa -lperfmon -lesp32 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lsdmmc -lesp_serial_slave_link -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lphy -lrtc -lxt_hal -lm -lnewlib -lgcc -lstdc++ -lpthread -lapp_trace -lgcov -lapp_trace -lgcov -lc -Wl,--end-group -Wl,-EL -o /tmp/arduino_build_349781/SD_Update.ino.elf
```

Error message:

```
~/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino_cache_328182/core/core_3c169ca6a9045a9187ddb2bfeb0989ba.a(MD5Builder.cpp.o):(.literal._ZN10MD5Builder5beginEv+0x0): undefined reference to `MD5Init'
~/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino_cache_328182/core/core_3c169ca6a9045a9187ddb2bfeb0989ba.a(MD5Builder.cpp.o):(.literal._ZN10MD5Builder3addEPht+0x0): undefined reference to `MD5Update'
~/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino_cache_328182/core/core_3c169ca6a9045a9187ddb2bfeb0989ba.a(MD5Builder.cpp.o):(.literal._ZN10MD5Builder9calculateEv+0x0): undefined reference to `MD5Final'
~/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino_cache_328182/core/core_3c169ca6a9045a9187ddb2bfeb0989ba.a(MD5Builder.cpp.o): in function `MD5Builder::begin()':
~/.arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/MD5Builder.cpp:32: undefined reference to `MD5Init'
~/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino_cache_328182/core/core_3c169ca6a9045a9187ddb2bfeb0989ba.a(MD5Builder.cpp.o): in function `MD5Builder::add(unsigned char*, unsigned short)':
~/.arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/MD5Builder.cpp:37: undefined reference to `MD5Update'
~/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2020r3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino_cache_328182/core/core_3c169ca6a9045a9187ddb2bfeb0989ba.a(MD5Builder.cpp.o): in function `MD5Builder::calculate()':
~/.arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/MD5Builder.cpp:97: undefined reference to `MD5Final'
collect2: error: ld returned 1 exit status
```

tobozo, 2021-06-13:
closing this as fixed on upstream and probably redundant

closed 2021-06-13 by tobozo

SelaliAdobor, 2021-06-28:
I ran into this same issue using the library as a component in esp-idf and solved it by adding `esp32.rom.redefined.ld` to the `LINKER_SCRIPTS` in my esp-idf folder

When you say fixed on upstream do you mean there's a fix for this in this repo, or in the esp-idf repo, since I was running into it on master of both

tobozo, 2021-06-28:
@SelaliAdobor yes it was solved by pulling the `.ld` file from the master of this repo.