espressif/arduino-esp32 #5287: Esp32c3 `Serial.baudRate()` is 2x too high

s-hadinger, 2021-06-14:
### Hardware:
Board:							ESP32-C3-DevKitM-1
Core Installation version:			2.0.0 alpha1
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Mac OSX

### Description:
Describe your problem here

The baudrate returned by `Serial.baudRate()` is twice the expected value.

The Serial port is opened with 115200, but `Serial.baudRate()` returns ~230400.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.printf(">> Serial baudrate: %d\n", Serial.baudRate());
  delay(1000);
}
// Output:
// >> Serial baudrate: 230423
```

lbernstone, 2021-06-14:
Yes, uart needs to be rewritten.  This should be a workaround until then:
```
#include <driver/uart.h>
void setup() {
  Serial.begin(115200);
  uint32_t br;
  uart_get_baudrate(0, &br);
  Serial.println(br);
}
```

s-hadinger, 2021-06-14:
Thanks, this works fine.

SuGlider, 2021-08-15:
UART is being refactored to be based on IDF and this issue shall be solved when it is released.

PR #5549 (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF
## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.
commit b1d072df (2021-08-23): Implements UART SerialHardware Refactoring on top of IDF (#5549)

## Summary
This PR is a complete reffactoring of UART Serial Hardware and respective HAL in order to use IDF instead of current Register manipulation approach. 

It  implements Arduino SerialEvent functionality. 

Fix #5287  
Fix #5273 
Fix #5519 
Fix #5247 
Fix #5403
Fix #5429
Fix #5047
Fix #5463
Fix #5362 
Fix #5112  
Fix #5443 

## Impact
It solves many reported issues related to UART.
It was tested and works fine for ESP32, ESP-S2 and ESP32-C3.
closed 2021-08-23 by me-no-dev