espressif/arduino-esp32 #5291: ESP8266 Not Saving Wifi Credentials - Loses after power loss

Abhesheksh, 2021-06-15:
Is there anyway to retain the wifi ssid & password on reset?

 Here is my code : 

`#include <ESP8266WiFi.h>;

void setup() {
  Serial.begin(115200);
  delay(10);

  WiFi.mode(WIFI_STA);
  delay(500);
IPAddress local_IP(192, 168, 1, 199);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 0, 0);
IPAddress primaryDNS(8, 8, 8, 8); // this is optional
IPAddress secondaryDNS(8, 8, 4, 4); // this is optional

// Print feedback if the settings are not configured
if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
  Serial.println("STA Failed to configure");
}
  WiFi.beginSmartConfig();

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    Serial.println(WiFi.smartConfigDone());
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

int value = 0;

void loop() {

}`

chegewara, 2021-06-15:
What it has to do with esp32?

Abhesheksh, 2021-06-15:
Oops sorry.

closed 2021-06-15 by Abhesheksh