espressif/arduino-esp32 #5297: ESP32 is three times slower than ESP8266 with MQTT and TCP related stuff

sblantipodi, 2021-06-16:
### Hardware:
Board:							ESP32 from D32 board from Lolin
Core Installation version:			1.0.6
IDE name:						Platformio (platform-espressif32 v.3.2.1)
LIB Used:                                                 pubsubclient for MQTT, world most used MQTT lib I think (v2.8)
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					500.000 or 1.000.000 or 2.000.000
Computer OS:                                         Windows 10 and Ubuntu 

### Description:
ESP32 is at least 3 times slower than ESP8266 when using MQTT or TCP more in general.
I have created a sketch that subscribe to an MQTT topic and increment a counter, every five seconds print the number of received message per second.

With ESP32 that sketch goes up to 45 messages per second,
ESP8266 scores 145 messages per second.

Isn't ESP32 supposed to be faster than ESP32?

### Sketch: 
This is a really small sketch that shows the problem, it works on both ESP8266 and ESP32

```cpp

#if defined(ESP8266)
#include <ESP8266WiFi.h>
#else
#include <WiFi.h>
#endif
#if defined(ESP8266)
#include <ESP8266WebServer.h>
#else
#include <WebServer.h>
#endif
#include <PubSubClient.h>
#include <FastLED.h>

const char *mqtt_server = "192.168.1.3"; //mqtt server
const char *ssid = "XXX";
const char *password = "XXX";
float framerate = 0;
float framerateCounter = 0;

WiFiClient espClient;
PubSubClient client(espClient); //lib required for mqtt

void callback(char *topic, byte *payload,
              unsigned int length) {   //callback includes topic and payload ( from which (topic) the payload is comming)
  framerateCounter++;

}

void reconnect() {
  while (!client.connected()) {
    Serial.println("Attempting MQTT connection...");
    if (client.connect("ESP32_clientID")) {
      Serial.println("connected");
      client.subscribe("lights/glowwormluciferin/set/stream"); //topic=Demo
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void connectmqtt() {
  client.connect("ESP32_clientID");  // ESP will connect to mqtt broker with clientID
  {
    Serial.println("connected to MQTT");
    client.subscribe("lights/glowwormluciferin/set/stream"); //topic=Demo
    if (!client.connected()) {
      reconnect();
    }
  }
}

void setup() {
  Serial.begin(1000000);
  pinMode(LED, OUTPUT);
  digitalWrite(LED, LOW);
  WiFi.begin(ssid, password);
  Serial.println("connected");
  client.setServer(mqtt_server, 1883);//connecting to mqtt server
  client.setCallback(callback);
  //delay(5000);
  connectmqtt();
}

void loop() {
  if (!client.connected()) {
    reconnect();
  }

  client.loop();

  EVERY_N_SECONDS(10) {
    framerate = framerateCounter > 0 ? framerateCounter / 10 : 0;
    framerateCounter = 0;
    Serial.println("framerate");
    Serial.println(framerate);
  }

}

```

Can you fix this performance problem on ESP32 please?
Is it a problem in its WiFiClient ?

VojtechBartoska, 2021-06-17:
Thanks @sblantipodi for reporting this issue.

@david-cermak , PTAL

sblantipodi, 2021-06-17:
Thank you for the answer @VojtechBartoska 

I forgot to say that the "EVERY_N_SECONDS" is a function offered by the FastLED library.
https://github.com/FastLED/FastLED

It's just a function that execute some lines of code every N seconds. You can change it with everything you want like millis() for example, the results doesn't change.

TLS1000, 2021-06-18:
Hi, Sorry I can't help you with the speed issue. But currently I'm also working on a project with mqtt (using MQTT library), and I'm using multi-thread. I suffer from frequent disconnects/reconnects. Is the pubsub library thread-safe, can the WIFI chip perform other concurrent sessions without completely disconnecting the mqtt-client? I don't think there's a wrapper around the idf-mqtt; why would that be? I suppose the idf-mqtt API will be thread-safe.

sblantipodi, 2021-06-18:
@TLS1000 please open another issue, I don't think that your question fits this issue.

amarotica, 2021-06-20:
> Hi, Sorry I can't help you with the speed issue. But currently I'm also working on a project with mqtt (using MQTT.h library), and I'm using multi-thread. I suffer from frequent disconnects/reconnects. Is the pubsub library thread-safe, can the WIFI chip perform other concurrent sessions without completely disconnecting the mqtt-client? I don't think there's a wrapper around the idf-mqtt; why would that be? I suppose the idf-mqtt API will be thread-safe.

Hi, I've experienced the same unstable connect/disconnects and found my solution was to remove WiFi.RSSI() calls! Somehow WiFi.RSSI() breaks TLS connections of WiFiClientSecure...

YordanYanakiev, 2021-07-05:
Is this for 1.06 or other SDK ?

sblantipodi, 2021-07-05:
@YordanYanakiev every Arduino SDK I tried, runs slower than ESP8266 but yes, I'm running it on 1.06

YordanYanakiev, 2021-07-06:
1.0.4 is a lot faster than 1.0.6, but it is having bugs with some stuff, but yes - ESP32 by some reason is slower than 8266 :|

sblantipodi, 2021-07-17:
hi @VojtechBartoska, some weeks ago you assigned the issue to @david-cermak but now the issue is unassigned.
do you know if there is there someone working on it, is this a known problem that is not fixable?

can you give some more infos on this topic please?

thank you!

YordanYanakiev, 2021-07-17:
P.S: A great way to see the difference and note the issue is to use some ESP32 which runs on battery ( a smaller capacity - a better ), then setup : Wifimulti with a few AP's on it's list, then scan the network for AP's, then connect to the WifiMulti, then get some data from some server, then go to LIGHT SLEEP ( very important ), then resume from it with a timer, and repeat the process. observe.

sblantipodi, 2021-08-03:
pretty sad to see this platform abandoned. it seems that good old ESP8266 will have longer life than ESP32 xD

TLS1000, 2021-08-03:
who needs +45 messages per second?

lbernstone, 2021-08-03:
Weird, I would have expected that trolling and complaining about how shitty the product is would get you all sorts of unpaid support.

TLS1000, 2021-08-03:
Hail to the developers and contributors of the magnificent arduino-ESP32 platform!

sblantipodi, 2021-08-05:
Trying with UDP too, ESP32 is still way slower than ESP8266 even when using UDP.

me-no-dev, 2021-08-10:
ESP32 and ESP8266 use vastly different approaches to handle the network. 8266 is single core, uses very low level LwIP API and can get away with it. ESP32 is dual-core, running on top of an actual OS. It's targets are not to be the fastest, but the safest option. How the libs are then implemented is also important. Try the Async libs if you want to have speed on ESP32.

closed 2021-08-10 by me-no-dev

sblantipodi, 2021-08-10:
@me-no-dev I see no one putting delays or moving functions between "core tasks" on ESP8266 just to not make it crash or to not trigger the watchdog.

I would not consider ESP32 the safer option but ok, your answer clearly shows that ESP32 is slower than ESP8266 in WiFi (TCP/UDP) related stuffs and that Espressif knows it and won't fix.
Thanks.

me-no-dev, 2021-08-10:
@sblantipodi I understood nothing from that comment?

TCP and UDP on ESP32 are not broken. I can safely push over 10Mbps TCP. The fact that some interactions are slower do not make them broken at all. While on ESP8266 is NOT possible to call the network from more than one task at the same time, on ESP32 that is a very possible thing. They are not the same thing at all. I am sure that whatever lib you are using, can be optimized for ESP32.

sblantipodi, 2021-08-10:
@me-no-dev don't want to be unpolite but how can I optimize this?

```
if (packetSize) {
      int len = UDP.read(packet, UDP_PACKET_SIZE);
      if (len > 0) {
        packet[len] = '\0';
        // I have my packet two times slower than ESP8266 here.
      }
    }
```

this is the easyest code I know for UDP and it still slower on ESP32.

me-no-dev, 2021-08-10:
2 times slower is not a good measurement :) how many milliseconds is the difference?
In any way, there isn't much that can be done. The OS requires it's own time to switch tasks, pass data and do other things, while on ESP8266 you are only waiting for the packet to be received and passed on to LwIP (while yielding). 1 task running on single core will always be faster than multiple tasks in OS environment.

GiuseppeCervo, 2023-08-03:
I've worked with n a distributed application that used a server (Pi) and some nodes (ESP8266), where we verified connection responsiveness through timed pings between the server and the nodes. When developing a new feature, I tried to implement them using the ESP32 and was struggling with ping delays (using the same communication protocol, the difference was just that I changed ESP8266WiFi.h to WiFi.h). What did the trick for me was applying WiFi.setSleep(false), a line I found when struggling with UDP async responsiveness and which is supposed to deactivate some energy saving feature. Maybe give it try if you still have the sketches you've used to compare the devices! Curious on the outcome.