espressif/arduino-esp32 #5302: ESP32 Arduino - Ble deinit

savagerex, 2021-06-18:
I use ESP32 Ble uart sample code in ESP32 arduino 1.0.4.

When i porting this sample code to ESP32 arduino 1.0.6.

the program will get stuck in the BLEDevice::deinit(false);

but i use ESP32 Ble uart sample code in ESP32 arduino 1.0.6.
the program is no problem.

so please help me to solve

Thank you

savagerex, 2021-06-18:
Board  : ESP32 dev module
Core install version:  1.0.4 & 1.0.6
IDE name : Arduino
Flash Frequency : 80MHz
PSRAM Enable : yes
upload speed :115200
computer os : win10

i mean i write program to stop ble, i use (BLEDevice::deinit(false);) in 1.0.6. 
the program is no problem.

 i write program to stop ble, i use (BLEDevice::deinit(false);) in 1.0.4.
then porting to 1.0.6.
the program will get stuck in the BLEDevice::deinit(false);

me-no-dev, 2021-06-18:
Hi @savagerex 

Is the program getting stuck, or are you getting an exception? Do you see any errors in the uart console?

me-no-dev, 2021-06-18:
@chegewara do you have any idea what could be causing this?

chegewara, 2021-06-18:
I dont understand what it means to port program from v1.0.4 to v1.0.6 and how it may differ from code wrote for v1.0.6 which seems to work.

me-no-dev, 2021-06-18:
issue is that `BLEDevice::deinit(false);` is not working. Porting here refers to just try everything on 1.0.6, not that code had to be changed

savagerex, 2021-06-18:
Hello me-no-dev

the program is getting stuck.  there is no message & no any error in uart console.

chegewara, 2021-06-18:
`deinit` function is fairly simple. You can try this code instead:
```
    esp_err_t err;
    err = esp_bluedroid_disable();
    err |= esp_bluedroid_deinit();
    err |= esp_bt_controller_disable();
    err |= esp_bt_controller_deinit();
    if(err) Serial.println("Error to deinit ble: %d", err);
```
But i dont see what could cause it to stuck, none of this is blocking API.

me-no-dev, 2021-06-18:
I have a faint memory that we had such issue at some point and one of the calls above was hanging, but we fixed it. @savagerex can you please check if that is your case and which call is hanging?

savagerex, 2021-06-18:
@chegewara 

complier this, it has problem .

'esp_bluedroid_disable' was not declared in this scope

me-no-dev, 2021-06-18:
@savagerex deinit is defined here: https://github.com/espressif/arduino-esp32/blob/release/v1.0/libraries/BLE/src/BLEDevice.cpp#L636-L650

You get the error above because you are missing some includes: https://github.com/espressif/arduino-esp32/blob/release/v1.0/libraries/BLE/src/BLEDevice.cpp#L14-L16

savagerex, 2021-06-18:
@me-no-dev 

can you see the below the printscreen???

[image]

me-no-dev, 2021-06-18:
@savagerex question is where inside `BLEDevice::deinit()` it's hanging. The code is calling only IDF functions and we need to find out first which one is failing

chegewara, 2021-06-18:
But if you call `btStop()` you cant call `deinit()`. It makes no sense.

savagerex, 2021-06-18:
@me-no-dev 

there is fail in esp_bluedroid_disable();

[image]

chegewara, 2021-06-18:
Again, when you call `btStop` it will disable controller and you cant call `deinit`. In `deinit` there is `esp_bluedroid_disable` which has to be called before disable bt controller. Maybe that esp-idf shouldnt cause stuck, but it is error in your app (and lack of fail check in BLE library too).

savagerex, 2021-06-18:
@chegewara 

got it!!. now, the program is no problem.

but  i will keep testing

Thank you!!!

closed 2021-09-19 by stale[bot]