espressif/arduino-esp32 #5305: ESP32- Connect to web server via Ethernet & WiFi at the same time

BoYi0126, 2021-06-18:
### Hardware:
Board: ESP32-Ethernet-Kit_A_V1.1
IDE: Arduino 1.8.13
OS: Win10

### Description:
I want to connect my AsyncWebServer via Ethernet and WiFi, and connect to the server by IP.
But it cannot work at same time.
If connect ethernet and wifi at same time, only ehternet can work...

||||
|:---|---|---|
|**Ethernet**|**WiFi**|**Connect to web server**|
|Connect|Disconnect|Ethernet|
|Disconnect|Connect|WiFi|
|Connect|Connect|Ethernet||

### Sketch:
#include <ETH.h>
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <ESPmDNS.h>

#define SOFT_AP

#define ETH_ADDR        1
#define ETH_POWER_PIN   5
#define ETH_MDC_PIN     23
#define ETH_MDIO_PIN    18
#define ETH_TYPE        ETH_PHY_IP101

// ETH variables
IPAddress ethIp(192, 168, 1, 18);
IPAddress gateway(192, 168, 1, 2);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns1(192, 168, 1, 10);
IPAddress dns2(192, 168, 1, 20);

// WiFi variables
const char *apSsid = "WIFI_AP";
IPAddress apIp(192, 168, 1, 19);

#define PORT_1 80

AsyncWebServer asyncWebServer(PORT_1);

static bool eth_connected = false;
    
void setupETH()
{
  // Start the ethernet
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE);
  ETH.config(ethIp, gateway, subnet);
}

void notFound(AsyncWebServerRequest *request)
{
  request->send(404, "text/plain", "Not found");
}

void setHtmlCallback()
{
  asyncWebServer.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    Serial.println("Run call back");
    request->send(200, "text/plain", "Hello world");
  });
  
  asyncWebServer.onNotFound(notFound);
}

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      Serial.print("ETH IP: ");
      Serial.println(ETH.localIP());
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void setup()
{
  Serial.begin(115200);

  WiFi.onEvent(WiFiEvent);
  

#ifdef SOFT_AP
  WiFi.mode(WIFI_AP);
  WiFi.softAP(apSsid);
  WiFi.softAPConfig(apIp, gateway, subnet);
#endif

  setupETH();
  
  // Start the Ethernet web server
  setHtmlCallback();
  asyncWebServer.begin();

  Serial.print("WiFi MAC: ");
  Serial.println(WiFi.macAddress());
  Serial.print("ETH MAC: ");
  Serial.println(ETH.macAddress());
}

void loop()
{
}

BoYi0126, 2021-06-18:
I catch some console:

### Step1. Start program and print mac
[D][esp32-hal-psram.c:60] psramInit(): PSRAM enabled
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 21 - ETH_START
ETH Started
WiFi MAC: 84:0D:8E:D3:64:6C
ETH MAC: 84:0D:8E:D3:64:6F

### Step2. Connect to the wifi
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 16 - AP_STACONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED

### Step3. Connect to the web server via wifi
Run call back
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4

### Step4. Wire the Ethernet cable
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 25 - ETH_GOT_IP
[D][WiFiGeneric.cpp:455] _eventCallback(): ETH IP: 192.168.1.18, MASK: 255.255.255.0, GW: 192.168.1.2
84:0D:8E:D3:64:6F, IPv4: 192.168.1.18, FULL_DUPLEX, 100Mbps
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 25 - ETH_GOT_IP
[D][WiFiGeneric.cpp:455] _eventCallback(): ETH IP: 192.168.1.18, MASK: 255.255.255.0, GW: 192.168.1.2
84:0D:8E:D3:64:6F, IPv4: 192.168.1.18, FULL_DUPLEX, 100Mbps

### Step5. Connect to the web server via ETH
Run call back
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4

### Step6. Refresh web via wifi, and the web server no refresh
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 17 - AP_STADISCONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 16 - AP_STACONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED

PilnyTomas, 2022-01-27:
Hi, could you please provide some info about the network setup around this sketch? What do I need to do to replicate your problem?

BoYi0126, 2022-02-08:
Hi Sir,

Sorry about replying late,
I deal with my boss, and now I don't need to solve this problem.
This issue will be closed after 14 days, thank you for your asking!

Best regards,
Bob Wu