espressif/arduino-esp32 #5308: ESP32_AzureIoT_Arduino questions

PDE-Paul, 2021-06-18:
Hi All,

I am getting random Core resets when I implement Esp32MQTTClient.h

I need to know what is the minimum heap required for stable execution of VSChina / ESP32_AzureIoT_Arduino so I can determine how to manage this random resetting issue.

Is anyone aware of a similar library that is actually being maintained as the Esp32MQTTClient library has not been looked at in 3 years.

Thanks in advance
Paul Brink

closed 2021-09-19 by stale[bot]

thaynapatricio, 2021-11-23:
Can you add the Esp32MQTTClient library for me to download? I'm not finding her anywhere.

thaynapatricio, 2021-11-23:
There's actually one version of Azure IoT that I didn't find in my directory.

thaynapatricio, 2021-11-23:
[AzureIoT.zip](https://github.com/espressif/arduino-esp32/files/7590876/AzureIoT.zip)

There's actually one version of Azure IoT that I didn't find in my directory.