espressif/arduino-esp32 #5313: ESP-IDF in Visual Studio Code wont compile Arduino as a component

Olgidos, 2021-06-21:
When creating a project form an Arduino template or adding a Arduino component to a project, it wont compile. Since I thought this was originally a problem of the VSC ESP IDF extension, I created an issue over there. [->issue](https://github.com/espressif/vscode-esp-idf-extension/issues/441). There I was informed that I should post an issue here as well. 

**Describe the bug**
Wont compile arduino-as-component. 

**To Reproduce**
I am vscode-esp-idf-extension version 1.0.3. pressed F1 "create project from extension template" -> arduino as component -> built

**Expected behavior**
A correct built

**Environment (please complete the following information):**
- OS Name & Version: [Win 10]
- VSCode Version: [1.57.1]
- ESP-IDF Version: [4.3]
- Python Version: [3.9]

**Logs**
> Executing task: cmake -G Ninja .. <

-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.27.0.windows.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/TheOlg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/TheOlg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Users/TheOlg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/TheOlg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/TheOlg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from C:\Users\TheOlg\esp\esp-idf\requirements.txt are satisfied.
-- Project sdkconfig file C:/Users/TheOlg/Desktop/svn/esp32/arduino-as-component/sdkconfig
-- Found PythonInterp: C:/Users/TheOlg/.espressif/python_env/idf4.3_py3.9_env/Scripts/python.exe (found version "3.9.5") 
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
CMake Warning (dev) at C:/Users/TheOlg/esp/esp-idf/components/mbedtls/CMakeLists.txt:114 (target_sources):
  Policy CMP0076 is not set: target_sources() command converts relative paths
  to absolute.  Run "cmake --help-policy CMP0076" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

  A private source from a directory other than that of target "mbedcrypto"
  has a relative path.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- App "arduino-as-component" version: 1
-- Adding linker script C:/Users/TheOlg/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/TheOlg/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/TheOlg/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/TheOlg/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/TheOlg/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/TheOlg/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/TheOlg/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Users/TheOlg/Desktop/svn/esp32/arduino-as-component/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/TheOlg/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/TheOlg/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update arduino asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/TheOlg/esp/esp-idf/components/app_trace C:/Users/TheOlg/esp/esp-idf/components/app_update C:/Users/TheOlg/Desktop/svn/esp32/arduino-as-component/components/arduino C:/Users/TheOlg/esp/esp-idf/components/asio C:/Users/TheOlg/esp/esp-idf/components/bootloader C:/Users/TheOlg/esp/esp-idf/components/bootloader_support C:/Users/TheOlg/esp/esp-idf/components/bt C:/Users/TheOlg/esp/esp-idf/components/cbor C:/Users/TheOlg/esp/esp-idf/components/cmock C:/Users/TheOlg/esp/esp-idf/components/coap C:/Users/TheOlg/esp/esp-idf/components/console C:/Users/TheOlg/esp/esp-idf/components/cxx C:/Users/TheOlg/esp/esp-idf/components/driver C:/Users/TheOlg/esp/esp-idf/components/efuse C:/Users/TheOlg/esp/esp-idf/components/esp-tls C:/Users/TheOlg/esp/esp-idf/components/esp32 C:/Users/TheOlg/esp/esp-idf/components/esp_adc_cal C:/Users/TheOlg/esp/esp-idf/components/esp_common C:/Users/TheOlg/esp/esp-idf/components/esp_eth C:/Users/TheOlg/esp/esp-idf/components/esp_event C:/Users/TheOlg/esp/esp-idf/components/esp_gdbstub C:/Users/TheOlg/esp/esp-idf/components/esp_hid C:/Users/TheOlg/esp/esp-idf/components/esp_http_client C:/Users/TheOlg/esp/esp-idf/components/esp_http_server C:/Users/TheOlg/esp/esp-idf/components/esp_https_ota C:/Users/TheOlg/esp/esp-idf/components/esp_https_server C:/Users/TheOlg/esp/esp-idf/components/esp_hw_support C:/Users/TheOlg/esp/esp-idf/components/esp_ipc C:/Users/TheOlg/esp/esp-idf/components/esp_local_ctrl C:/Users/TheOlg/esp/esp-idf/components/esp_netif C:/Users/TheOlg/esp/esp-idf/components/esp_pm C:/Users/TheOlg/esp/esp-idf/components/esp_ringbuf C:/Users/TheOlg/esp/esp-idf/components/esp_rom C:/Users/TheOlg/esp/esp-idf/components/esp_serial_slave_link C:/Users/TheOlg/esp/esp-idf/components/esp_system C:/Users/TheOlg/esp/esp-idf/components/esp_timer C:/Users/TheOlg/esp/esp-idf/components/esp_websocket_client C:/Users/TheOlg/esp/esp-idf/components/esp_wifi C:/Users/TheOlg/esp/esp-idf/components/espcoredump C:/Users/TheOlg/esp/esp-idf/components/esptool_py C:/Users/TheOlg/esp/esp-idf/components/expat C:/Users/TheOlg/esp/esp-idf/components/fatfs C:/Users/TheOlg/esp/esp-idf/components/freemodbus C:/Users/TheOlg/esp/esp-idf/components/freertos C:/Users/TheOlg/esp/esp-idf/components/hal C:/Users/TheOlg/esp/esp-idf/components/heap C:/Users/TheOlg/esp/esp-idf/components/idf_test C:/Users/TheOlg/esp/esp-idf/components/jsmn C:/Users/TheOlg/esp/esp-idf/components/json C:/Users/TheOlg/esp/esp-idf/components/libsodium C:/Users/TheOlg/esp/esp-idf/components/log C:/Users/TheOlg/esp/esp-idf/components/lwip C:/Users/TheOlg/Desktop/svn/esp32/arduino-as-component/main C:/Users/TheOlg/esp/esp-idf/components/mbedtls C:/Users/TheOlg/esp/esp-idf/components/mdns C:/Users/TheOlg/esp/esp-idf/components/mqtt C:/Users/TheOlg/esp/esp-idf/components/newlib C:/Users/TheOlg/esp/esp-idf/components/nghttp C:/Users/TheOlg/esp/esp-idf/components/nvs_flash C:/Users/TheOlg/esp/esp-idf/components/openssl C:/Users/TheOlg/esp/esp-idf/components/partition_table C:/Users/TheOlg/esp/esp-idf/components/perfmon C:/Users/TheOlg/esp/esp-idf/components/protobuf-c C:/Users/TheOlg/esp/esp-idf/components/protocomm C:/Users/TheOlg/esp/esp-idf/components/pthread C:/Users/TheOlg/esp/esp-idf/components/sdmmc C:/Users/TheOlg/esp/esp-idf/components/soc C:/Users/TheOlg/esp/esp-idf/components/spi_flash C:/Users/TheOlg/esp/esp-idf/components/spiffs C:/Users/TheOlg/esp/esp-idf/components/tcp_transport C:/Users/TheOlg/esp/esp-idf/components/tcpip_adapter C:/Users/TheOlg/esp/esp-idf/components/tinyusb C:/Users/TheOlg/esp/esp-id [... cut ...]
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

chegewara, 2021-06-21:
The logs looks exactly how suppose to look.
You just didnt copy sdkconfig from arduino-esp32 and you dont have enabled required configurations.

me-no-dev, 2021-06-22:
Looks like wrong version of ESP-IDF. It is planned to support a few older versions, but it's not yet done. Please make sure that you are using ESP-IDF master

Olgidos, 2021-06-22:
I was just using the vscode-esp-idf-extension methods to integrate arduino and therefore expected, that everything is working. do you think it is then an issue of the extension?

LearnDude, 2021-06-25:
@chegewara I have the same error. From where to where should I compy sdkconfig?

chegewara, 2021-06-25:
From here (depends what chip family):
https://github.com/espressif/arduino-esp32/tree/master/tools/sdk

to your project. You will find sdkconfig in there after running `idf.py build`.

nemccarthy, 2021-07-21:
I tired copying the sdkconfig however still getting the same error when running `idf.py build`

closed 2022-03-30 by VojtechBartoska