espressif/arduino-esp32 #5316: Is it possible to use BLE Link-level Encryption? How?

pscs, 2021-06-22:
Board:							ESP-DevkitC-WRover
Core Installation version:			1.0.6
IDE name:						Arduino 1.8.13

### Description:

I'm trying to connect to a device (a Victron SmartSolar charge controller) to read data from it. That uses BLE Encryption.  I've used a BLE packet sniffer to watch the connection between the official app and the device and it goes somewhat like:

M->S  - LE_LL -   LL_VERSION_IND
S->M -  LE_LL -   LL_VERSION_IND
M->S -  LE_LL -   LL_FEATURE_REQ
S->M -  LE_LL -   LL_FEATURE_RSP
M->S -  LE_LL -   LL_ENC_REQ
S->M -  LE_LL -   LL_ENC_RSP
M->S -  LE_LL -   LL_START_ENC_RSP
(then everything is encrypted)

When trying to achieve similar from my ESP32 I've used this as a basis for testing: 

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/security/SampleClient_authentication_passkey.cpp

But, the BLE Sniffer shows that the ESP32 never sends the LL_ENC_REQ opcode and just sends a pairing request (which fails)

M->S  - LE_LL -   LL_VERSION_IND
S->M -  LE_LL -   LL_VERSION_IND
M->S -  LE_LL -   LL_FEATURE_REQ
S->M -  LE_LL -   LL_FEATURE_RSP
....
M->S - SMP - Sent Pairing Request:....
S->M - SMP -> Rcvd Pairing Failed: Invalid parameters

I have fiddled with all the parameters that look relevant. eg 'BLEDevice::setEncryptionLevel' - but that just alters the parameters used in the Pairing Request

So - my question is - is there any way to have the ESP32 use BLE encryption using the LL_REQ_ENC etc opcodes, or am I flogging a dead horse? If there is a way, how do I enable it? 

I've Googled everywhere and can't find anything...

pscs, 2021-06-22:
Yes, I tried that - exactly the same - the ESP32 still doesn't send the LL_REQ_ENC, still sends a pairing request, and it still gets an 'invalid parameters' response back

FWIW, I'm puzzled because I can't see where the ESP32 is getting the PIN code from for the pairing request (with ESP_IO_CAP_IO). The SecurityCallbacks class isn't being called except for the onAuthenticationComplete method to say the pairing failed.

The sample would suggest that the PIN is provided in the onPassKeyRequest method, but that never gets called (I've added logging to that method to check).

This is the ESP32 log I'm getting

```[  2343][V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: RegEvt (0x3ffd98dc), owner: <N/A>
[  2352][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: OpenEvt (0x3ffd9c30), owner: <N/A> for connect
[  2362][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: OpenEvt (0x3ffd9c30), owner: connect
[  2370][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3ffd9c30), owner: connect for connect
[  2636][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT
[  2638][V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: ESP_GATTC_CONNECT_EVT
[  2644][V][BLEUtils.cpp:1311] dumpGattClientEvent(): [conn_id: 0, remote_bda: c4:17:54:d3:f4:c2]
[  2652][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT
[  2663][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
[  2674][V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
[  2681][V][BLEUtils.cpp:1402] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, remote_bda: c4:17:54:d3:f4:c2, mtu: 23]
[  2692][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
*********************MPPT Connected

[  2708][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: OpenEvt (0x3ffd9c30), owner: connect
[  2715][V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: OpenEvt (0x3ffd9c30), owner: <N/A>
[  2724][V][BLEClient.cpp:149] connect(): << connect(), rc=1
Connected to Server

[  2731][V][BLEClient.cpp:448] getServices(): >> getServices
[  2737][V][BLEClient.cpp:73] clearServices(): >> clearServices
[  2742][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_AUTH_CMPL_EVT
[  2742][V][BLEClient.cpp:80] clearServices(): << clearServices
[  2756][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: SearchCmplEvt (0x3ffd9c94), owner: <N/A> for getServices
[  2767][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SearchCmplEvt (0x3ffd9c94), owner: getServices
[  2751][V][BLEUtils.cpp:1105] dumpGapEvent(): [bd_addr: c4:17:54:d3:f4:c2, key_present: 0, key: ***, key_type: 0, success: 0, fail_reason: 87, addr_type: ***, dev_type: ESP_BT_DEVICE_TYPE_BLE]
[  2776][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffd9c94), owner: getServices for getServices
[  2793][I][BLEDevice.cpp:253] gapEventHandler(): ESP_GAP_BLE_AUTH_CMPL_EVT
*********************onAuthenticationComplete
*********************pair status = fail
[  2819][D][BLEClient.cpp:493] handleGAPEvent(): BLEClient ... handling GAP event!
[  2827][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT

chegewara, 2021-06-22:
In that case maybe try to remove all of this:
```
		BLESecurity *pSecurity = new BLESecurity();
		pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_ONLY);
		pSecurity->setCapability(ESP_IO_CAP_OUT);
		pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
```

and leave only that code:
```
		BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
		BLEDevice::setSecurityCallbacks(new MySecurity()); // probably this line is also not needed
```

pscs, 2021-06-22:
I've tried that - almost exactly the same result,.

BUT, I think I may be getting somewhere. I've been doing lots of reading about BLE encryption :-)

I think I was probably getting confused because it looks as if the Android App had already got an LTK (long-term key) so it was skipping part of the encryption setup process. I tried the app on a different phone and the BLE sniffer showed a lot more stuff that I wasn't seeing before.

It looks as if a key part was that this particular BLE server device requires the pairing request to use ESP_IO_CAP_KBDISP, not _IO and not _NONE - once I set that, then the pairing progresses better. It still fails, but given that the ESP program is going through some sort of passcode exchange now, where it wasn't before, I think I'm getting somewhere, and it's probably something more obvious now ;-)

Thanks for helping

chegewara, 2021-06-22:
Thats good.

Yes, if smartphone is already paired with device then there is a lot less communication to authenticate.
If that peripheral has static pin then you are good and you can easy make it works, in other case you will need display and or keypad.

pscs, 2021-06-22:
Yes, I've got it pairing now (it's a 'static' pin in this case). I can't seem to get notifications to work, but that's a different problem.

In case anyone else comes here looking at this issue, this is what the code snippet ended up like
```class MySecurityCallbacks: public BLESecurityCallbacks {
  public:
  uint32_t onPassKeyRequest() override {//gets called to obtain PIN
    printf("onPasskeyrequest\n");
    return 123456;  
  }
  void onPassKeyNotify(uint32_t pk) override {//doesn't get called
    printf("OnPassKeyNotify %d\n", pk);
  }
  bool onSecurityRequest() override {//doesn't get called
    printf("onsecurity request\n");
    return true;
  }
  void onAuthenticationComplete(esp_ble_auth_cmpl_t auth_cmpl) override {
    printf("onAuthenticationComplete\n");
    printf("pair status = %s\n", auth_cmpl.success ? "success" : "fail");
  }
  bool onConfirmPIN(uint32_t pin) override { //doesn't get called
    printf("onConfirmPin %d\n", pin);
    return true;
  }
};

bool connectToMPPT() {
  if (pMPPTDevice) {

  BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT_MITM);
  BLEDevice::setSecurityCallbacks(new MySecurityCallbacks());

  BLESecurity *pSecurity = new BLESecurity();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_ONLY | ESP_LE_AUTH_BOND);
  pSecurity->setCapability(ESP_IO_CAP_KBDISP);
  pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK | ESP_BLE_LINK_KEY_MASK);

    pMPPTClient = BLEDevice::createClient();
    pMPPTClient->setClientCallbacks(new MPPTClientCallback());

    if (pMPPTClient->connect(pMPPTDevice)) {
      printf("Connected to Server\r\n");
    
    std::map< std::string, BLERemoteService * > *pServices = pMPPTClient->getServices();

    if (pServices) {
      for (const auto &s: *pServices) {
        printf(" --- Service %s\r\n", s.second->toString().c_str());
      }
    } else {
      printf("No services found\r\n");
    }

closed 2021-06-22 by pscs