espressif/arduino-esp32 #5331: esp32 C3 arduino环境中不能使用while(1)

BG2CRW, 2021-06-25:
代码如下：
void setup() {
  while(1);
}
void loop(){
}

报错：
E (285276) task_wdt: CPU 0: loopTask
E (290269) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (290269) task_wdt:  - IDLE (CPU 0)
E (290269) task_wdt: Tasks currently 

之前的esp32中是可以程序中while(1)卡死等待条件的，不需要自己手动喂狗，期待改进

zoumiaoer, 2021-06-28:
> 代码如下：
> void setup() {
> while(1);
> }
> void loop(){
> }
>
> 报错：
> E (285276) task_wdt: CPU 0: loopTask
> E (290269) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E [...]

PLATFORMIO ARDUINO框架下怎么支持ESP32-C3啊

me-no-dev, 2021-06-28:
please use English :)
I would wildly guess that you think that there is an issue and that `while(1);` should not cause WDT, but in fact that is to be expected. You are running a "busy" loop that does not allow for other tasks on the chip to run. While loops need to have delay or to wait on some sort of event.

closed 2021-06-28 by me-no-dev