espressif/arduino-esp32 #5334: Esp32-(xQueueGenericReceive)- assert failed

eswarkeshav2001, 2021-06-27:
### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module|
|<B>IDE name</B>|Arduino IDE|
|<B>Flash Frequency</B>|80Mhz|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|

### Description:
Describe your problem here
Hey guys i have been working on a project in which the analog values sampled at a particular frequency and stored in array. Then the value will be sent to user application ESP32 BLE. But I got stuck in this error (xQueueGenericReceive)- assert failed! abort().
I am Using Esp32arduino and Free Rtos for programming i error is in the semaphore from the interrupt but i couldn't able to find out exact solution. Please help me out guys

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
##include <ArduinoJson.h>
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
 
/*#if CONFIG_FREERTOS_UNICORE
  static const BaseType_t app_cpu = 0;
#else
  static const BaseType_t app_cpu = 1;
#endif*/
 
static const BaseType_t pro_cpu = 0;
static const BaseType_t app_cpu = 1;
 
//ADC Related Global Variables
static const uint16_t timer_divider =80;
static const uint64_t timer_max_count=1000;
 
static const int adc_pin=A0;
 
static const int BUF_SIZE=1000;
 
static int buf[BUF_SIZE];
int Buff_Len=0;
static int Read=0;
static int Write=0;
static int count=0;
static float avg=0;
int i=0;
int BLE_flag=0;
String cmd;
 
static hw_timer_t *timer=NULL;
static uint16_t val;
static int count1=0;
 
static SemaphoreHandle_t bin_sem=NULL;
static SemaphoreHandle_t bin_sem2=NULL;
static portMUX_TYPE spinlock = portMUX_INITIALIZER_UNLOCKED;
//ADC Related Global Variables
 
//BLE Global Variable
char Reading[4];
BLEServer *pServer = NULL;
BLECharacteristic * pTxCharacteristic;
bool deviceConnected = false;
bool oldDeviceConnected = false;
 
//Declaration BLE necessary Classes
#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
 
class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };
 
    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};
//BLE Global Variables
 
//Task Section 
void IRAM_ATTR onTimer()
{
  BaseType_t task_woken=pdFALSE;
 
  val=analogRead(adc_pin);
  count1++;
  
  xSemaphoreGiveFromISR(bin_sem2,&task_woken);
  
  if (task_woken) {
    portYIELD_FROM_ISR();
  }
}
 
void move_to_Queue (void *parameters)
{
  while(1)
  {
   xSemaphoreTake(bin_sem2,portMAX_DELAY);
   
    if(Buff_Len==BUF_SIZE||count1>2000)
    {
      Serial.println("Buffer is full");
      xSemaphoreGive(bin_sem);
      //BLE_flag=1;
      Buff_Len=0;
      count1=0;    
    }
    else
    {
      buf[Write]=val;
      Write = (Write + 1) % BUF_SIZE;
      Buff_Len++;
    } 
  }
}
 
void BLE_Task(void *parameters)
{
   while(1)
  {
 
    xSemaphoreTake(bin_sem,portMAX_DELAY);
    Serial.println("BLE");
    for (i=0;i<BUF_SIZE;i++)
  {
   if(i==0)
  {
    sprintf(Reading,"[%d",buf[i]);
  }
  else if(i==BUF_SIZE-1)
  {
    sprintf(Reading,",%d]",buf[i]);
  }
  else{
  sprintf(Reading,",%d",buf[i]); 
  }
   Serial.print(Reading);
    if (deviceConnected) {
        pTxCharacteristic->setValue(Reading);
        pTxCharacteristic->notify();
    delay(10); // bluetooth stack will go into congestion, if too many packets are sent
  }   
 }
 Serial.println();
 
 }
}
 
 
void setup() 
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  vTaskDelay(1000/portTICK_PERIOD_MS);
  
  //BLE Declarations
  BLEDevice::init("UART Service");
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pTxCharacteristic = pService->createCharacteristic(
                    CHARACTERISTIC_UUID_TX,
                    BLECharacteristic::PROPERTY_NOTIFY
                  );
      
  pTxCharacteristic->addDescriptor(new BLE2902());
  pService->start();
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
  //BLE Declaration
 
  
  //ADC Semaphore and Timer Declarations
  bin_sem = xSemaphoreCreateBinary();
  bin_sem2 = xSemaphoreCreateBinary();
  
 
  if(bin_sem==NULL||bin_sem2==NULL)
  {
    Serial.println("Could not create semaphore");
    ESP.restart();
  }
    xTaskCreatePinnedToCore(move_to_Queue,
                          "move_to_Queue",
                          1024,
                          NULL,
                          2,
                          NULL,
                          app_cpu);
     xTaskCreatePinnedToCore(BLE_Task,
                          "BLE_Task",
                          2048,
                          NULL,
                          2,
                          NULL,
                          app_cpu);
                  
     timer = timerBegin(0, timer_divider, true);
 
  // Provide ISR to timer (timer, function, edge)
     timerAttachInterrupt(timer, &onTimer, true);
 
  // At what count should ISR trigger (timer, count, autoreload)
     timerAlarmWrite(timer, timer_max_count, true);
 
  // Allow ISR to trigger
     timerAlarmEnable(timer);
      vTaskDelete(NULL);
}
 
void loop() {
  // put your main code here, to run repeatedly:
  
 
}
```

### Debug Messages:
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1443 (xQueueGenericReceive)- assert failed! abort() was called at PC 0x4008e1d5 on core 1

Backtrace: 0x40091b38:0x3ffe0b20 0x40091d69:0x3ffe0b40 0x4008e1d5:0x3ffe0b60 0x400d1a2d:0x3ffe0ba0 0x4008e525:0x3ffe0be0

Rebooting... ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) configsip: 0, SPIWP:0xee clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00 mode:DIO, clock div:1 load:0x3fff0018,len:4 load:0x3fff001c,len:1044 load:0x40078000,len:8896 load:0x40080400,len:5816 entry 0x400806ac
```
Enable Core debug level: 
```
D][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegisterAppEvt (0x3ffde180), owner: <N/A> for registerApp
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegisterAppEvt (0x3ffde180), owner: registerApp
[D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffde3d8), owner: <N/A> for createService
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffde3d8), owner: createService
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffde63c), owner: <N/A> for executeCreate
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffde63c), owner: executeCreate
[D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e to service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x0028
[D][BLECharacteristic.cpp:90] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e, service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x0028
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffdf058), owner: <N/A> for executeCreate
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffdf058), owner: executeCreate
[D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffdf26c), owner: <N/A> for executeCreate
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffdf26c), owner: executeCreate
[D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: StartEvt (0x3ffde6fc), owner: <N/A> for start
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: StartEvt (0x3ffde6fc), owner: start
[D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[I][BLEDevice.cpp:554] getAdvertising(): create advertising
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:196] start(): - no services advertised
Waiting a client connection to notify...
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 0]
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 6]

chegewara, 2021-06-27:
There is no obvious issue, because code seem to not use semaphore before it is created, but just in case move that code into top of setup():
```
  bin_sem = xSemaphoreCreateBinary();
  bin_sem2 = xSemaphoreCreateBinary();
  
 
  if(bin_sem==NULL||bin_sem2==NULL)
  {
    Serial.println("Could not create semaphore");
    ESP.restart();
  }
  ```

The other issue i can see is that both task stack size is too small. Try 3-4kB.

eswarkeshav2001, 2021-06-27:
> There is no obvious issue, because code seem to not use semaphore before it is created, but just in case move that code into top of setup():
>
> ```
>   bin_sem = xSemaphoreCreateBinary();
>   bin_sem2 = [...]

No man it didn't help when i move the semaphore above the setup i getting an error "bin_sem doesn't name a type" and i also increased stack size to 4096 but still i got the same xqueuegenericreceive error. But i appreciate your help brother.

chegewara, 2021-06-27:
> when i move the semaphore above the setup

Not above, just before other code in setup:
```
void setup()
{
  Serial.begin(115200);
  bin_sem = xSemaphoreCreateBinary();
  bin_sem2 = xSemaphoreCreateBinary();
  
 
  if(bin_sem==NULL||bin_sem2==NULL)
  {
    Serial.println("Could not create semaphore");
    ESP.restart();
  }

eswarkeshav2001, 2021-06-27:
> Not above, just before other code in setup:
>
> ```
> void setup()
> {
>   Serial.begin(115200);
>   bin_sem = xSemaphoreCreateBinary();
>   bin_sem2 = xSemaphoreCreateBinary(); [...]

Still same error buddy

chegewara, 2021-06-27:
The `BUFF_SIZE` is 1000, but:
`char Reading[4];`
You have memory corruption, which leads to that weird error.

eswarkeshav2001, 2021-06-27:
> The `BUFF_SIZE` is 1000, but:
> `char Reading[4];`
> You have memory corruption, which leads to that weird error.

Damn man! it worked .very thanks for your help brother .But i still don't get how it worked. let me explain what i thought was happening 
I have an array of 1000 values and i am taking value one by one from that array and i am storing that single value in char array then after sending the value i am overwriting the 4 bits of char array for next value. now here why the size of char should be 1000.however i just gonna store only one number in char array and overwrite the array for the next value right? i think for a single number size of 1000 is very big. So could you please explain me how you figured it out what my mistake is .because my curiosity is killing me here.

chegewara, 2021-06-27:
It is simple, just read code.
You want to do what you described, but there is issue. Maybe i am mistaken too and 100 is wrong value, because i didnt read code very carefully.
The clue was log output when all values are printed. I thought you create full json and then send it, but maybe you just missing trailing `0` in char string. In that case char array size is 5.
Important is to read log output, thats why you add it to code. When last log before crash is array, then something after that is causing crash.

eswarkeshav2001, 2021-06-27:
got it buddy. Thanks for helping me out.it really means a lot

closed 2021-06-28 by eswarkeshav2001