espressif/arduino-esp32 #5335: How to use httpclient instead of  wificlient in this example

shariq-azim, 2021-06-28:
Hello All,
I am following the example mentioned in AWS ota esp32 example
https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/AWS_S3_OTA_Update/AWS_S3_OTA_Update.ino

But in this case i am having problems with wificlient,so want to use httpclient instead..my question how can we do (what parts of the code should we modify) the OTA  replacing wificlient .

Do we need to include SPIFFS /littleFs/Fatfs here at all?

Also which method should we prefer in case either of the above three are used?

Please suggest.

vlastahajek, 2021-07-09:
@shariq-azim, What issues do you have with WifiClient?
You don't need to rewrite this at all. Just use [httpUpdate example](../blob/master/libraries/HTTPUpdate/examples/httpUpdate/httpUpdate.ino) and change URL to your S3 bucket URL.

VojtechBartoska, 2022-03-30:
Hello @shariq-azim, sorry for replying to you late. Are you still looking for help?

shariq-azim, 2022-04-08:
> @shariq-azim, What issues do you have with WifiClient? You don't need to rewrite this at all. Just use [httpUpdate example](../blob/master/libraries/HTTPUpdate/examples/httpUpdate/httpUpdate.ino) and change URL to your S3 bucket URL.

Hey,
Thank you somehow i managed working with wificlient itself. But thank you for your interest.

Regards,
Shariq

closed 2022-04-08 by shariq-azim