espressif/arduino-esp32 #5337: Timer functions do not work at all in PlatformIO

maxgerhardt, 2021-06-28:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation version:			?1.0.6 as well as 2.0.0-alpha1 and current master 21947ebe76cf6380fe06aa885390b86cc477709a? 
IDE name:						?Platform.io? 
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no?
Upload Speed:					?921600?
Computer OS:                                        ?Windows 10? 

### Description:

I am attempting to replicate the tutorial [Hardware Interrrupts](https://www.youtube.com/watch?v=qsflCf6ahXU&list=PLEBQazB0HUyQ4hAPU1cJED6t3DU0h34bz&index=10) here, specifically the code [here](https://github.com/ShawnHymel/introduction-to-rtos/blob/main/09-hardware-interrupts/esp32-freertos-09-demo-timer-interrupt/esp32-freertos-09-demo-timer-interrupt.ino). The sketch uses functions from `esp32-hal-timer.c` to setup timer 0 with an alarm that will trigger the execution of an ISR in which a pin is toggled at a rate of 1Hz.

When uploading this sketch in PlatformIO, using all default settings (meaning also core version 1.0.6), this does not work. There is no blinking going on the defined LED pin at all.

The following sketch was used, which additionally prints some debug information on the `Serial`. A LED is connected to IO2. 

```cpp
#include <Arduino.h>
// Settings
static const uint16_t timer_divider = 80;
static const uint64_t timer_max_count = 1000000;

// Pins (change this if your Arduino board does not have LED_BUILTIN defined)
static const int led_pin = 2;

// Globals
static hw_timer_t *timer = NULL;

//*****************************************************************************
// Interrupt Service Routines (ISRs)

// This function executes when timer reaches max (and resets)
void IRAM_ATTR onTimer() {
  // Toggle LED
  int pin_state = digitalRead(led_pin);
  digitalWrite(led_pin, !pin_state);
}

//*****************************************************************************
// Main (runs as its own task with priority 1 on core 1)

void setup() {
  Serial.begin(115200);
  Serial.println("Firmware start");

  // Configure LED pin
  pinMode(led_pin, OUTPUT);

  // Create and start timer (num, divider, countUp)
  timer = timerBegin(0, timer_divider, true);

  // Provide ISR to timer (timer, function, edge)
  timerAttachInterrupt(timer, &onTimer, true);

  // At what count should ISR trigger (timer, count, autoreload)
  timerAlarmWrite(timer, timer_max_count, true);

  // Allow ISR to trigger
  timerAlarmEnable(timer);
}

void loop() {
  Serial.println("Timer seconds: " + String(timerReadSeconds(timer)));
  Serial.println("Timer alarm seconds: " + String(timerAlarmReadSeconds(timer)));
  Serial.println("Timer is alarm enabled: " + String(timerAlarmEnabled(timer)));
  Serial.println("Timer is started: " + String(timerStarted(timer)));
  delay(250);
}
```
[... 195 lines cut ...]
Timer is alarm enabled: 1
Timer is started: 1
Timer seconds: 0.50
Timer alarm seconds: 1.00
Timer is alarm enabled: 1
Timer is started: 1
Timer seconds: 0.75
Timer alarm seconds: 1.00
Timer is alarm enabled: 1
Timer is started: 1
Timer seconds: 0.00
Timer alarm seconds: 1.00
Timer is alarm enabled: 0
Timer is started: 1
Timer seconds: 0.25
```
Now the timer seconds to read correctly as `1.00` but after 1 elapsed second, the alarm is just disabled and never continues...

However from there, no matter where I add `volatile` in the register struct definition, I can not get it to function correctly. There might be additional issues somewhere else.

In any case, I see this is a bug. For 1.0.6 it's a bug reproducable with PlatformIO and also fixable as I've shown, As for 2.0.0-alpha1 and the current master version, I see the same bug source (still no `volatile` on timer regs) but that is not the only needed thing to keep it going. 

Are you able to reproduce the issue in PlatformIO? 

Magically per the linked video, Arduino IDE seems to work with some versions though? Can anyone test this with the latest master?

me-no-dev, 2021-06-28:
Hi @maxgerhardt ! Thanks for the detailed submission :)
Good hunch on the `volatile` as for the other issue, it seems that edge interrupt does not work on ESP32, so giving `false` for `edge` in `timerBegin` would have done the job. Anyway I will add a "fix" for this in just a sec :)

closed 2021-06-28 by me-no-dev

maxgerhardt, 2021-06-28:
You are right, with `volatile` in the timer regs and calling `timerAttachInterrupt(timer, &onTimer, false);` instead of `timerAttachInterrupt(timer, &onTimer, true);` in the sketch, this now properly works on the latest core version, too :)

williamesp2015, 2021-08-29:
I was very frustrated with ESP32 RepeatTimer example (above example) but this solved the problem hanging on timer using PlatformIo.
volatile uint8_t InterruptFlag = 0;
hw_timer_t * timer = NULL;
void onTimer(){
   // Serial.println(millis());
    InterruptFlag = 1;
}
void startTimer(void){ //long Timerrus) {
  InterruptFlag = 0;
  timer = timerBegin(0, 80, true); // timer_id = 0; divider=80; countUp = true;
  timerAttachInterrupt(timer, &onTimer, true); // edge = true
  timerAlarmWrite(timer, 100000, true);  //100 ms
  timerAlarmEnable(timer);
  delay(10);
}

void stopTimer() {
  if(timer)
  {
    timerEnd(timer);
  timer = NULL;
  InterruptFlag = 0;
  Serial.println("Timer Stopped");
}
}