espressif/arduino-esp32 #5338: Stop the bot

Garito, 2021-06-28:
Please...

VojtechBartoska, 2021-07-07:
Hi @Garito,

can you please specify more what is the main problem you are facing with the Stale bot? if you provide more descriptive feedback we will take it in consideration.

At this point using a Stale bot is necessary for us to keep track of current issues and being able to handle them. We are trying to provide you as much help as we can but still our resources are limited and not all the issues can be solved.

If your issue was closed by Stale Bot, just comment it to open it again. 

Thanks in advance for your answer

Garito, 2021-07-07:
My issue with the bot is the mail storms it does from time to time

Garito, 2021-07-08:
For instance the one of today at 2 am UTC with 31 bot messages

JohnMacrae, 2021-07-08:
I just have a filter in my email :)

me-no-dev, 2021-07-08:
> My issue with the bot is the mail storms it does from time to time

unfortunately the bot was off for some time, so issues accumulated and caused the email storms. After the one this morning, I hope everything will work good from now on and we will see occasional few emails.

Garito, 2021-07-08:
Another one at 9 UTC
I'm leaving this project
Sorry

closed 2021-07-12 by VojtechBartoska