espressif/arduino-esp32 #5340: 2.0.0-alpha1 BLEScan Compile Error with ESP32S2 Dev Profile

GaiaBlitz, 2021-06-29:
### Hardware:
Board:							Gravitech Cucumber MI (ESP32S2-WROOM-I, using ESP32S2 Dev Module profile)
Core Installation version:			2.0.0-alpha1
IDE name:						Arduino IDE 1.8.15
Flash Frequency:					80Mhz
PSRAM enabled:                                     No
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
I am using my board in a project which utilises BLE scanning, and as such wanted to test whether the board could scan for BLE servers/beacons. However, I came across the below compile error when attempting to compile the "BLE_scan" example code (re-included below) for the board, using the ESP32S2 Dev Module profile in the Arduino IDE.

I tested compiling the same code for ESP32 Dev Module as well as FireBeetle-ESP32 and did not get this error, thus I believe there may be something not right with the ESP32S2 profile as it is not associating the BLE classes as they should be.

### Sketch:  
```cpp

/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 5; //In seconds
BLEScan* pBLEScan;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
  // put your main code here, to run repeatedly:
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
  delay(2000);
}
```

### Debug Messages:
```

BLE_scan:12:1: error: 'BLEScan' does not name a type
 BLEScan* pBLEScan;
 ^~~~~~~
BLE_scan:14:72: error: expected class-name before '{' token
 class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
                                                                        ^
BLE_scan:15:19: error: 'BLEAdvertisedDevice' has not been declared
     void onResult(BLEAdvertisedDevice advertisedDevice) {
                   ^~~~~~~~~~~~~~~~~~~
C:\Users\jiaju\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\BLE\examples\BLE_scan\BLE_scan.ino: In member function 'void MyAdvertisedDeviceCallbacks::onResult(int)':
BLE_scan:16:66: error: request for member 'toString' in 'advertisedDevice', which is of non-class type 'int'
       Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
                                                                  ^~~~~~~~
C:\Users\jiaju\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\BLE\examples\BLE_scan\BLE_scan.ino: In function 'void setup()':
BLE_scan:24:3: error: 'BLEDevice' has not been declared
   BLEDevice::init("");
   ^~~~~~~~~
BLE_scan:25:3: error: 'pBLEScan' was not declared in this scope
   pBLEScan = BLEDevice::getScan(); //create new scan
   ^~~~~~~~
BLE_scan:25:14: error: 'BLEDevice' has not been declared
   pBLEScan = BLEDevice::getScan(); //create new scan
              ^~~~~~~~~
C:\Users\jiaju\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\BLE\examples\BLE_scan\BLE_scan.ino: In function 'void loop()':
BLE_scan:34:3: error: 'BLEScanResults' was not declared in this scope
   BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
   ^~~~~~~~~~~~~~
BLE_scan:36:18: error: 'foundDevices' was not declared in this scope
   Serial.println(foundDevices.getCount());
                  ^~~~~~~~~~~~
BLE_scan:38:3: error: 'pBLEScan' was not declared in this scope
   pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
   ^~~~~~~~
Multiple libraries were found for "BLEDevice.h"
 Used: C:\Users\jiaju\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-alpha1\libraries\BLE
 Not used: C:\Users\jiaju\Documents\Arduino\libraries\ArduinoBLE
exit status 1
'BLEScan' does not name a type

```

chegewara, 2021-06-29:
I hope you know that esp32 S2 does not have bluetooth?

GaiaBlitz, 2021-06-29:
Yes, I understand that the S2 does not have Bluetooth, however I am connecting a HM-10 module to it. Would I still have to use another BLE library instead of the BLE library included in the ESP32 package in this scenario?

chegewara, 2021-06-29:
Yes. This library works only with internal BLE hardware.

GaiaBlitz, 2021-06-29:
I see. Thanks for your quick response @chegewara , and apologies for any inconvenience caused.

closed 2021-06-29 by GaiaBlitz

chegewara, 2021-06-29:
No problem.