espressif/arduino-esp32 #5347: What is the schedule to support USB DFU for ESP32 S2 and C3

jonesfu, 2021-07-02:
We have project with ESP32 S2 chip, we hope use USB DFU function for low cost concern.

igrr, 2021-07-02:
@jonesfu Could you clarify, what kind of support do you expect from the Arduino-ESP32 project? USB DFU is a feature of ESP32-S2 chip, you can use with existing tool such as [dfu-util](http://dfu-util.sourceforge.net/).

Regarding ESP32-C3, unlike ESP32-S2 this chip doesn't have a programmable USB device controller, so it doesn't support DFU.

jonesfu, 2021-07-02:
How to use dfu-util on ESP32 S2?

jonesfu, 2021-07-02:
Another, the best way is doing DFU by arduino

me-no-dev, 2021-07-02:
@jonesfu any particular reason why not use USB-CDC instead? It uses the same USB lines and works out of the box.

lbernstone, 2021-07-02:
I get that C3 doesn't have a USB controller, but is it possible to get simple console output on the USB for debugging purposes?  I can program easily with CDC in the bootloader, but am essentially blind after that.  Sorry to hijack.

jonesfu, 2021-07-05:
> @jonesfu any particular reason why not use USB-CDC instead? It uses the same USB lines and works out of the box.

Because we expect remove USB-CDC chip for cost concern

me-no-dev, 2021-07-05:
@jonesfu I do not mean to use the USB-UART chip connected to TX and RX, I mean using a USB port connected to GPIOs 18 and 19, booting into download and uploading using esptool. You do not need external chip for that. Download mode on S2 supports both DFU and CDC flashing over the internal USB port

jonesfu, 2021-07-08:
@me-no-dev Yes! After connect USB port by USB D- to GPIO20 and USB D+ to GPIO19,  and press boot key when ESP32 S2 chip power up, then we can found USB CDC device in windows. But there are some problem as blew:

1. Under Win10, there is no default USB CDC driver for ESP32 S2, need install another driver as https://www.esp32.com/viewtopic.php?t=15222

2. under Arduino PlatformIO build environment, it will error report after flash firmware bin as:
ERROR: ESP32-S2FH32 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this error, set --after option to 'no_reset'.

We think it should be WARNING, not treated as ERROR, could you please fix it?

me-no-dev, 2021-07-08:
the message comes from esptool to notify you that it can not auto-reboot the board and you need to do it manually. You can submit an issue in esptool's repository, if you want to talk to the devs about changing the message. But even if changed, esptool as binary needs to return error anyway, because executables either succeed or fail in execution.

igrr, 2021-07-08:
https://github.com/espressif/esptool/issues/631 has changed the "ERROR" message to "WARNING", this fix can be included next time esptool is updated.

igrr, 2022-04-05:
This is fixed in the v2.0.3-rc1 release.

closed 2022-04-05 by igrr