espressif/arduino-esp32 #5348: Error abort () was called at PC 0x4013ed4f on core 0 in ESPAsyncWebServer

IhorAntiukhov, 2021-07-02:
Hello everyone! I am using ESP32-CAM Ai-Thinker board. I made a sketch below, with which I try to asynchronously send a video stream in MJPEG format to an HTTP server (for this I use the built-in camera module that comes with the board). Before downloading this sketch, you need to install a set of ESP32 boards and the (ESPAsyncWebServer)[https://github.com/me-no-dev/ESPAsyncWebServer] and (AsyncTCP)[https://github.com/me-no-dev/AsyncTCP] libraries (If you have an ESP8266 board, you need the (ESPAsyncTCP)[https://github.com/me-no-dev/ESPAsyncTCP] library).

`#include <WiFi.h>
#include <WiFiClient.h>
#include "esp_camera.h"

#include <Arduino.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"

#define PWDN_GPIO_NUM  32
#define RESET_GPIO_NUM -1
#define XCLK_GPIO_NUM   0
#define SIOD_GPIO_NUM  26
#define SIOC_GPIO_NUM  27

#define Y9_GPIO_NUM    35
#define Y8_GPIO_NUM    34
#define Y7_GPIO_NUM    39
#define Y6_GPIO_NUM    36
#define Y5_GPIO_NUM    21
#define Y4_GPIO_NUM    19
#define Y3_GPIO_NUM    18
#define Y2_GPIO_NUM     5
#define VSYNC_GPIO_NUM 25
#define HREF_GPIO_NUM  23
#define PCLK_GPIO_NUM  22

#define SSIDName "1"
#define SSIDPass "0000000000"

camera_fb_t * fb = NULL;
AsyncWebServer server(80);

String ESP32IP;

const char header[] = "HTTP/1.1 200 OK\r\n" \
                      "Access-Control-Allow-Origin: *\r\n" \
                      "Content-Type: multipart/x-mixed-replace; boundary=123456789000000000000987654321\r\n";
const char boundary[] = "\r\n--123456789000000000000987654321\r\n";
const char ctntType[] = "Content-Type: image/jpeg\r\nContent-Length: ";
const int hdrLength = strlen(header);
const int bdrLength = strlen(boundary);
const int cntLength = strlen(ctntType);
char jpegBuffer[32];
int photoSize;

bool setupCamera() {
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;

  if (psramFound()) {
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Возникла следующая ошибка камеры: 0x%x ", err);
    return false;
  }

  sensor_t * s = esp_camera_sensor_get();
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1);
    s->set_brightness(s, 1);
    s->set_saturation(s, -2);
  }
  return true;
}

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);
  Serial.begin(115200);

  if (setupCamera()) {
    Serial.println("Камера успешно настроена!");
  } else {
    Serial.println("Не удалось настроить камеру :(");
    return;
    ESP.restart();
  }
}

void loop() {
  if (WiFi.status() != WL_CONNECTED) {
    WiFi.begin(SSIDName, SSIDPass);
    if (WiFi.waitForConnectResult() == WL_CONNECTED) {
      Serial.println("Подключились к WiFi сети!");
      IPAddress LocalIP = WiFi.localIP();
      for (int i = 0; i < 4; i ++) {
        ESP32IP += i ? "." + String(LocalIP[i]) : String(LocalIP[i]);
      }
      Serial.println("Local IP: " + String(ESP32IP));
      Serial.println("Video stream IP: http://" + String(ESP32IP) + "/mjpeg");

      server.on("/mjpeg", HTTP_GET, [] (AsyncWebServerRequest *request) {
        AsyncResponseStream *response = request->beginResponseStream("text/plain");
        
        response->write(header, hdrLength);
        response->write(boundary, bdrLength);

        while(true) {
          if (fb) {
            esp_camera_fb_return(fb);      
          }
          fb = esp_camera_fb_get();

          photoSize = fb->len;
          response->write(ctntType, cntLength);
          sprintf(jpegBuffer, "%d\r\n\r\n", photoSize);
          response->write(jpegBuffer, strlen(jpegBuffer));
          response->write((char *)fb->buf, photoSize);
          response->write(boundary, bdrLength);
        }
      });
      server.begin();
    }
  }
}`

When I run this sketch, after starting the Web server, a message appears in the COM port:

abort () was called at PC 0x4013ed3b on core 0

Backtrace: 0x400911a4: 0x3ffddc90 0x400913d5: 0x3ffddcb0 0x4013ed3b: 0x3ffddcd0 0x4013ed82: 0x3ffddcf0 0x4013e697: 0x3ffddd10 0x4013e786: 0x3ffddd30 0x4013e73d: 0x3ffddd50 0x400da73f: 0x3ffddd70 0x400da785: 0x3ffddd90 0x400d6e9c: 0x3ffdddb0 0x400d6ec0: 0x3ffdddd0 0x4014dc25: 0x3ffdddf0 0x400d1325: 0x3ffdde10 0x400d8735: 0x3ffdde40 0x400d63e1: 0x3ffdde90 0x400d64af: 0x3ffdded0 0x400d6735: 0x3ffddf10 0x400d2895: 0x3ffddf30 0x400d2915: 0x3ffddf70 0x400d2fae: 0x3ffddf90 0x4008d8ed: 0x3ffddfc0

lbernstone, 2021-07-02:
https://github.com/espressif/arduino-esp32#decoding-exceptions

IhorAntiukhov, 2021-07-03:
I decoded the backtrace with EspExceptionDecoder and got this:

0x400911a4: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x400913d5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x4013ed3b: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x4013ed82: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x4013e697: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x4013e786: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x4013e73d: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc line 32
0x400da73f: cbuf::resize(unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\cbuf.cpp line 49
0x400da785: cbuf::resizeAdd(unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\cbuf.cpp line 35
0x400d6e9c: AsyncResponseStream::write(unsigned char const*, unsigned int) at D:\projects\tech\arduino\libraries\ESPAsyncWebServer-master\src\WebResponses.cpp line 690
0x4014dc25: Print::write(char const*, unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h line 71
0x400d1331: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ) at D:\projects\tech\arduino\home\Async_MJPEG_Stream_Server/Async_MJPEG_Stream_Server.ino line 139
0x400d8735: AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d63e1: AsyncWebServerRequest::_parseLine() at D:\projects\tech\arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 581
0x400d64af: AsyncWebServerRequest::_onData(void*, unsigned int) at D:\projects\tech\arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 123
0x400d6735: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ,  ,  ,  ) at D:\projects\tech\arduino\libraries\ESPAsyncWebServer-master\src\WebRequest.cpp line 76
0x400d2895: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d2915: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at D:\projects\tech\arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 1210
0x400d2fae: _async_service_task(void*) at D:\projects\tech\arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 162
0x4008d8ed: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

me-no-dev, 2021-07-05:
https://gist.github.com/me-no-dev/d34fba51a8f059ac559bf62002e61aa3

closed 2021-09-19 by stale[bot]