espressif/arduino-esp32 #5349: Installation instructions for Mac OS have some problems

Tinyu-Zhao, 2021-07-02:
system:macos Big Sur 11.4 

An error occurred during the execution of the following code and failed to solve it after many attempts
`mkdir -p ~/Documents/Arduino/hardware/espressif && \
cd ~/Documents/Arduino/hardware/espressif && \
git clone https://github.com/espressif/arduino-esp32.git esp32 --depth 1 && \
cd esp32 && \
git submodule update --init --recursive --depth 1 && \
cd tools && \
python get.py `

Error code in the terminal:
[image]

lbernstone, 2021-07-02:
Are you able to download https://github.com/espressif/crosstool-NG/releases/download/esp-2020r3/xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-macos.tar.gz in a browser?  You do not seem to have the necessary SSL certificate root.

Tinyu-Zhao, 2021-07-02:
> Are you able to download https://github.com/espressif/crosstool-NG/releases/download/esp-2020r3/xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-macos.tar.gz in a browser? You do not seem to have the necessary SSL certificate root.

Yes, it works in my browser

marcelstoer, 2021-07-03:
Isn't this exactly what's documented in the 2nd to last bullet point at https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md#installation-instructions-for-mac-os?

closed 2021-09-03 by Tinyu-Zhao