espressif/arduino-esp32 #5351: Security status about wifi vulnerability: fragattacks

Rotzbua, 2021-07-03:
## Description

It seems that there is currently no release available which uses a fixed idf version.
This issue should help to show the current status about the patches and releases.

## Status of arduino-esp32 

| arduino-esp32 |  ESP-IDF | vulnerable |
| ---------------- | ----------| ---------|
| older than 1.0.6 |  x | yes ⚠️  |
| 1.0.6 |  v3.3.5 | yes ⚠️  |
| 2.0.0-alpha1 | IDF master cf457d412 | yes ⚠️  |

## Fixed IDF releases

* Master ( ef127ab9 )
* Release v4.3.1 ( 46144f70 )
* Release v4.2.2 ( 60ccb3fe )
* Release v4.1.2 ( 97c8be71 )
* Release v4.0.4 ( 7504329e )
* Release v3.3.6 ( b403b0db )

## Reference

* Bug homepage: https://www.fragattacks.com/
* Bug report `esp-idf`: https://github.com/espressif/esp-idf/issues/7019
* Advisory: https://www.espressif.com/sites/default/files/advisory_downloads/AR2021-003%20Security%20Advisory%20for%20WFA%20vulnerability.pdf

## Disclaimer

I am not a maintainer. Ping team if necessary.

VojtechBartoska, 2021-09-22:
opening

me-no-dev, 2021-09-22:
This is already in master and in 2.0.0

closed 2021-09-22 by me-no-dev

Rotzbua, 2021-09-22:
@me-no-dev Will older v1 get a fix?

me-no-dev, 2021-09-22:
not in Arduino. We can not re-release 1.0.6 for example.

Rotzbua, 2021-09-23:
I agree, a re-release of an existing version is always a mess. But it is possible to release a new `1.0.7` or are there problems with the Arduino IDE which I am not familiar? I thought the branch `release/v1.0` is kept for critical fixes?