espressif/arduino-esp32 #5352: Enable websocket support in 2.0.0?

simon-jouet, 2021-07-03:
### Question:
Would it be possible to enable CONFIG_HTTPD_WS_SUPPORT in the config now that the esp-idf has been updated? 

I've noticed #4925 that referenced the old esp-idf as being the issue but now that it's been updated it should be fine? Or is there any other reason preventing this?

### Hardware:
Board:							esp32-s2 dev module
Core Installation version:			2.0.0-alpha1
IDE name:						Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        XUbuntu

me-no-dev, 2021-07-16:
Support is now in github master and 2.0.0 will be out in just a bit :)

closed 2021-07-16 by me-no-dev

simon-jouet, 2021-07-17:
Thanks @me-no-dev

SebiTimeWaster, 2021-08-29:
Is there a way to enforce this option until "2.0.0 will be out in just a bit"?

lbernstone, 2021-08-29:
rc2 was released 6 days ago.  It includes the [CONFIG](https://github.com/espressif/arduino-esp32/blob/2.0.0-rc2/tools/sdk/esp32/sdkconfig#L674)

SebiTimeWaster, 2021-08-29:
and how do i set this up? i cant find anything on platformio to specify a framework version, and if i set `framework = arduino@2.0.0-rc2` i just get `Error: This board doesn't support arduino@2.0.0-rc2 framework!` on compile...

SebiTimeWaster, 2021-08-29:
ok, i think i can set it up with 
```
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git@2.0.0-rc2
```
it downloads it, but then i get 
```
Error: Package version 0.0.0+sha.a62979d doesn't satisfy requirements 2.0.0-rc2 based on PackageMetaData <type=tool name=framework-arduinoespressif32 version=0.0.0+sha.a62979d spec={'owner': 'platformio', 'id': None, 'name': 'framework-arduinoespressif32', 'requirements': '2.0.0-rc2', 'url': 'git+https://github.com/espressif/arduino-esp32.git'}
```
and i have no clue where this `0.0.0+sha.a62979d` even comes from...

SebiTimeWaster, 2021-08-30:
...it comes from the package json. but that doesnt help, even after forking and setting the right value i run into linker exceptions.

SebiTimeWaster, 2021-09-02:
...i removed the arduino core.