espressif/arduino-esp32 #5353: Maximum indicated program size is wrong when using custom partitions.csv

pgrawehr, 2021-07-04:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:		1.0.6
IDE name:						Arduino IDE 1.8.13 on Windows 10
Flash Frequency:				80Mhz
PSRAM enabled:                  no

### Description:
From an ESP32 example code, I extracted the following part, that just prints the flash partition list to the console. 

```cpp
#include <string.h>
#include <assert.h>
#include "esp_partition.h"
#include "esp_log.h"

static const char *TAG = "example";

void setup(void)
{
    Serial.begin(115200);
    delay(200); // Otherwise, the first log entries get lost
    ESP_LOGI(TAG, "Start test");
    esp_partition_iterator_t iter = esp_partition_find(ESP_PARTITION_TYPE_APP, ESP_PARTITION_SUBTYPE_ANY, NULL);
    ESP_LOGI(TAG, "Name, type, subtype, offset, length");
    while (iter != nullptr)
    {
      const esp_partition_t *partition = esp_partition_get(iter);
      ESP_LOGI(TAG, "%s, app, %d, 0x%x, 0x%x (%d)", partition->label, partition->subtype, partition->address, partition->size, partition->size);
      iter = esp_partition_next(iter);
    }
    
    esp_partition_iterator_release(iter);
    iter = esp_partition_find(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, NULL);
    while (iter != nullptr)
    {
      const esp_partition_t *partition = esp_partition_get(iter);
      ESP_LOGI(TAG, "%s, data, %d, 0x%x, 0x%x (%d)", partition->label, partition->subtype, partition->address, partition->size, partition->size);
      iter = esp_partition_next(iter);
    }
    
    esp_partition_iterator_release(iter);

    
}

void loop()
{
}
```

In the project folder, I also have a custom `partitions.csv` file as follows:
```
nvs,        data, nvs,      0x9000,  0x6000,
phy_init,   data, phy,      0xf000,  0x1000,
factory,    app,  factory,  0x10000, 512k,
storage,    data, ,         ,        0x40000,
```

If I run the code, it correctly shows the custom partitions with the above sizes, everything like it should be.

**However**, the build output says the program uses 207278 Bytes from a maximum of 1310720. If I change the partition scheme selection in the tools menu, the second number changes. I would expect the maximum size to be equal to the size of the `factory` partition from the custom partition scheme. 

If I change the factory partition to be only 100k in size, the code can successfully be uploaded, but the ESP32 fails to boot and enters a core-dump-restart-loop. 

### Expected behavior

If a custom `partitions.csv` file is present, it's factory partition size is used as the maximum program size, and the selection in the partition scheme menu of the Arduino IDE is ignored. Uploading a firmware with a size that exceeds the factory partition size should fail.

me-no-dev, 2021-07-05:
ArduinoIDE has no way of reading the partition scheme you have in your sketch folder, so it is not possible for it to judge how much space there is. I recommend you to set in the menu to use the "huge app" scheme, so that you do not get errors for space.

closed 2021-07-05 by me-no-dev

pgrawehr, 2021-07-05:
@me-no-dev Are these values not coming from the toolchain? When looking at the output, it seems to be the result of running `xtensa-esp32-elf-size`. Note that only a few lines above it, `gen_esp32part.exe` runs to generate the partition map. 

This is the end of the build log (messages are in german, but you should get the point):
```
"C:\\Users\\Patrick\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\3.0.0/esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\Patrick\\AppData\\Local\\Temp\\arduino_build_625578/ExtendedConfigurableFirmata.ino.bin" "C:\\Users\\Patrick\\AppData\\Local\\Temp\\arduino_build_625578/ExtendedConfigurableFirmata.ino.elf"
esptool.py v3.0-dev
"C:\\Users\\Patrick\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/gen_esp32part.exe" -q "C:\\Users\\Patrick\\AppData\\Local\\Temp\\arduino_build_625578/partitions.csv" "C:\\Users\\Patrick\\AppData\\Local\\Temp\\arduino_build_625578/ExtendedConfigurableFirmata.ino.partitions.bin"
Bibliothek ConfigurableFirmata in Version 2.10.1 im Ordner: C:\Users\Patrick\Documents\Arduino\libraries\ConfigurableFirmata  wird verwendet
Bibliothek Wire in Version 1.0.1 im Ordner: C:\Users\Patrick\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Wire  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\Patrick\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SPI  wird verwendet
Bibliothek FS in Version 1.0 im Ordner: C:\Users\Patrick\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS  wird verwendet
Bibliothek FFat in Version 1.0 im Ordner: C:\Users\Patrick\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FFat  wird verwendet
"C:\\Users\\Patrick\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\Patrick\\AppData\\Local\\Temp\\arduino_build_625578/ExtendedConfigurableFirmata.ino.elf"
Der Sketch verwendet 401378 Bytes (30%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 15976 Bytes (4%) des dynamischen Speichers, 311704 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
```

Of course I can set the partition scheme to "huge". This will prevent an error during build, but it will not prevent that I can upload a program that is larger than the allocated code partition. And since for my project, I need a data partition that is as big as possible, I want to keep the code partition as small as possible, but have the compiler complain when added features extend the size.

me-no-dev, 2021-07-05:
the values do not come from the toolchain but from `boards.txt` menu: https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L71

You could edit the menu and add an potion with your necessary size, but that will get overwritten when you update arduino-esp32