espressif/arduino-esp32 #5354: Failed to use WiFi.setDNS()

hutje, 2021-07-05:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.6
IDE name:						Arduino IDE v1.8.8
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Can't use WiFi.setDNS()

I want to set the DNS server without a static IP.
Only change the DNS, but it wont work.

### Debug Messages:
I get an error:
```
init.h:351:8: error: 'class WiFiClass' has no member named 'setDNS'

   WiFi.setDNS(mqtt_server.c_str());

        ^
```

killgroove, 2021-07-14:
Hello,

Same problem here, did you found out any solution ?

Thanks!

hutje, 2021-07-14:
> Hello,
> 
> Same problem here, did you found out any solution ?
> 
> Thanks!

For me the DNS was an option to enable secure certificate SSL by checking the server. This way I hoped I could work with a secure SSL, but due to the core update of ESP32 currently it is not working.
A workaround for me was to enable insecure SSL connection. This way the certificate is only being used to encrypt the message, but not to check the end user.
So currently I do not need it, but hope it will work in the future.

AdamMiltonBarker, 2021-07-23:
> For me the DNS was an option to enable secure certificate SSL by checking the server. This way I hoped I could work with a secure SSL, but due to the core update of ESP32 currently it is not [...]

Hi mate what has changed with ESP32 ? I had two boards that were connected to a private MQTT secured by letsencrypt. I reuploaded code to one of them and it could no longer connect to the broker, after a week of trying to get it working again I gave up. I have another device that has been connected to the broker since December last year, today I uploaded the code again with some minor modifications and it now has exactly the same issue it doesn't even reach the broker, so there is an issue with ESP32 core and SSL ?

AdamMiltonBarker, 2021-07-23:
> For me the DNS was an option to enable secure certificate SSL by checking the server. This way I hoped I could work with a secure SSL, but due to the core update of ESP32 currently it is not [...]

I worked out how to solve this, just revert the ESP32 board back to 1.0.4 and it will work again.

hutje, 2021-07-23:
Currently you need to choose setInsecure() to disable server IP check with New core version

AdamMiltonBarker, 2021-07-23:
It does not work with the new version

floek, 2022-02-13:
The answer is here: https://github.com/espressif/arduino-esp32/issues/5590

VojtechBartoska, 2022-02-15:
@floek Thanks for the link! Closing this one.

closed 2022-02-15 by VojtechBartoska