espressif/arduino-esp32 #5357: Ble mode  ON/OFF

savagerex, 2021-07-06:
when i turn on / off  ble .mode .

about 20 times.

then esp32 will reboot

### Hardware:
Board : ESP32 dev module
Core install version: 1.0.4 & 1.0.6
IDE name : Arduino
Flash Frequency : 80MHz
PSRAM Enable : yes
upload speed :115200
computer os : win10

below is code

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

#define BUTTON_BLE        34
int sensorValueblewifi = 0;
BLEServer *pServer = NULL;
BLECharacteristic * pTxCharacteristic;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint8_t txValue = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();

      if (rxValue.length() > 0) {
        Serial.println("*********");
        Serial.print("Received Value: ");
        for (int i = 0; i < rxValue.length(); i++)
          Serial.print(rxValue[i]);

        Serial.println();
        Serial.println("*********");
      }
    }
};

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("UART Service");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
                    CHARACTERISTIC_UUID_TX,
                    BLECharacteristic::PROPERTY_NOTIFY
                  );
                      
  pTxCharacteristic->addDescriptor(new BLE2902());

  BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
                       CHARACTERISTIC_UUID_RX,
                      BLECharacteristic::PROPERTY_WRITE
                    );

  pRxCharacteristic->setCallbacks(new MyCallbacks());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");

}

void BleUartSetup() {
  //Serial.begin(115200);
  Serial.println("[BleUart] MethodCall BleUartSetup");

  // Create the BLE Device
  BLEDevice::init("UART Service");
  //BLEDevice::setMTU(mtu);

  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
                    CHARACTERISTIC_UUID_TX,
                    BLECharacteristic::PROPERTY_NOTIFY
                  );

  pTxCharacteristic->addDescriptor(new BLE2902());

  BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
                       CHARACTERISTIC_UUID_RX,
                      BLECharacteristic::PROPERTY_WRITE
                    );

  pRxCharacteristic->setCallbacks(new MyCallbacks());

  // Start the service
  pService->start();

  // Start advertising
  // pServer->getAdvertising()->start(); // for backward compability
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.printf("[BleUart] UART Service started(BLE address: %s), waiting a client connection to notify ...\n", BLEDevice::getAddress().toString().c_str());
}

void BleStart()
{
    BleUartSetup();
}

void BleStop()
{
   BLEDevice::deinit(false);
}

void loop() {
  // put your main code here, to run repeatedly:
  sensorValueblewifi = analogRead(BUTTON_BLE);

  //Serial.println(sensorValueblewifi);
  //delay(100);

  if (2715 < sensorValueblewifi && sensorValueblewifi < 3374)
  {
      Serial.println("Start");
      BleStart();
  }

  if (1430 < sensorValueblewifi && sensorValueblewifi < 1744)
  {
      Serial.println("Stop");
      BleStop();
  }

  

}

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
![image](https://user-images.githubusercontent.com/79502707/124541978-1c1a3c80-de55-11eb-9bf5-e9a902154444.png)

savagerex, 2021-07-06:
looks like memory not enough!!!

[image]

me-no-dev, 2021-07-07:
Hi @savagerex , please use https://github.com/me-no-dev/EspExceptionDecoder to decode the backtrace

savagerex, 2021-07-07:
below is right???

the file only have (.java) ,  no ( .jar)   ?????

[image]

me-no-dev, 2021-07-07:
You need the zip from releases: https://github.com/me-no-dev/EspExceptionDecoder/releases/tag/1.1.0

savagerex, 2021-07-07:
How to use ??

[image]

me-no-dev, 2021-07-07:
https://github.com/me-no-dev/EspExceptionDecoder#usage there is an explanation in the repository readme

savagerex, 2021-07-07:
i has been change ble mode, many time.

but don't have exception.

[image]

me-no-dev, 2021-07-07:
You need to copy the lines starting with `abort()` and `Backtrace: ` to the top pane of the exception decoder

savagerex, 2021-07-07:
Hello   here  !!

[image]

me-no-dev, 2021-07-07:
Thanks! We will look into it

me-no-dev, 2021-07-07:
Seems that the version you are running is a bit old. Can you please try and use the development board manager link and install 2.0.0-beta https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json

Or even better to install the github version which contains quite a few fixes (including BT/BLE): https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

savagerex, 2021-07-07:
OK ,   I  try it  .

THX

savagerex, 2021-07-07:
there is no 2.0.0-beta ????

[image]

savagerex, 2021-07-07:
this website
(https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json)

paste it in preference .   right???

[image]

me-no-dev, 2021-07-07:
yes, where currently says `https://dl.espressif.com/dl/package_esp32_index.json`. Replace it with the new link

savagerex, 2021-07-07:
choose 2.0.0

[image]

savagerex, 2021-07-07:
One question!!

i saw memory keeping low.

is normal??

me-no-dev, 2021-07-07:
It is normal to have less heap on the new version. But generally should not continuously drop when you are starting/stopping BLE

savagerex, 2021-07-07:
OK!!  we wiil test other function in 2.0.0.

if this version is stable, i afraid this version have other issue.

THX

closed 2021-09-22 by stale[bot]