espressif/arduino-esp32 #5358: Sharing variables on between tasks

eswarkeshav2001, 2021-07-06:
Make your question, not a Statement, inclusive. Include all pertinent information:
Why i am not able to write typical Global variable from multiple tasks and Why the if loops continuously iterating in a Task when there are other task to run.

What you are trying to do?
I am trying to Sample a ADC and Send the values to the Mobile Using BLE. Here i have Used a Hardware timer to Sample ADC once i done sampling i want to stop the timer for optimize efficiency .And when there is  input from User Via Bluetooth connection the timer should start again and ADC sample has to be sent again. Here i haven't Implement my BLE receiving part So well . But Used a global variable to simulate it. 

Describe your system( Hardware, computer, O/S, core version, environment).
Processor :Intel i5
Windows :10
Ide: arduino

Describe what is failing.

When the Global variable set by the BLE task(Task2) the Timer_turn on task (Task3 has to run).But it didnt't work when i used GLOBAL variable t3 to act like a flag it didn't start the Timer_turn on task(tassk 3) didn't run.So i Used actual Flag this time Timer Turnon  task in only running rest of the task are not running.It feels like if  condition at timer turn on task is running continously without giving proper timer for other task. And then i Used Seamphore to signal the Timerturon once BLE task has completed ts work it owrks perfectly.BUt still i want to know why Flag method is going crazy and Global variable method is not working. and i used freertos here.Please help me out guys.Without knowing the answer i am not even able to sleep. 
 
Show the shortest possible code that will duplicate the error.

void Timer_Turnon(void *parameters)
{
  while(1)
  {
    if(t3==1)
  //xSemaphoreTake(bin_sem2,portMAX_DELAY);
  { 
  { 
     timerStart(timer);
     Serial.println("Timer started again"); 
     t3=0;
  }
 }
}
} 

void BLE_Task(void *parameters)
{
   while(1)
  {
    //Serial.println("Task3 on work");
    //if(BLE_flag==1)
    {
    //BLE_flag=0;
    xSemaphoreTake(bin_sem,portMAX_DELAY);
    Serial.println("BLE");
    for (i=0;i<BUF_SIZE;i++)
  {
   if(i==0)
  {
    sprintf(Reading,"[%d",buf[i]);
  }
  else if(i==BUF_SIZE-1)
  {
    sprintf(Reading,",%d]",buf[i]);
  }
  else{
  sprintf(Reading,",%d",buf[i]); 
  }
   Serial.print(Reading);
    if (deviceConnected) {
        pTxCharacteristic->setValue(Reading);
        pTxCharacteristic->notify();
        delay(10); // bluetooth stack will go into congestion, if too many packets are sent
  }   
 }
 Serial.println();
 
  }
  t3=1;
  //xSemaphoreGive(bin_sem2);
 }
}
[... 289 lines cut ...]
12:19:36.585 -> [D][BLEService.cpp:225] addCharacteristic(): Adding characteristic: uuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e to service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x0028
12:19:36.585 -> [D][BLECharacteristic.cpp:90] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e, service: UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, handle: 0x0028
12:19:36.632 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffdf4ec), owner: <N/A> for executeCreate
12:19:36.632 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffdf4ec), owner: executeCreate
12:19:36.632 -> [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
12:19:36.632 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3ffdf700), owner: <N/A> for executeCreate
12:19:36.632 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3ffdf700), owner: executeCreate
12:19:36.679 -> [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
12:19:36.679 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: StartEvt (0x3ffdeb90), owner: <N/A> for start
12:19:36.679 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: StartEvt (0x3ffdeb90), owner: start
12:19:36.679 -> [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
12:19:36.679 -> [I][BLEDevice.cpp:554] getAdvertising(): create advertising
12:19:36.725 -> [D][BLEDevice.cpp:556] getAdvertising(): get advertising
12:19:36.725 -> [D][BLEAdvertising.cpp:196] start(): - no services advertised
12:19:36.725 -> Waiting a client connection to notify...
12:19:36.725 -> [D][BLEDevice.cpp:556] getAdvertising(): get advertising
12:19:36.725 -> [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 0]
12:19:36.725 -> [D][BLEDevice.cpp:556] getAdvertising(): get advertising
12:19:36.725 -> [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
12:19:36.725 -> [D][BLEDevice.cpp:556] getAdvertising(): get advertising
12:19:36.725 -> [D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 6]
12:19:37.711 -> Buffer is full
12:19:37.711 -> [4095,312,288,292,292,284,276,286,234,184,74,.........1000]

[image]

PilnyTomas, 2022-02-24:
Hi, your post was very difficult to read and the code is confusing too. I tried to edit the post (not the code) to make it more readable - please check that I didn't change the meaning.
I would suggest you edit the code:

1. Remove unused (commented) code.
2. Remove unnecessary curly brackets: '{' and '}'.
3. **Instead of global variable use [FreeRTOS queue](https://www.freertos.org/a00018.html) to pass data between tasks**
4. NEVER use an infinite loop in a task without any yielding or blocking calls - no other task will get to be executed!
  
   - Yield in tasks i.e. use blocking functions to wait on something - for example [xQueueReceive](https://www.freertos.org/a00118.html) or use [vTaskDelay](https://www.freertos.org/a00127.html) for periodic execution.

6. `bin_sem` and `bin_sem2` are created twice in the `setup()` function - remove one of the creation.
7. Let the FreeRTOS decide on which core the task will run - instead of `xTaskCreatePinnedToCore` use [xTaskCreate](xTaskCreate) - this does the same as the previous function, only without core specification.

Please try these tips and if it doesn't fix your issue let me know and post the updated code.

VojtechBartoska, 2022-03-16:
Closing due no response.

closed 2022-03-16 by VojtechBartoska