espressif/arduino-esp32 #5360: fatal error: soc/soc_caps.h: No such file or directory

Gokula-Krishnan-Sekar, 2021-07-06:
Hi. i just followed your instruction but i got this error. i'm using esp-idf V4.0.

root@kali:~/Desktop/esp-idf-arduino-test# idf.py build
Checking Python dependencies...
Python requirements from /root/esp/esp-idf/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running cmake in directory /root/Desktop/esp-idf-arduino-test/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /root/Desktop/esp-idf-arduino-test"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.30.2") 
CMake Warning at /root/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/lwip/lwip is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /root/esp/esp-idf/tools/cmake/build.cmake:78 (git_submodule_check)
  /root/esp/esp-idf/tools/cmake/build.cmake:159 (__build_get_idf_git_revision)
  /root/esp/esp-idf/tools/cmake/idf.cmake:43 (__build_init)
  /root/esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:5 (include)

-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 418fb4d-dirty
-- Building ESP-IDF components for target esp32
-- Found PythonInterp: /root/.espressif/python_env/idf4.0_py2.7_env/bin/python (found version "2.7.18") 
-- Found Perl: /usr/bin/perl (found version "5.32.1") 
-- Adding linker script /root/Desktop/esp-idf-arduino-test/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /root/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /root/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /root/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /root/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /root/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /root/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /root/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update arduino asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /root/esp/esp-idf/components/app_trace /root/esp/esp-idf/components/app_update /root/Desktop/esp-idf-arduino-test/components/arduino /root/esp/esp-idf/components/asio /root/esp/esp-idf/components/bootloader /root/esp/esp-idf/components/bootloader_support /root/esp/esp-idf/components/bt /root/esp/esp-idf/components/coap /root/esp/esp-idf/components/console /root/esp/esp-idf/components/cxx /root/esp/esp-idf/components/driver /root/esp/esp-idf/components/efuse /root/esp/esp-idf/components/esp-tls /root/esp/esp-idf/components/esp32 /root/esp/esp-idf/components/esp_adc_cal /root/esp/esp-idf/components/esp_common /root/esp/esp-idf/components/esp_eth /root/esp/esp-idf/components/esp_event /root/esp/esp-idf/components/esp_gdbstub /root/esp/esp-idf/components/esp_http_client /root/esp/esp-idf/components/esp_http_server /root/esp/esp-idf/components/esp_https_ota /root/esp/esp-idf/components/esp_https_server /root/esp/esp-idf/components/esp_local_ctrl /root/esp/esp-idf/components/esp_ringbuf /root/esp/esp-idf/components/esp_rom /root/esp/esp-idf/components/esp_websocket_client /root/esp/esp-idf/components/esp_wifi /root/esp/esp-idf/components/espcoredump /root/esp/esp-idf/components/esptool_py /root/esp/esp-idf/components/expat /root/esp/esp-idf/components/fatfs /root/esp/esp-idf/components/freemodbus /root/esp/esp-idf/components/freertos /root/esp/esp-idf/components/heap /root/esp/esp-idf/components/idf_test /root/esp/esp-idf/components/jsmn /root/esp/esp-idf/components/json /root/esp/esp-idf/components/libsodium /root/esp/esp-idf/components/log /root/esp/esp-idf/components/lwip /root/Desktop/esp-idf-arduino-test/main /root/esp/esp-idf/components/mbedtls /root/esp/esp-idf/components/mdns /root/esp/esp-idf/components/mqtt /root/esp/esp-idf/components/newlib /root/esp/esp-idf/components/nghttp /root/esp/esp-idf/components/nvs_flash /root/esp/esp-idf/components/openssl /root/esp/esp-idf/components/partition_table /root/esp/esp-idf/components/protobuf-c /root/esp/esp-idf/components/protocomm /root/esp/esp-idf/components/pthread /root/esp/esp-idf/components/sdmmc /root/esp/esp-idf/components/soc /root/esp/esp-idf/components/spi_flash /root/esp/esp-idf/components/spiffs /root/esp/esp-idf/components/tcp_transport /root/esp/esp-idf/components/tcpip_adapter /root/esp/esp-idf/components/ulp /root/esp/esp-idf/components/unity /root/esp/esp-idf/components/vfs /root/esp/esp-idf/components/wear_levelling /root/esp/esp-idf/components/wifi_provisioning /root/esp/esp-idf/components/wpa_supplicant /root/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /root/Desktop/esp-idf-arduino-test/build
Running ninja in directory /root/Desktop/esp-idf-arduino-test/build
Executing "ninja all"...
[268/938] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.30.2") 
CMake Warning at /root/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/lwip/lwip is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /root/esp/esp-idf/tools/cmake/build.cmake:78 (git_submodule_check)
  /root/esp/esp-idf/tools/cmake/build.cmake:159 (__build_get_idf_git_revision)
  /root/esp/esp-idf/tools/cmake/idf.cmake:43 (__build_init)
  /root/esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:20 (include)

-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
[... 178 lines cut ...]
 const IPAddress INADDR_NONE(0, 0, 0, 0);
                                ^
../components/arduino/cores/esp32/IPAddress.h:94:35: error: expected declaration specifiers or '...' before numeric constant
 const IPAddress INADDR_NONE(0, 0, 0, 0);
                                   ^
../components/arduino/cores/esp32/IPAddress.h:94:38: error: expected declaration specifiers or '...' before numeric constant
 const IPAddress INADDR_NONE(0, 0, 0, 0);
                                      ^
In file included from ../components/arduino/libraries/WiFi/src/WiFi.h:29,
                 from ../main/main.c:7:
../components/arduino/cores/esp32/IPv6Address.h:29:1: error: unknown type name 'class'
 class IPv6Address: public Printable
 ^~~~~
../components/arduino/cores/esp32/IPv6Address.h:29:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 class IPv6Address: public Printable
                  ^
In file included from ../components/arduino/libraries/WiFi/src/WiFiSTA.h:28,
                 from ../components/arduino/libraries/WiFi/src/WiFi.h:32,
                 from ../main/main.c:7:
../components/arduino/libraries/WiFi/src/WiFiGeneric.h:28:10: fatal error: functional: No such file or directory
 #include <functional>
          ^~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

me-no-dev, 2021-07-06:
Use IDF master instead :) We have not tested against all IDF versions yet

Krish402, 2021-07-07:
Please tell me. Which idf version you have tried. Now I'm using idf V3.3 and arduino core 1.0.6. still it showing error

lbernstone, 2021-07-07:
You can see the idf commit versions associated with arduino-esp32 releases at https://github.com/lbernstone/arduino-esp32-dockerfiles.  There is also a link there to docker containers with ready-to-compile source code.

Gokula-Krishnan-Sekar, 2021-07-07:
I want to compile lwip custom library.

Krish402, 2021-07-08:
recently i'm getting this error. 

In file included from /home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiSTA.h:28:0,
                 from /home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFi.h:32,
                 from /home/pi/esp/wifirepeater/components/arduino/libraries/ArduinoOTA/src/ArduinoOTA.h:4,
                 from /home/pi/esp/wifirepeater/components/arduino/libraries/ArduinoOTA/src/ArduinoOTA.cpp:6:
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:81:2: error: 'wifi_event_sta_scan_done_t' does not name a type
  wifi_event_sta_scan_done_t wifi_scan_done;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:82:2: error: 'wifi_event_sta_authmode_change_t' does not name a type
  wifi_event_sta_authmode_change_t wifi_sta_authmode_change;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:83:2: error: 'wifi_event_sta_connected_t' does not name a type
  wifi_event_sta_connected_t wifi_sta_connected;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:84:2: error: 'wifi_event_sta_disconnected_t' does not name a type
  wifi_event_sta_disconnected_t wifi_sta_disconnected;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:85:2: error: 'wifi_event_sta_wps_er_pin_t' does not name a type
  wifi_event_sta_wps_er_pin_t wps_er_pin;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:86:2: error: 'wifi_event_sta_wps_fail_reason_t' does not name a type
  wifi_event_sta_wps_fail_reason_t wps_fail_reason;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:87:2: error: 'wifi_event_ap_probe_req_rx_t' does not name a type
  wifi_event_ap_probe_req_rx_t wifi_ap_probereqrecved;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:88:2: error: 'wifi_event_ap_staconnected_t' does not name a type
  wifi_event_ap_staconnected_t wifi_ap_staconnected;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:89:2: error: 'wifi_event_ap_stadisconnected_t' does not name a type
  wifi_event_ap_stadisconnected_t wifi_ap_stadisconnected;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:90:2: error: 'wifi_event_ftm_report_t' does not name a type
  wifi_event_ftm_report_t wifi_ftm_report;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:91:2: error: 'ip_event_ap_staipassigned_t' does not name a type
  ip_event_ap_staipassigned_t wifi_ap_staipassigned;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:92:2: error: 'ip_event_got_ip_t' does not name a type
  ip_event_got_ip_t got_ip;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:93:2: error: 'ip_event_got_ip6_t' does not name a type
  ip_event_got_ip6_t got_ip6;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:94:2: error: 'smartconfig_event_got_ssid_pswd_t' does not name a type
  smartconfig_event_got_ssid_pswd_t sc_got_ssid_pswd;
  ^
/home/pi/esp/wifirepeater/components/arduino/libraries/WiFi/src/WiFiGeneric.h:95:2: error: 'esp_eth_handle_t' does not name a type
  esp_eth_handle_t eth_connected;
  ^
make[1]: *** [/home/pi/esp/esp-idf/make/component_wrapper.mk:290: libraries/ArduinoOTA/src//ArduinoOTA.o] Error 1
make: *** [/home/pi/esp/esp-idf/make/project.mk:552: component-arduino-build] Error 2

closed 2021-09-22 by stale[bot]