espressif/arduino-esp32 #5361: arduino-esp32 for specific ESP-IDF

superuser789, 2021-07-07:
I am trying to build micropython  project which is based on ESPIDFv3.3.2  - 9e70825d1e1cbf7988cf36981774300066580ea7. I am using some libraries from arduino-esp32 in my project.  Building micropython with latest arduino-esp32 or any other releases always shows compilation error. 
For getting specific ESP-IDF for particular arduno-esp32,  I checkout the specific ESP-IDF commit from [CONFIG_ARDUINO_IDF_COMMIT](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/config/sdkconfig.h#L661).

How can I do that in reverse i.e. getting compatible `arduino-esp32` for  `ESP-IDF v3.3.2  - 9e70825d1e1cbf7988cf36981774300066580ea7`   ?

me-no-dev, 2021-07-07:
you can try to use the release/v1.0 branch of this repo, but you might need to go further back. Posting some of those build errors might help

superuser789, 2021-07-07:
> you can try to use the release/v1.0 branch of this repo, but you might need to go further back. Posting some of those build errors might help

Here is the common error, I am getting : 
```

root@localhost:~/twatch# idf.py reconfigure && idf.py build
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from /root/esp/esp-idf/requirements.txt are satisfied.
Running cmake in directory /root/twatch/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized /root/twatch"...
Warn about uninitialized values.
-- Building for target esp32
-- ccache will be used for faster builds
-- Building empty aws_iot component due to configuration
-- Component names: soc log heap xtensa-debug-module app_trace freertos vfs newlib esp_ringbuf driver esp_event ethernet mbedtls micro-ecc efuse bootloader_support partition_table app_update spi_flash nvs_flash lwip tcpip_adapter pthread smartconfig_ack wpa_supplicant espcoredump esp32 cxx console mdns esp_adc_cal wear_levelling sdmmc fatfs spiffs openssl nimble bt arduino ArduinoJson AsyncTCP ESP32Ping ESP8266_Spiram ESP8266Audio ESP8266SAM ESPAsyncWebServer IRremoteESP8266 TTGO_TWatch_Library micropython pubsubclient asio jsmn aws_iot bootloader coap nghttp esp-tls tcp_transport esp_http_client esp_http_server esp_https_ota esp_https_server esptool_py expat freemodbus idf_test json libsodium mqtt protobuf-c protocomm ulp unity wifi_provisioning main
-- Component paths: /root/esp/esp-idf/components/soc;/root/esp/esp-idf/components/log;/root/esp/esp-idf/components/heap;/root/esp/esp-idf/components/xtensa-debug-module;/root/esp/esp-idf/components/app_trace;/root/esp/esp-idf/components/freertos;/root/esp/esp-idf/components/vfs;/root/esp/esp-idf/components/newlib;/root/esp/esp-idf/components/esp_ringbuf;/root/esp/esp-idf/components/driver;/root/esp/esp-idf/components/esp_event;/root/esp/esp-idf/components/ethernet;/root/esp/esp-idf/components/mbedtls;/root/esp/esp-idf/components/micro-ecc;/root/esp/esp-idf/components/efuse;/root/esp/esp-idf/components/bootloader_support;/root/esp/esp-idf/components/partition_table;/root/esp/esp-idf/components/app_update;/root/esp/esp-idf/components/spi_flash;/root/esp/esp-idf/components/nvs_flash;/root/esp/esp-idf/components/lwip;/root/esp/esp-idf/components/tcpip_adapter;/root/esp/esp-idf/components/pthread;/root/esp/esp-idf/components/smartconfig_ack;/root/esp/esp-idf/components/wpa_supplicant;/root/esp/esp-idf/components/espcoredump;/root/esp/esp-idf/components/esp32;/root/esp/esp-idf/components/cxx;/root/esp/esp-idf/components/console;/root/esp/esp-idf/components/mdns;/root/esp/esp-idf/components/esp_adc_cal;/root/esp/esp-idf/components/wear_levelling;/root/esp/esp-idf/components/sdmmc;/root/esp/esp-idf/components/fatfs;/root/esp/esp-idf/components/spiffs;/root/esp/esp-idf/components/openssl;/root/esp/esp-idf/components/nimble;/root/esp/esp-idf/components/bt;/root/twatch/components/arduino;/root/twatch/components/ArduinoJson;/root/twatch/components/AsyncTCP;/root/twatch/components/ESP32Ping;/root/twatch/components/ESP8266_Spiram;/root/twatch/components/ESP8266Audio;/root/twatch/components/ESP8266SAM;/root/twatch/components/ESPAsyncWebServer;/root/twatch/components/IRremoteESP8266;/root/twatch/components/TTGO_TWatch_Library;/root/twatch/components/micropython;/root/twatch/components/pubsubclient;/root/esp/esp-idf/components/asio;/root/esp/esp-idf/components/jsmn;/root/esp/esp-idf/components/aws_iot;/root/esp/esp-idf/components/bootloader;/root/esp/esp-idf/components/coap;/root/esp/esp-idf/components/nghttp;/root/esp/esp-idf/components/esp-tls;/root/esp/esp-idf/components/tcp_transport;/root/esp/esp-idf/components/esp_http_client;/root/esp/esp-idf/components/esp_http_server;/root/esp/esp-idf/components/esp_https_ota;/root/esp/esp-idf/components/esp_https_server;/root/esp/esp-idf/components/esptool_py;/root/esp/esp-idf/components/expat;/root/esp/esp-idf/components/freemodbus;/root/esp/esp-idf/components/idf_test;/root/esp/esp-idf/components/json;/root/esp/esp-idf/components/libsodium;/root/esp/esp-idf/components/mqtt;/root/esp/esp-idf/components/protobuf-c;/root/esp/esp-idf/components/protocomm;/root/esp/esp-idf/components/ulp;/root/esp/esp-idf/components/unity;/root/esp/esp-idf/components/wifi_provisioning;/root/twatch/main
CMake Warning at /root/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain /root/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
  crosstool-ng version crosstool-ng-1.22.0-97-gc752ad5d doesn't match
  supported version crosstool-ng-1.22.0-80-g6c4433a.  Check Getting Started
  documentation or proceed at own risk.
Call Stack (most recent call first):
  /root/esp/esp-idf/tools/cmake/idf_functions.cmake:214 (crosstool_version_check)
  /root/esp/esp-idf/tools/cmake/idf_functions.cmake:373 (idf_verify_environment)
  /root/esp/esp-idf/tools/cmake/project.cmake:173 (idf_import_components)
  CMakeLists.txt:16 (project)

-- IDF_VER: v3.3.2
-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Adding linker script /root/twatch/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /root/esp/esp-idf/components/esp32/ld/esp32.rom.ld
[... cut ...]
../components/arduino/libraries/WiFi/src/ETH.cpp: In member function 'IPAddress ETHClass::dnsIP(uint8_t)':
../components/arduino/libraries/WiFi/src/ETH.cpp:196:37: error: conversion from 'const ip_addr_t* {aka const ip_addr*}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested
     ip_addr_t dns_ip = dns_getserver(dns_no);
                                     ^
[86/1097] Building CXX object esp-idf/arduino/CMakeFiles/idf_component_arduino.dir/libraries/WiFi/src/WiFiProv.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

me-no-dev, 2021-07-07:
you can try commit 6e5be7883811ba910743ebd0215708ca658bbaeb for that ETH change. In a similar way if you have other errors, just check when they were changed in the file and back one commit

closed 2021-09-22 by stale[bot]