espressif/arduino-esp32 #5363: Cannot Initialise Bluetooth inside called function ESP32-Wroom

AtefMohamed, 2021-07-07:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1-rc4
IDE name:						?Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    ?no?
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I need to initialise and uninitialised Bluetooth inside the code for ESP32 it works with a simple toggle switch but in my code, I cannot fix or found the reason

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch

#include "Functions.h"
//*************************************************
bool flag = false;
bool flag2 = false;
extern bool LCD_Flag;
extern bool Config_Flag;
//extern  char* pref_ssid;
//extern  char* pref_pass;
extern BluetoothSerial SerialBT;
extern uint8_t count;
extern bool Button_Pressed;
float Sensors_Data [6]= {0,0,0,0,0,0};
//enum Sensors {Air_Temp , Air_Pressure , Soil_Temp , Soil_Pressure , Auto_1 , Led_1 };
unsigned long str, crt , elpsd = 0;
unsigned long str2, crt2 , elpsd2 = 0;
void setup() {
  Serial.begin(9600);
  
  delay(50);
  LCD_Init();
  Sensors_Init();
  delay(10);
  WiFi_Connect();
  delay(50);
  Firebase_Init();
  delay(50);
  Button_init();
  delay(10);
//  WiFi.end();
//  delay(100);
//  Serial.println("disconnected bluetooth  ");
//  Bluetooth_Connect();
//  delay(7000);
//  SerialBT.end();
//  delay(10000);
//  Serial.println("disconnected bluetooth  ");
//  WiFi_Connect();
//str = millis();

}

void loop()
{
// crt = millis();
// elpsd = crt - str;
// crt2 = millis();
// elpsd2 = crt2 - str2;
// if( elpsd > 120000     ){
//
// // if (flag == true){
//      Serial.println(" bawasl bluetooth  ");
//  Bluetooth_Connect();
//  str  = crt;
//  }
////  flag = false;
////  }
////  else flag== true;
//// }
// 
// if( elpsd2 > 200000   ){
////  if (flag2 == true){
//    Serial.println("End bluetooth  ");
//    SerialBT.end();
//    str2  = crt2;
//  }
//    flag2 =false;
//  }
//  }
//  else flag2 = true;
// 
 
//   WiFi_Connect();
   Read_Sensors();
   Read_Data();
   Serial.print(Sensors_Data[Air_Temp]);
   Serial.print("  ");
   Serial.print(Sensors_Data[Air_Pressure]);
   Serial.print("  ");
   Serial.print(Sensors_Data[Soil_Temp]);
   Serial.print("  ");
   Serial.print(Sensors_Data[Soil_Pressure]);
   Serial.print("  ");
   Serial.print(Sensors_Data[Auto_1]);
   Serial.print("  ");
   Serial.println(Sensors_Data[Led_1]);
   //LCD_Print();
   Data_Send();
   delay(50); 
   LCD_Clear();
   Button_State();
   
    if(LCD_Flag){
      Serial.println(" LCD  ");
      LCD_Print();
      delay(2000);
      LCD_Flag = false;
     }
    if(Config_Flag){
//      Serial.println(" Config ");
      LCD_Print_Config();
      delay(1000);
//      Bluetooth_Connect();
    btStart();
//   Serial.println(" khagrt mn Btstart  ");
   bool stateBT = SerialBT.begin("ESP32test"); //Bluetooth device name 
   delay(5000);
   Serial.println(" khagrt mn Bluetooth");
   
      delay(1000);
      Config_Mode();
      Config_Flag = false;
    } 
    
   while(Wifi_Check()){  // handle Wifi failure 
    LCD_Wifi_Failure();   // print to the screen that wifi doesnt connected
    WiFi_Connect();   // restablish wifi connection
    
  }
}
 

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c46c  PS      : 0x00060730  A0      : 0x80133491  A1      : 0x3ffffbd0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00002398  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000239  A8      : 0x80082750  A9      : 0x3ffffb70  
A10     : 0x00000000  A11     : 0x00001800  A12     : 0x00000000  A13     : 0x00000003  
A14     : 0x00000020  A15     : 0x3fffc510  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000238  

ELF file SHA256: 0000000000000000

Backtrace: 0x4000c46c:0x3ffffbd0 0x4013348e:0x3ffffbe0 0x4013bb3b:0x3ffffc00 0x4013b7db:0x3ffffc20 0x4013b81b:0x3ffffc40 0x400912be:0x3ffffc70

me-no-dev, 2021-07-07:
https://github.com/me-no-dev/EspExceptionDecoder

AtefMohamed, 2021-07-07:
Decoding stack results

0x4008ffcc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x40090249: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x40090bf6: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 719
0x4013c945: btu_task_post at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c line 265
0x4013ce65: BTU_StartUp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_init.c line 184
0x4016b53f: bte_main_enable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/main/bte_main.c line 132
0x4016b56c: bte_main_boot_entry at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/main/bte_main.c line 89
0x4012853c: btc_init_bluetooth at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_main.c line 64
0x4012861c: btc_main_call_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_main.c line 109
0x40126c4d: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/btc/core/btc_task.c line 160
0x400912be: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

AtefMohamed, 2021-07-07:
PC: 0x4000c46c
EXCVADDR: 0x00000000

Decoding stack results
0x40146d9f: l2c_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c line 831
0x4013c57e: btu_init_core at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_init.c line 89
0x4013c21b: btu_task_start_up at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c line 278
0x4013c25b: btu_task_thread_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c line 226
0x400912be: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

P-R-O-C-H-Y, 2021-11-03:
Hi @AtefMohamed,
Try updating arduino-esp32 to latest version `2.0.1-RC1`.
Tested `BluetoothSerial` and it's working perfectly.

VojtechBartoska, 2021-11-18:
@AtefMohamed Any updates?

closed 2021-12-14 by VojtechBartoska