espressif/arduino-esp32 #5365: https problem

savagerex, 2021-07-08:
when i set static ip into esp32, then use https communication.  it's work.

but after esp32 reset. ip is static ip.  https communication , fail.

Hardware:
Board : ESP32 dev module
Core install version: 1.0.4 & 1.0.6
IDE name : Arduino
Flash Frequency : 80MHz
PSRAM Enable : yes
upload speed :115200
computer os : win10

me-no-dev, 2021-07-08:
Please provide a minimal sketch example, enable debug in the board menu and post the logs from the serial port.

savagerex, 2021-07-08:
Do you install "anydesk".  it can Remote control my computer.

I can show you.

me-no-dev, 2021-07-08:
Let's try to resolve it the normal way first :)
Here is what you need to enable, run the sketch and then copy here the log from the IDE Monitor
[image]

savagerex, 2021-07-08:
when i sending get 
as below
[image]

 but there is no any log

savagerex, 2021-07-08:
after esp32 reset.   http  "ok"  but  https "fail"

lbernstone, 2021-07-08:
I have no idea what we are seeing in that image, but setting a gateway outside of the subnet range is unlikely to ever work.

me-no-dev, 2021-07-08:
@savagerex I need the log from ArduinoIDE's serial monitor. The one that shows what the board outputs via Serial

savagerex, 2021-07-08:
below is log

but those are we made.

Version: 1.0.5
Build: Jul  8 2021 20:22:29
DeviceId: bedside-c4dd5779b0ac
E (1363) esp_himem: esp_himem_init: already initialized
========================================
PSRAM total size     : 8388608 
----------------------------------------
PSRAM first 4MB size : 3932108 
PSRAM first 4MB free : 3932108 
PSRAM HI-MEM    size : 4456448 
PSRAM HI-MEM    free : 4456448 
PSRAM HI-MEM reserved: 262144 
========================================
Internal RAM  size   : 240020 
Internal RAM  free   : 212340 
========================================
SPIFFS        size   : 52961 
SPIFFS        used   : 6024 
========================================
Power On GPIO INIT
LedAndBuzzerTask create success.
[threadInital]
LedBuzzerTask running on core 0
timerCallback running on core 1
timerCallback create success.
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
WifiTask running on core 1
WifiTask create success.
[BleUart] BLE already stopped
[Wifi] Start connection to V2041
[Wifi] Status: WL_DISCONNECTED
[W][SPIFFS.cpp:54] begin(): SPIFFS Already Mounted!
Reading certificate from SPIFFS.
Read 1950 bytes of certificate and 3324 bytes of key from SPIFFS
[WebServer] setup
You are running NFC_getUids.
You are running NFC_putUids put API./api/bedside/v1/nfc/uids
GetPreferences
DeletePreferences
Starting HTTPS server...
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.15.200, MASK: 255.255.255.0, GW: 192.168.186.254
[V2041 with IP: 192.168.15.200 RSSI: -34
Starting HTTPS server...
[WebServer] start
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.15.200, MASK: 255.255.255.0, GW: 192.168.186.254
V2041 with IP: 192.168.15.200 RSSI: -32
Starting HTTPS server...
[WebServer] start
BedsideWebServerTask create success.
setupSockets failed
[WebServer] start
Http Server ready.
Https Server ready.
{"UserImage":true,"DeviceId":"bedside-c4dd5779b0ac","ModelName":"ES133ERBM10","WifiMac":"C4:DD:57:79:B0:AC","BluetoothMac":"C4:DD:57:79:B0:AE","Version":"1.0.5","Build":"Jul  8 2021 20:22:29","Sn":"⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮","Mcu":"CHIP_ESP32","Ram":"8 MB","Rom":"4 MB","Vcom":1270,"It8951FwVer":".v.1","It8951LutVer":"2v02"}
[Psram] Free:3921508, Size:3932060, MinFree:3921508, MaxAlloc:3921508
[Heap] Free:125888, Size:235804, MinFree:104260, MaxAlloc:90552
[Stack] TaskGetStackHighWaterMark: 5452
[D][ulptool.h:50] ulptool_load_binary(): program_size_bytes: 120 total_size: 120 offset: 12 .text: 108, .data: 0, .bss: 8
[V][Preferences.cpp:342] getInt(): nvs_get_i32 fail: KeyTwo NOT_FOUND
[V][Preferences.cpp:342] getInt(): nvs_get_i32 fail: KeyFour NOT_FOUND

me-no-dev, 2021-07-08:
I do not know why you get GOT_IP twice and started the server twice, but as @lbernstone said, your IP configuration does not seem correct.  IP: 192.168.15.200 with MASK: 255.255.255.0 will not be able to reach GW: 192.168.186.254. GW must start with 192.168.15.

savagerex, 2021-07-08:
ok ,   i try to change gateway

savagerex, 2021-07-08:
Thank you very much!!!

when i change gateway ip , then https work!!!

me-no-dev, 2021-07-08:
great!

closed 2021-07-08 by me-no-dev