espressif/arduino-esp32 #5366: ESP32 4M & 8m

savagerex, 2021-07-08:
i use code and upload to esp32 4M. it's ok

but same code upload to esp32 4M. esp32 always reboot.

below is exception tool loag.

[image]

savagerex, 2021-07-08:
sorry 

 same code upload to esp32 8M. esp32 always reboot.

me-no-dev, 2021-07-08:
So once you flashed 4M chip and then 8M chip with the same firmware and you got different result?
The trace seem to point into stack error. Are you running out of memory? The flash size should not play a role into that.

pipi61, 2021-07-08:
try: select 8M and compile again

lbernstone, 2021-07-08:
Perhaps you are choosing 8M, but your board does not have 8M?  Please read the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md ) carefully.

savagerex, 2021-07-09:
@me-no-dev   Yes, I got different result.

same firmware upload.

me-no-dev, 2021-07-09:
@savagerex that makes no sense, unless there is something physically wrong with the board/chip. Error points at stack/memory error and should have been the same on all boards. Can you try on more boards?

savagerex, 2021-07-09:
Sorry ,  it is solved.   i choose wrong flash size.

Thx

closed 2021-07-09 by me-no-dev