espressif/arduino-esp32 #5367: SSID and passphrase/PSK may be truncated or be treated as broken

inagaki83, 2021-07-09:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Arduino IDE
Flash Frequency:					40MHz
PSRAM enabled:                                    no
Upload Speed:					460800
Computer OS:                                        Debian

### Description:
1. SSID is up to 32-octet length according to the standard. But 32-octet SSID is rejected in WiFiSTA.cpp or is truncated to 31 octets in WiFiAP.cpp.
2. 64-char PSK is truncated to 63 characters (to be broken) in WiFiAP.cpp by `strlcpy()`.
3. SoftAP with 62-char or shorter passphrases might cause "Password error" in client side (10% of board caused this issue). Filling the `wifi_ap_config_t::password[64]` using `strncpy()` resolved the issue. So it's an ESP-IDF bug and affected by some noise bytes after NUL, I guess. 
4. On the HEAD, SSID or passphrase including `%` will cause unintentional printf formatting and will be broken in WiFiAP.cpp and WiFiSTA.cpp.

SSID and passphrase/PSK should always be copied using `strncpy()`, I think.

### Sketch:
In the official example WiFiAccessPoint.ino, you should use...:
- 32-octet SSID.
- 64-character PSK.
- some `%` character in SSID and passphrase.

In the official example WiFiClient.ino, you should use...:
- 32-octet SSID.
- some `%` character in SSID and passphrase.

### Debug Messages:
No debug messages.

inagaki83, 2021-07-09:
`strcmp` is used to compare SSIDs or passphrases in WiFiAP.cpp. You should use `strncmp` for it in case 32-octet SSID or 64-char PSK comes.

closed 2021-07-16 by me-no-dev

inagaki83, 2021-08-05:
The third problem is not fixed (once fixed and newly made!) on the current [HEAD](https://github.com/espressif/arduino-esp32/blob/5bb8177aa1feec665938be97607b526a4f0ec0bb/libraries/WiFi/src/WiFiAP.cpp#L53). Or is the ESP-IDF bug is fixed?

me-no-dev, 2021-08-10:
Are you saying we should add `\0` at the end of `wifi_ap_config_t::password[64]` up to 64 bytes?

inagaki83, 2021-08-10:
Yes.
You should fill the space with \0 from the end of the string up to 64 bytes (if the string is shorter) or the ESP-IDF will be affected by noise bytes there, I think.

me-no-dev, 2021-08-10:
do you have any way to confirm this? If so, we can add the fix here and notify IDF team of the issue.

inagaki83, 2021-08-16:
I've made a test sketch but I can't reproduce the problem now.
So it doesn't have to be changed. Thank you.