espressif/arduino-esp32 #5368: SD_MMC sketch begin() function get crash with case SD-card is not present or removed out slot socket SD.

TienHuyIoT, 2021-07-09:
esp32 module: WROOM
version: install latest 2.0.0

SDMMC begin and work succeed within case the SD-card have present in slot socket SD.
If the SD-card is removed out slot socket SD. Esp32 shall crash and reset log print as below.

with version: release v1.0.6
ESP32 example SD_MMC work normal with this case.

```
rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1412
load:0x40078000,len:13416
load:0x40080400,len:3672
entry 0x400805f8
E (79) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (80) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
assertion "xQueue" failed: file "../esp-idf/components/freertos/queue.c", line 1946, function: uxQueueMessagesWaiting

abort() was called at PC 0x400eafe7 on core 1

Backtrace:0x400d8051:0x3ffb25a00x40088d41:0x3ffb25c0 0x4008e74d:0x3ffb25e0 0x400eafe7:0x3ffb2660 0x40089e89:0x3ffb2690 0x400dd4b7:0x3ffb26b0 0x400dbac1:0x3ffb26e0 0x400dbc2a:0x3ffb2720 0x400e4f95:0x3ffb2740 0x400d30e1:0x3ffb2770 0x400d2278:0x3ffb27f0 0x400d402f:0x3ffb2820 
```

_Originally posted by @TienHuyIoT in https://github.com/espressif/arduino-esp32/issues/5195#issuecomment-876983887_

me-no-dev, 2021-07-09:
please decode the `Backtrace`

TienHuyIoT, 2021-07-09:
Hi @me-no-dev ,
**This is my SDMMC_Test.ino Code.**
```
/*
 * Connect the SD card to the following pins:
 *
 * SD Card | ESP32
 *    D2       12
 *    D3       13
 *    CMD      15
 *    VSS      GND
 *    VDD      3.3V
 *    CLK      14
 *    VSS      GND
 *    D0       2  (add 1K pull up after flashing)
 *    D1       4
 */

#include "FS.h"
#include "SD_MMC.h"

#define SD_POWER_PIN     		        22
#define SD_POWER_PINMODE_INIT()         pinMode(SD_POWER_PIN, OUTPUT_OPEN_DRAIN)
#define SD_POWER_ON()                   digitalWrite(SD_POWER_PIN, LOW)
#define SD_POWER_OFF()                  digitalWrite(SD_POWER_PIN, HIGH)

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void createDir(fs::FS &fs, const char * path){
    Serial.printf("Creating Dir: %s\n", path);
    if(fs.mkdir(path)){
        Serial.println("Dir created");
    } else {
        Serial.println("mkdir failed");
    }
}

void removeDir(fs::FS &fs, const char * path){
    Serial.printf("Removing Dir: %s\n", path);
    if(fs.rmdir(path)){
        Serial.println("Dir removed");
[... 156 lines cut ...]
    Serial.printf("Used space: %lluMB\n", SD_MMC.usedBytes() / (1024 * 1024));
}

void loop(){

}

```
**And my Exception Deocder**

```
Decoding stack results
0x400d8051: panic_abort at ../esp-idf/components/esp_system/panic.c line 390
0x40088d41: esp_system_abort at ../esp-idf/components/esp_system/esp_system.c line 126
0x4008e74d: abort at ../esp-idf/components/newlib/abort.c line 46
0x400eafe7: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c line 62
0x40089e89: uxQueueMessagesWaiting at ../esp-idf/components/freertos/queue.c line 1946
0x400dd4b7: spi_bus_remove_device at ../esp-idf/components/driver/spi_master.c line 434
0x400dbac1: deinit_slot at ../esp-idf/components/driver/sdspi_host.c line 223
0x400dbc2a: sdspi_host_remove_device at ../esp-idf/components/driver/sdspi_host.c line 267
0x400e4f95: esp_vfs_fat_sdmmc_mount at ../esp-idf/components/fatfs/vfs/vfs_fat_sdmmc.c line 361
0x400d30e1: fs::SDMMCFS::begin(char const*, bool, bool) at C:\Users\HUY-GAMING\Documents\Arduino\hardware\espressif\esp32\libraries\SD_MMC\src\SD_MMC.cpp line 76
0x400d2278: setup() at C:\Users\HUY-GAMING\Documents\Arduino\generated_examples\SDMMC_Test/SDMMC_Test.ino line 187
0x400d402f: loopTask(void*) at C:\Users\HUY-GAMING\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 35
```

diegonunesbr, 2021-08-16:
I have the same problem. With the card present, the code works very well.

Without the card, the driver try to deinit SPI bus, but SDMMC was initied in SDMMC 4 bits mode, not SPI.
My board all 6 data pins connected, exactly as source code provied by TienHuyIoT