espressif/arduino-esp32 #5370: Core 2.0.0 complie error

savagerex, 2021-07-09:
when i complie the code in core 1.0.4 , it's ok

but i complie the code in core 2.0.0 , it's fail
----------------------------- Remove above -----------------------------

### Hardware:
Hardware:
Board : ESP32 dev module
Core install version: 2.0.0
IDE name : Arduino
Flash Frequency : 80MHz
PSRAM Enable : yes
upload speed :115200
computer os : win10

below is error

[image]

me-no-dev, 2021-07-09:
did you restart the IDE? something is missing from the package

savagerex, 2021-07-09:
Yes!!  i has restarted Arduino IDE

it still appear error

[image]

me-no-dev, 2021-07-09:
then try uninstalling and reinstalling the package. You have missing/wrong files.

savagerex, 2021-07-09:
i has install esp32 https server package.

[image]

but still have error.

savagerex, 2021-07-09:
all lib ,  i has reinstall.

me-no-dev, 2021-07-09:
not the libs! the core! the esp32 core version 2.0.0beta1 is that you need to reinstall

savagerex, 2021-07-09:
OK!!!  i try it

savagerex, 2021-07-09:
what device i need to choose??

before i choose "ESP32 Dev Module".

[image]

savagerex, 2021-07-09:
there is 2.0.0 alpha1.  don't have 2.0.0beta1

[image]

Jason2866, 2021-07-10:
alpha is the correct one. There is no beta

savagerex, 2021-07-11:
After reinstall core  2.0.0 alpha1, still have error!!!

me-no-dev, 2021-07-12:
rename the library folder to NOT have chinese symbols.

savagerex, 2021-07-12:
the path and folder is no chinese.

[image]

me-no-dev, 2021-07-12:
Not the core... your libraries:

[image]

savagerex, 2021-07-12:
now i can complie.

Crashes during compilation.

Arduino ide will close by itself.

savagerex, 2021-07-12:
it always stuck in here!!

[image]

me-no-dev, 2021-07-12:
there is no error shown. Did you try another computer as Ma Li suggested?

savagerex, 2021-07-12:
Ok,i try another computer

savagerex, 2021-07-13:
I use "Arduino 1.8.15" & core 2.0.0.
it can complie.

test https.  
if all tesk enable, https fail.
if all tesk disable, https ok.

the memory is not enough.

me-no-dev, 2021-07-13:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#restrictions
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig#L529

So short answer is NO, long is that you should look into optimizing your task's stack sizes, allocate using malloc, rather than on the stack, or build your custom SDK with CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY enabled (complicated task).

savagerex, 2021-07-13:
OK!!! THX

savagerex, 2021-07-14:
@me-no-dev 

finally, we use https communication several times. the heap will Keep decreasing.
the memory don't release.

when memory is not enough, the https fail.

savagerex, 2021-07-14:
@me-no-dev 

i find use  https  "Put"  command. ESP32 heap will decrease.

how to release memory???

VojtechBartoska, 2022-03-29:
Hello @savagerex, I'm closing this issue. If you still need help you can reopen this one or create a new issue. Thanks for your contribution.

closed 2022-03-29 by VojtechBartoska